weewx/weewx #510: SQLite - Memory-Mapped IO

GarryALockyer, 2020-02-20:
Given that RPi4 can have up to 4GB of memory, would enabling SQLite Memory-Mapped IO provide a performance advantage?

Regards,

Garry

tkeffer, 2020-02-20:
It might, but it can also lead to more erratic performance. It's easy enough to give it a try: there is an undocumented feature that allows `pragmas` to be passed to the sqlite driver. Go into `weewx.conf` and change the `[[SQLite]]` stanza so it looks like this:

```
    [[SQLite]]
        driver = weedb.sqlite
        # Directory in which the database files are located
        SQLITE_ROOT = %(WEEWX_ROOT)s/archive
	[[[pragmas]]]
		mmap_size=536870912 
```

This would allocate 512MB to the memory-mapped buffers. 

Give it a try. I'll be interested to hear about your experiences.

GarryALockyer, 2020-02-20:
Thanks!  I will likely give it a try, but won’t be able to for 4+ weeks.

Regards,

Garry

closed 2020-02-20 by tkeffer