weewx/weewx #511: Could daemon.py be replaced with python-daemon?

bjmc, 2020-02-22:
Would there be any interest in replacing [`daemon.py`](https://github.com/weewx/weewx/blob/master/bin/daemon.py) with usage of the [python-daemon](https://pypi.org/project/python-daemon/) library that implements [PEP 3143?](https://www.python.org/dev/peps/pep-3143/) It would mean incurring another dependency, but it would remove some boilerplate from this codebase.

Thoughts?

tkeffer, 2020-02-22:
I don't want to incur another dependency, but, looking at the license that comes with `python-daemon` we could just include a copy.

Would you want to check that out?

bjmc, 2020-02-25:
Only if you think it's valuable. If the existing `daemon.py` has been working fine, maybe there's not much to be gained by swapping one implementation for another.

tkeffer, 2020-02-25:
It's always nice to be able to replace some bespoke code with something supported, but `daemon.py` has given me no trouble for nearly 10 years.

Let's leave it be!

closed 2020-02-25 by tkeffer