weewx/weewx #517: `wee_import` should better handle source data files containing BOM

gjr80, 2020-03-23:
Particularly applies to CSV imports as the source data files can be produced by any number of means that may or may not include a BOM. Less likely to affect WD or Cumulus imports as source data files for such imports are the actual log files produced by WD and Cumulus respectively.

`wee_import`currently strips any HTML tags and blank lines from source data files so BOMs could easily be handled there. Looking at the python csv module there may be some scope to change the encoding used by the csv reader.

tkeffer, 2020-03-23:
Another way of looking at it: the original source was a set of code points. Some source program at some point wrote them out, using an encoding that involved a BOM. You are now trying to read it back in as a text file.

The problem here is that the `open()` statement being used to read the file back in is not using the same encoding as what was used to write by the source program. Instead, it is using the locale's default encoding (probably 'UTF-8'), which is not prepared to encounter a BOM. You may get lucky and this encoding is a simple byte-to-character mapping, but that's not guaranteed.

Net-net? Another solution would be to make the encoding a user-supplied option.

gjr80, 2020-03-23:
Yes, dealing with it via the `open()` statement is the way to go, was hung up on reading about `csv.DictReader()` and BOM. I need to read a bit more about usage of BOM. Would prefer `wee_import` to automatically determine encoding but I guess a silent config option would be a good safety net.

gjr80, 2020-03-24:
So it seems detecting the file encoding is not an exact science. So how about this for an approach:

1. have a silent `encoding` config item and if set use that encoding to open the file
2. otherwise we check for UTF 8, 16-LE, 16-BE, 32-LE and 32-BE BOMs and if found open the file with that encoding
3. if 2. is indeterminate open the file as we do now but catch the error and exit gracefully advising the user they might need to check/specify/change their file encoding

tkeffer, 2020-03-24:
I think that's more elaborate than necessary. I would make it a config file option, defaulting to UTF-8. The vast majority of files will be encoded in ASCII or UTF-8.

gjr80, 2020-03-24:
> I think that's more elaborate than necessary.

What a nice way of saying 'overly complicated' or 'could be done in a much simpler way', it did not even dent my enthusiasm :)

Have submitted a PR so you can review the code before anything is committed. I guess there has been a bit of mission creep; csv is not the only source that should handle file encoding, any of the sources that import log files should similarly be treated - hence the changes to `weeimport.py`.

Python 2 `open()` doesn't do encoding so hence the `io` import, believe there are other approaches that could be taken but `io` seemed the most succinct. Believe that `io.open()` is somewhat slower than `open()` but I don't think that is a great issue here. Have briefly tested against a UTF-16 file and it failed gracefully (well more gracefully than before). UTF-16 file was correctly decoded when `source_encoding=utf-16` was set in the import config file.

If you are happy with this approach I will treat the other sources similarly and add `source_encoding` to the docs.

gjr80, 2020-03-24:
Have treated Cumulus and WD imports in the same manner as the CSV; `utf-8-sig` is assumed unless the user overrides via the `source_encoding` config option in the import config file. Commit fa081d028297d285b0c38e12f0ba0a909ae16d79 refers.

Have added `source_coding` to the `wee_import` documentation. Commit 2f87f51f1393a8bda51371e1e53d8178db93945b refers. Have refrained from including `source_encoding` in the example import config files since (1) it will likely be seldom used and (2) adding it but not setting it messed up the default used with `.get()` and adding just the description but no config option just looked plain silly.

I think we are done here.

closed 2020-04-03 by tkeffer