weewx/weewx #519: Weewx: 4.0.0b17- Missing Interpolation Option exception from line 248 of reportEngine

AlgerP572, 2020-03-29:
Hi I've been trying the new 4.0 beta and I have run into an issue in _build_skin_dict in ReportEngine.

The line is 248 from the file reportengine.py :
merge_dict = copy.deepcopy(self.config_dict['StdReport']['Defaults'])

The Exception is:
MissingInterpolationOption: 'missing option "levelname" in interpolation'

Setup:
RaspberryPi 4 with Debian Stretch
Python 3.7.3

For now I have commented this line out and everything appears to work without issue.

I have enabled the new logging if this could be related?

Regards.

tkeffer, 2020-03-29:
Could we see the full stack trace? Also, in case the trace doesn't show it, which skin?

Option `levelname` is supplied by the Python library module `logging`. Not sure how we could supply it.

AlgerP572, 2020-03-30:
Hi.

Attached is the image of where I catch the exception. Line 144 or reportengine.py
The skin is the SeasonsReport part of weewx. (Same code worked fine is b15 and b16 if this helps)

[image]

I added a traceback.print_exc(file=sys.stdout) as you can see above, below are the results.

If you need anything more I'm happy to help let me know.

Traceback (most recent call last):
  File "/home/weewx4/bin/weewx/reportengine.py", line 144, in run
    skin_dict = self._build_skin_dict(report)
  File "/home/weewx4/bin/weewx/reportengine.py", line 251, in _build_skin_dict
    merge_dict = copy.deepcopy(self.config_dict['StdReport']['Defaults'])
  File "/usr/lib/python3.7/copy.py", line 180, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/usr/lib/python3.7/copy.py", line 280, in _reconstruct
    state = deepcopy(state, memo)
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 220, in _deepcopy_tuple
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 220, in <listcomp>
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 240, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "/usr/lib/python3.7/copy.py", line 180, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/usr/lib/python3.7/copy.py", line 280, in _reconstruct
    state = deepcopy(state, memo)
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 220, in _deepcopy_tuple
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 220, in <listcomp>
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 240, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "/usr/lib/python3.7/copy.py", line 180, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/usr/lib/python3.7/copy.py", line 280, in _reconstruct
    state = deepcopy(state, memo)
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 220, in _deepcopy_tuple
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 220, in <listcomp>
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 240, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "/usr/lib/python3.7/copy.py", line 180, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/usr/lib/python3.7/copy.py", line 280, in _reconstruct
    state = deepcopy(state, memo)
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 220, in _deepcopy_tuple
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 220, in <listcomp>
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 240, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "/usr/lib/python3.7/copy.py", line 180, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/usr/lib/python3.7/copy.py", line 280, in _reconstruct
    state = deepcopy(state, memo)
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 220, in _deepcopy_tuple
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 220, in <listcomp>
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 240, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "/usr/lib/python3.7/copy.py", line 180, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "/usr/lib/python3.7/copy.py", line 280, in _reconstruct
    state = deepcopy(state, memo)
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 220, in _deepcopy_tuple
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 220, in <listcomp>
    y = [deepcopy(a, memo) for a in x]
  File "/usr/lib/python3.7/copy.py", line 150, in deepcopy
    y = copier(x, memo)
  File "/usr/lib/python3.7/copy.py", line 240, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "/usr/lib/python3.7/copy.py", line 169, in deepcopy
    rv = reductor(4)
  File "/usr/lib/python3/dist-packages/configobj.py", line 482, in __reduce__
    state = (dict(self), self.__dict__)
  File "/usr/lib/python3/dist-packages/configobj.py", line 557, in __getitem__
    return self._interpolate(key, val)
  File "/usr/lib/python3/dist-packages/configobj.py", line 549, in _interpolate
    return engine.interpolate(key, value)
  File "/usr/lib/python3/dist-packages/configobj.py", line 352, in interpolate
    value = recursive_interpolate(key, value, self.section, {})
  File "/usr/lib/python3/dist-packages/configobj.py", line 330, in recursive_interpolate
    k, v, s = self._parse_match(match)
  File "/usr/lib/python3/dist-packages/configobj.py", line 417, in _parse_match
    value, section = self._fetch(key)
  File "/usr/lib/python3/dist-packages/configobj.py", line 386, in _fetch
    raise MissingInterpolationOption(key)
  File "<string>", line None
configobj.MissingInterpolationOption: missing option "levelname" in interpolation.

tkeffer, 2020-03-30:
Thanks. 

Can you remove any passwords and send me your copy of `weewx.conf` privately? My address is tkeffer at gmail dot com.

tkeffer, 2020-03-31:
Thanks for the file. I am remembering why I did not document the use of `[Logging]`!

Still, I think I figured out a workaround. Fixed, I hope, in commit https://github.com/weewx/weewx/commit/d077a8bddc98a7b00f84ac33b17a5fa2f0823736

Give it a try.

AlgerP572, 2020-04-01:
Thank you for the fast response and quick turn around time. I'm happy to report that with this change set all seems to be functioning fine now.  Here is a screenshot from my testing:

[image]

It's been running since a little before noon today (PDT).  As you can see the console logging is working great now.  This is a great new feature as it will make debugging much easier. :)

Thanks again for your help.

kobuki, 2020-07-18:
I'm getting the same error in the logs on latest 4.1.1 (installed from Debian Buster repo [per instructions](http://weewx.com/docs/debian.htm)), and no html report is generated at all. Using default config with metric settings and the weewx-meteoRX driver that I've just finished porting to Python 3. Are the changes discussed here will get integrated to the upstream packages any time soon?

tkeffer, 2020-07-18:
Log?

kobuki, 2020-07-18:
Sorry?

tkeffer, 2020-07-18:
Python 3.6 changed how deepcopy works. Effectively, rather than a faithful copy of a ConfigObj, the copy did the interpolation early.

This is fixed in the current `master` on GitHub. You have two choices:
1. Use the current version on GitHub, which is quite stable, or
2. Use Python 3.5 or Python 2.7.

kobuki, 2020-07-18:
Yeah, thanks. Please also see my comment about the log stanza. If that's not present (copy-pasted from the Wiki), the error disappears. For now, that's the workaround I use (not using file logging). I have no idea how the logging relates to deepcopy(), though.

closed 2020-07-18 by tkeffer

tkeffer, 2020-07-18:
The configuration file is read in and converted to a `ConfigObj` object, a dictionary-like structure. Internally, WeeWX needs to modify this structure, so it makes a deep copy first. Unfortunately, a deep copy for later versions of Python tries to evaluate the interpolation expressions, instead of just copying them over as is. Naturally, there is no information for the value of the interpolation variable like `levelname` --- this is just a deep copy --- so Python raises an error.

We had an earlier fix for this, but it had the flaw of not working reliably in a multi-threaded environment. The latest fix is a custom deep copy for `ConfigObj` objects, instead of relying on Python's `copy.deepcopy()` function. 

Hopefully, that's the last we'll see of this issue!