weewx/weewx #520: APRS/CWOP can't connect when only one server listed

MoTLD, 2020-04-01:
As a licensed ham, I tried using "server_list = noam.aprs2.net:14580" in the CWOP section of weewx.conf, with my call sign and passcode of course, and weewx wouldn't connect. Checking the logs, I found the following:

Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 'n'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 'o'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 'a'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 'm'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: '.'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 'a'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 'p'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 'r'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 's'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: '2'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: '.'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 'n'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 'e'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: 't'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: ':'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: '1'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: '4'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: '5'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: '8'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Bad server address: '0'; ignored
Apr  1 01:22:48 monstrosity weewx[7258]: restx: CWOP: Failed to publish record 2020-04-01 01:22:00 EDT (1585718520): Tried 20 servers 3 times each

Adding extra servers to server_list fixed the problem; apparently weewx can't handle a "comma delimited" list of just one item. Not exactly a high priority bug, but it had me perplexed for a bit so I figured I'd mention it here. And no, this is not an April Fools joke. ;)

Weewx is awesome, BTW, and I love it! So does my nephew, a youthful new ham who got me wanting to set up a weather station of my own after I helped him with the one he got for his birthday, and with the help of weewx-sdr and a few tweaks here and there I got mine working pretty quickly with an existing thermometer/hygrometer my folks already had outside. Now to shop for my own...

Thanks for the great software, helpful documentation, and all the hard work that went into it!

gjr80, 2020-04-01:
The problem will be due to the way configobj parses a single entry. A single entry is returned as a string and a comma delimited entry is returned as a list. As a workaround a single server should be able to be used by placing a comma immediately after the server entry, eg:
```
        server_list = noam.aprs2.net:14580,
```

MoTLD, 2020-04-01:
Good to know, thanks!

gjr80, 2020-04-01:
But we can and will fix it.....
Not everybody knows about the trailing comma

MoTLD, 2020-04-01:
That works too... :)

Thanks again!

closed 2020-04-03 by tkeffer