weewx/weewx #522: Missing support for wind-cup option 'other'

krbaker, 2020-04-03:
The new Ultrasonic Anemometer requires the use of the cup-size 'other' but wee-wx can't configure that setting.

tkeffer, 2020-04-03:
Apologies, but the Airmar stations are not supported by WeeWX. You will have to talk to the author of the driver.

krbaker, 2020-04-03:
Oh wow, sorry I was not being descriptive when I wrote this!  Davis released the 6415 Ultrasonic Anemometer (its really a rebadged LCJ).  The only missing piece to support it in weewx is setting the 'other' cup mode on the Davis VantagePro2 (vs. small and large). 

[Davis 6415](https://www.davisinstruments.com/product/sonic-anemometer/)
[Manual](https://www.rainwise.com/products/attachments/9005/20161202072626.pdf)

> 4. Configuration of the DAVIS VP2
> The anemometer cups must be set to "OTHER" in the Setup Menu.

tkeffer, 2020-04-03:
OK, that helps.

Looking through the Davis API, it seems that there are two EEPROM bytes for wind cup size.

Byte `0x2B`: This byte is writeable, and the one we presently use. It offers only two choices: "small" and "large". 

Byte `0xC3`: This one offers small, large, and "other (no speed correction applied by console)". I don't know whether it is writeable, or simply a reflection of what was chosen on the console display. I also assume that the software would have to provide some sort of calibration curve to convert the raw data into something with physical units. Any idea where that would come from?

tkeffer, 2020-04-16:
@krbaker Any thoughts?

tkeffer, 2020-04-24:
No response. Closing issue. We can always reopen later.

closed 2020-04-24 by tkeffer