weewx/weewx #524: Add TLSv1.2 support to mqtt plugin

arnebjarne, 2020-04-12:
Hi.

I am running WeewX 3.9.0 (on python 2.7) and mosquitto 1.6.9.
The default TLS implementation in weewx-mqtt is too old and will not connect to mosquitto.

This patch enabled TLSv1.2 which mosquitto supports:
--- cut ---
--- mqtt.py     2019-04-02 18:35:53.000000000 +0200
+++ /usr/share/weewx/user/mqtt.py       2020-04-12 15:41:14.409520543 +0200
@@ -289,6 +289,7 @@
             pass
         self.TLS_VER_OPTIONS['sslv23'] = ssl.PROTOCOL_SSLv23
         self.TLS_VER_OPTIONS['tlsv1'] = ssl.PROTOCOL_TLSv1
+        self.TLS_VER_OPTIONS['tlsv12'] = ssl.PROTOCOL_TLSv1_2
         try:
             self.TLS_VER_OPTIONS['tls'] = ssl.PROTOCOL_TLS
         except AttributeError:
--- cut ---
You can then specify in weewx.conf:
--- cut ---
    [[MQTT]]
        server_url = mqtt://weewx:REMOVED@mqttbroker.local:1883/
        topic = weather/weewx
        unit_system = METRIC
        binding = archive, loop
        aggregation = aggregate
        [[[tls]]]
            tls_version = tlsv12
            ca_certs = /etc/pki/tls/certs/ca-bundle.crt

--- cut ---

tkeffer, 2020-04-12:
Hello,

My apologies, but the MQTT plugin is not part of WeeWX.  I would suggest contacting the author.

closed 2020-04-12 by tkeffer

arnebjarne, 2020-04-12:
> 
> 
> Hello,
> 
> My apologies, but the MQTT plugin is not part of WeeWX. I would suggest contacting the author.

Hi Tom.
Check, I'll try to find the contact information on the developer :)

matthewwall, 2020-04-12:
fixed at commit 91b55d2 in the weewx-mqtt repository

https://github.com/matthewwall/weewx-mqtt

arnebjarne, 2020-04-12:
> 
> 
> fixed at commit 91b55d2 in the weewx-mqtt repository
> 
> https://github.com/matthewwall/weewx-mqtt

Thanks Matthew for the quick reply and fix (remember to update the wiki page with new options ;) )