weewx/weewx #525: occasional abnormal rain readings from acurite

zerog2k, 2020-04-22:
[image]

It did not actually rain 50+ inches in one hour here (was sunny and dry), and also the Acurite Console did not show any rainfall.

weewx log (relevant parts) with debugging on:

```
Apr 21 11:02:26 spire weewx[1650]: acurite: R1: 01 8e 01 71 00 7b 10 12 03 ff
Apr 21 11:03:02 spire weewx[1650]: acurite: R1: 01 8e 01 71 00 78 7b 46 03 ff
Apr 21 11:03:38 spire weewx[1650]: acurite: R1: 01 8e 01 71 00 6b 10 12 03 ff
Apr 21 11:03:38 spire weewx[1650]: acurite: rain counter decrement ignored: new: 52.4764 old: 193.5988
Apr 21 11:03:38 spire weewx[1650]: wxformulas: rain counter reset detected: new=52.4764 old=193.5988
Apr 21 11:04:14 spire weewx[1650]: acurite: R1: 01 8e 01 71 00 6b 10 12 03 ff
Apr 21 11:04:50 spire weewx[1650]: acurite: R1: 01 8e 01 71 00 6d 10 12 03 ff
```

zerog2k, 2020-04-22:
https://github.com/weewx/weewx/blob/master/bin/weewx/drivers/acurite.py#L510-L514

```
    def _augment_packet(self, packet):
        # calculate the rain delta from the total
        if 'rain_total' in packet:
            total = packet['rain_total']
            if (total is not None and self.last_rain is not None and
                total < self.last_rain):
                log.info("rain counter decrement ignored:"
                         " new: %s old: %s" % (total, self.last_rain))
            packet['rain'] = weewx.wxformulas.calculate_rain(total, self.last_rain)
            self.last_rain = total
```

seems here like `calculate_rain` should _not_ be called (ignored) if rain counter is decremented, possibly just indentation typo? otherwise rain counter decrement is not actually getting ignored - only logging this
?

thunderdome2, 2020-04-22:
Hello.
I don't know if anyone else has this issue with a FineOffset weather station. I seen this issue randomly for years, but until now I've been thinking that issue was from a defect rain gauge.

zerog2k, 2020-04-22:
fyi @tkeffer 
so the underlying issue is that sometimes (for me maybe once every other month, etc) I will see an unreal spike of rain on weewx (which of course the console does not register).
Here is an old google groups thread describing the issue:
https://groups.google.com/forum/#!topic/weewx-user/qCTkcG_8Nbs

I have caught it in the act, as we can see the raw packets data, which are clearly wrong. Rain is supposed to be 7th byte:
1. R1: 01 8e 01 71 00 7b 10 12 03 ff (rain field is 0x12)
2. R1: 01 8e 01 71 00 78 7b 46 03 ff (rain field is now 0x46)
3. R1: 01 8e 01 71 00 6b 10 12 03 ff (rain field is again 0x12)

So here the console reported (incorrectly) that the rain incremented by 0.52 in (46h-12h = 52 decimal * 0.01 in)

Considering readings are (usually, and in this case) 36 seconds apart, it seems unlikely that we received 0.5in of rain in 36 seconds - so maybe one possible workaround is to reject rainfall rates exceeding some threshold within some (small) interval. (It might be possible to receive such a large rainfall rate between two readings if, for example, there are missed readings due to various legitimate reasons.)

So in summary, I'm not sure what the actual cause of this issue is, but it's clear that it happens with some (in)frequency, to multiple users. However, what is not obvious is how to detect and/or mitigate, as it always seems transient.

So looking for feedback and ideas here on how to handle, such that our rain data is not "polluted".

tkeffer, 2020-04-22:
Wow. That sucks. But, I didn't quite follow your comments. You said "... of course the console does not register [the bad data]", but then later said, "...the console reported (incorrectly) that the rain incremented by 0.52 in".

Did you mean the display in the console for the former, but what was emitted down the wire for the latter? Or, something else?

Is there a checksum available?

zerog2k, 2020-04-22:
Sorry - I meant that in all cases (I have seen, and what I have read of reports) the console display itself does not show the (abnormal) rainfall difference - but the console is sending the data over usb to weewx where this data stream apparently has this issue. Interestingly, the Acurite RF protocol has both a crc/checksum byte at the end, individual parity bits (0x80 mask) on some? bytes/words in the message, and sends the messages three times, but what is sent by console over usb does not include this.  (I suspect that it may be some bug in the way the console decodes/verifies/forwards the data, but may never be able to know. I don't run the Acurite PC software, so not sure if this has "fixes" for this occasional issue or not.)

So I have a few thoughts on possible workarounds:
1. check parity bits (?), and/or
2. reject extreme rainfall amounts within some very small time window

tkeffer, 2020-05-05:
Looking through the acurite driver, the only messages with a checksum are the R3 history records. The R1 records, which contain rain, do not include any checks that we know of.

So, garbage in, garbage out. Not sure what we can do here.

Try setting a value for [`[StdQC]`](http://www.weewx.com/docs/usersguide.htm#StdQC). This can at least filter out the largest `rain` values.

But clearly, the problem is in the hardware, not the driver.

closed 2020-05-05 by tkeffer

edi-x, 2020-06-09:
> 
> 
> Hello.
> I don't know if anyone else has this issue with a FineOffset weather station. I seen this issue randomly for years, but until now I've been thinking that issue was from a defect rain gauge.

@thunderdome2 , I had the same issue with my FineOffset station and I have made some changes in the driver (see https://github.com/weewx/weewx/pull/565/) to avoid these effects. The driver was already set up to handle such spurious readings but as the problem is in the hardware it's a little bit tricky to detect an compensate such problems in a suitable way for all or at least most circumstances.