weewx/weewx #532: weewx v4 import rain data issue

MksRasp, 2020-05-03:
imported weewx 3.9.2 data base (CSV file) to new install of weewx v4 all data are the same across both version except rain data which is way lower than what v3.9.2 display.
I think the import calculation is not right!

gjr80, 2020-05-03:
A few questions:

1. Can you also explain exactly what you mean by ‘ I think the import calculation is not right!’? 

1. Can you explain exactly what steps you did both for the export and import? 

1. Can you provide the data that did not import correctly and a copy of the import config file? 

1. Were there any errors/warnings/relevant messages on the console or in the log during/after the import?

Gary

MksRasp, 2020-05-03:
1-Iam runing weewx 3.9.2 on 3 device with different sensors but all share the output of acurite 5in1 weather station
sometimes when one device is off for longtime I import the missing data from the other device and never had problem, I liked the idea of upgrading to weewx v4 and it the upgrade went well then I tried to do the usual data import just to make sure it will work before going ahead and upgrade the other devices.
all data like temerture , humidity,pressure and so on are all identical on all station except rain data and wind data on weewx v4 device the data number is low instead of 62.2mm it shows 37.8mm.

2-I do copy the database to pc then I use "DB Browser for SQLite" to export it to CSV format then I move this file to the target station and use the import utility while weewx is stopped and never failed even with v4 but rain data and wind data somehow shows different numbers that other stations and my manual records from acurite display (2.5 yesrs of data).( I live in the desert so littel rain we have per year so it was easy to spot).

3-only warninig that the interval is different but I accept and yes to contenue the operation but no major error at all and the import goes through the steps and finaly display sucessfully.
[station 2.txt](https://github.com/weewx/weewx/files/4571920/station.2.txt)
[station 3.txt](https://github.com/weewx/weewx/files/4571921/station.3.txt)
[missing.conf.txt](https://github.com/weewx/weewx/files/4571925/missing.conf.txt)

gjr80, 2020-05-04:
My first thought regards the rain data is your import config file is saying that the source rain data is cumulative whereas if you are doing a simple export from a WeeWX archive your rain data will be (what I term) discrete, ie the rain data is the the total seen by the station during that archive interval not a cumulative value. That would certainly result in lower rainfall data appearing in the imported data. I suggest you try using the following setting in your import config file:
```
rain = discrete
```
Regards your wind data, given that the daily times for highest wind speed are identical but the magnitudes are different on both NOAA format reports I am wondering if there is a units mismatch going on somewhere. The import config file has no equivalent wind speed setting as there is for rain, the only relevant setting is the field name and units in the field map. Though as this is an export from a WeeWX archive (I expect) there will also be a field `usUnits` in the import source data and that could be having some sort of adverse effect. Would help if you could post the import source (exported) csv data, impossible to say much more without seeing that data.

MksRasp, 2020-05-04:
Thank you gjr80
using rain = discrete solved the rain problem but the wind data still shows different data but it is not important for me. for this reason I will close this issue

closed 2020-05-04 by MksRasp