weewx/weewx #533: weewx resists death

DavidCWGA, 2020-05-03:
If you "kill -2" the weewx process, a new one immediately spawns in its place.

If you "systemctl stop weewx", systemctl hangs.

If you attempt to stop an lxc container with weewx running inside, lxc gets very upset and hangs, or sometimes barfs a timeout error.

Only "kill -9"ing the process is a reliable way of terminating it.

tkeffer, 2020-05-03:
1. Not sure why you'd want to do this. Normally, you send `SIGTERM`.
2. `systemctl stop weewx` works properly for me. Can you show us a log?
3. Don't know anything about lxc.

DavidCWGA, 2020-05-04:
The only thing written to the log when you do systemctl stop is:
"May  3 16:57:35 weather weewx[952] INFO __main__: Received signal TERM (15)."

But weewx doesn't exit. systemctl exits after exactly 2 minutes, which I guess is the timeout, and weewx is still running afterwards, as shown by ps:

  952 ?        S      1:19 python3 /usr/share/weewx/weewxd --daemon --pidfile=/var/run/weewx.pid /etc/weewx/weewx.conf

tkeffer, 2020-05-04:
I just tried `systemctl stop weewx` and this is what I got:

```
May  3 17:23:34 rpi2 systemd[1]: Stopping LSB: weewx weather system...
May  3 17:23:34 rpi2 weewx[1326] INFO __main__: Received signal TERM (15).
May  3 17:23:34 rpi2 weewx[1326] INFO weewx.engine: Main loop exiting. Shutting engine down.
May  3 17:23:34 rpi2 weewx[1326] INFO __main__: Terminating weewx version 4.0.0
May  3 17:23:34 rpi2 weewx[1326] INFO __main__:     ****  Traceback (most recent call last):
May  3 17:23:34 rpi2 weewx[1326] INFO __main__:     ****    File "/usr/share/weewx/weewxd", line 154, in main
May  3 17:23:34 rpi2 weewx[1326] INFO __main__:     ****      engine.run()
May  3 17:23:34 rpi2 weewx[1326] INFO __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 188, in run
May  3 17:23:34 rpi2 weewx[1326] INFO __main__:     ****      for packet in self.console.genLoopPackets():
May  3 17:23:34 rpi2 weewx[1326] INFO __main__:     ****    File "/usr/share/weewx/weewx/drivers/simulator.py", line 148, in genLoopPackets
May  3 17:23:34 rpi2 weewx[1326] INFO __main__:     ****      time.sleep(sleep_time)
May  3 17:23:34 rpi2 weewx[1326] INFO __main__:     ****    File "/usr/share/weewx/weewxd", line 257, in sigTERMhandler
May  3 17:23:34 rpi2 weewx[1326] INFO __main__:     ****      raise Terminate
May  3 17:23:34 rpi2 weewx[1326] INFO __main__:     ****  Terminate
May  3 17:23:34 rpi2 weewx[1438]: Stopping weewx weather system: weewx.
May  3 17:23:34 rpi2 systemd[1]: weewx.service: Succeeded.
May  3 17:23:34 rpi2 systemd[1]: Stopped LSB: weewx weather system.
```

This is under Buster.

One thing to note: if you try to terminate while the reporting thread is running, the weewx engine will attempt to stop it, but it may take some time. The attempt will eventually time out, resulting in program termination.

tkeffer, 2020-05-05:
David?

DavidCWGA, 2020-05-05:
I'm still trying to figure out what's going on, but it's not a huge problem because I can just modify the init script to do "kill -9" for now.

Incidentally it really should exit when sent SIGINT, if only because that's the standard, and everything else does.

tkeffer, 2020-05-05:
Not everything. I just tried `kill -2` on the mysqld daemon on my machine and it didn't do anything. 

Still, it's trivial to test. In the file `weewx/weewxd.py` change this

```python
    signal.signal(signal.SIGHUP, sigHUPhandler)
    signal.signal(signal.SIGTERM, sigTERMhandler)
```

to this

```python
    signal.signal(signal.SIGHUP, sigHUPhandler)
    signal.signal(signal.SIGTERM, sigTERMhandler)
    signal.signal(signal.SIGINT, sigTERMhandler)
```

and see if that works for you. If it does, I have no problem with adding the patch to the code base.

tkeffer, 2020-05-10:
David?

DavidCWGA, 2020-05-12:
Well making that change doesn't stop systemctl from hanging, but I guess that shouldn't be a surprise, I don't think systemctl sends SIGINT.

I still can't figure out why hup and term aren't killing it.

tkeffer, 2020-05-13:
weewxd responds to a `HUP` signal by rereading its configuration file.

Can't help you on `TERM`.

Closing issue.

closed 2020-05-13 by tkeffer