weewx/weewx #534: Unable to upgrade from 3.9.2 to 4.0.0 on Debian 9/Stretch using package repo

kobuki, 2020-05-04:
Package repo per docs is installed, installation fails with the following message:

```
python3: can't open file '/home/weewx/bin/wee_config': [Errno 2] No such file or directory
dpkg: error processing package weewx (--configure):
 subprocess installed post-installation script returned error exit status 2
Processing triggers for systemd (232-25+deb9u9) ...
Errors were encountered while processing:
 weewx
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
There is no weewx user and home dir.

- - -

On another note, installation of prereqs also fails in various ways for the setup.py method on D9 (docimented requirement package doesn't exist, cheetah cannot be installed, etc), but that's a different issue. If needed, I can document that too.

tkeffer, 2020-05-04:
Can you give us a few more details?

I assume you had previously used a package install. Then you followed [these instructions](http://www.weewx.com/docs/debian.htm) for Debian 9?

Did this happen after

    sudo apt-get install weewx

Don't quite understand why it's trying to use Python 3 under Debian 9. Do you have anything in `/etc/default/weewx`?

Re: `setup.py`. Did you follow the Debian 9 instructions [here](http://www.weewx.com/docs/setup.htm)?

kobuki, 2020-05-04:
> Can you give us a few more details?

Sure. 

> I assume you had previously used a package install.

Yes, used `dpkg -i package.deb` for older versions.

> Then you followed [these instructions](http://www.weewx.com/docs/debian.htm) for Debian 9?

Yes, but I simply did an `apt-get install package-4.0.0.deb` as root.

> Don't quite understand why it's trying to use Python 3 under Debian 9. Do you have anything in `/etc/default/weewx`?

```
# cat /etc/default/weewx
export LANG=xx_XX.UTF-8
```
I also suspect that when the installer detects Python 3, it tries to do something using this version.

> Re: `setup.py`. Did you follow the Debian 9 instructions [here](http://www.weewx.com/docs/setup.htm)?

Yes, I did everything per docs (or tried to). There is no package named `python3-distutils` in D9.

Also:
```
# python3 -V
Python 3.5.3
# pip3 install cheetah
Collecting cheetah
  Using cached https://files.pythonhosted.org/packages/cd/b0/c2d700252fc251e91c08639ff41a8a5203b627f4e0a2ae18a6b662ab32ea/Cheetah-2.4.4.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-_5obdaga/cheetah/setup.py", line 10, in <module>
        import SetupTools
      File "/tmp/pip-build-_5obdaga/cheetah/SetupTools.py", line 50
        except DistutilsPlatformError, x:
                                     ^
    SyntaxError: invalid syntax

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-_5obdaga/cheetah/
```

matthewwall, 2020-05-04:
you clearly have python3 installed on your system, and i'm guessing that you also have python2 since you were running weewx 3.9.2.  is that correct?

the installer does not put LANG into /etc/default/weewx - you must have done that.  but why is /etc/default/weewx NOT the file installed by the installer?  was there some warning/failure output when you did the 'apt-get install package-4.0.0.deb'?

without the proper variables in /etc/default/weewx, the weewx scripts in /usr/bin/wee* will try to invoke the actual entrypoints in /usr/share/weewx with whatever python interpreter is default for the system, which in your case seems to be python3

kobuki, 2020-05-04:
> you clearly have python3 installed on your system, and i'm guessing that you also have python2 since you were running weewx 3.9.2. is that correct?

Yes.

> the installer does not put LANG into /etc/default/weewx - you must have done that.

Probably, yes, 4-5 years ago. I've long forgotten about it.

> but why is /etc/default/weewx NOT the file installed by the installer? was there some warning/failure output when you did the 'apt-get install package-4.0.0.deb'?

You're right, but there are so many questions about overwriting files (one for each localized skin file) that I've skipped over this by mistake.

> without the proper variables in /etc/default/weewx, the weewx scripts in /usr/bin/wee* will try to invoke the actual entrypoints in /usr/share/weewx with whatever python interpreter is default for the system, which in your case seems to be python3

Hm, you're half right here. `/etc/default/weewx.dpkg-dist` installed by the v4 installer has this content, but python default is v2.
```
# python -V
Python 2.7.13
# cat /etc/default/weewx.dpkg-dist
WEEWX_PYTHON=python2
WEEWX_BINDIR=/usr/share/weewx
WEEWX_BIN=/usr/bin/weewxd
WEEWX_CFG=/etc/weewx/weewx.conf
```
> there is no 'cheetah' for python3 - it is called 'cheetah3'. so you should do 'pip3 install cheetah3'

That's another mistake by me. I'll try again and report back.

matthewwall, 2020-05-04:
>>    without the proper variables in /etc/default/weewx, the weewx scripts in /usr/bin/wee* will try to invoke the actual entrypoints in /usr/share/weewx with whatever python interpreter is default for the system, which in your case seems to be python3

> Hm, you're half right here. /etc/default/weewx.dpkg-dist installed by the v4 installer has this content, but python default is v2.

the shebang in each /usr/share/weewx/wee_* file is this:
```
#!/usr/bin/env python
```
what happens when you type that in, i.e.,
```
/usr/bin/env python -V
```

kobuki, 2020-05-04:
> what happens when you type that in, i.e.,

```
# /usr/bin/env python -V
Python 2.7.13
```

tkeffer, 2020-05-12:
Any progress on this?

kobuki, 2020-05-12:
Sorry, I haven't been able to do the upgrade the system where the errors occurred since then, I left it on 3.x. I'll do the upgrade in the coming days so I'll be able to report back soon.

tkeffer, 2020-05-24:
Were you able to do the upgrade?

kobuki, 2020-05-24:
I decided to upgrade that particular OS to Debian Buster/10. It also upgraded WeeWx to 4.0 and it went without problems. I haven't tried to run it with Python 3 yet. It's running with Python 2 in the Buster VM now with logging issues I mentioned in the Google user group (some of the older drivers/services aren't using the new logging facility). I'm in the process of updating and testing 3 separate drivers/services for Python 3/WeeWx 4 so I haven't looked at it since then. I'll publish the code of the public ones after testing. A 4th one might have a fix for WeeWx 4 already (MesoWx), if not, I'll have to update that one too since I'm using it since almost the beginning.

However to end this cleanly and if you think it would be useful to see how WeeWx 4 behaves under Debian 9+P3.6, I can create a clean VM and attempt the upgrade processes there to see if there are any remaining upgrade issues and report back.

tkeffer, 2020-05-24:
WeeWX V4 runs run under Debian 9 and Python 3 without problems. It's just that the install cannot be automated with a package installer due to missing dependencies. So, the user has to resort to using `pip`, as you did. We chose to not make it the default.

I'm not sure what we learned from this. That the presence of an unexpected `/etc/default/weewx` can cause the installer to fail?

kobuki, 2020-05-24:
If that's the case, it should be removed/renamed during install. But please don't close this ticket yet, I want to do a full install on a system without previous installation of weewx, to make sure.

kobuki, 2020-05-24:
OK, I've installed WeeWx 4.0 on a Debian 9 system, using the setup.py method and Python 3. The takeaway:

- no python3-distutils on Debian 9, installation step fails
- /etc/default/weewx should be renamed/removed by the installer or manually when using setup.py (possibly in other cases, too) - systemd startup method is likely unaffected
- setup.py method should recommend creating a systemd unit instead of sysvinit script where appropriate (basically everywhere)

Other than this, install went through and the daemon is running without issues. Using systemd unit and using simulator. You can close this if you want, addressing the points above could be considered.

matthewwall, 2020-05-24:
> no python3-distutils on Debian 9, installation step fails

i'm pretty sure that doing `apt install python3-pip` will also install `python3-distutils`.  did you do `apt install python3-pip`?

>  /etc/default/weewx should be renamed/removed by the installer or manually when using setup.py (possibly in other cases, too) - systemd startup method is likely unaffected

a setup.py installation does not use `/etc/default/weewx`.  the file `/etc/default/weewx` is used by `/etc/init.d/weewx` (and thus systemd), but that happens only on systems where weewx4 was installed from rpm or deb package.

>    setup.py method should recommend creating a systemd unit instead of sysvinit script where appropriate (basically everywhere)

systemd is not used everywhere.  however, we probably should refactor the 'run' section of the setup.py instructions so that there is one recipe for systemd, another for various init.d, another for bsd (and another for macos?)

kobuki, 2020-05-24:
> i'm pretty sure that doing `apt install python3-pip` will also install `python3-distutils`. did you do `apt install python3-pip`?
```
root@weewx-test:~# apt-cache search python3-distutils
python3-distutils-extra - enhancements to the Python3 build system
```
Maybe you need that on D9? Without the python3-pip package, it's not possible to use pip3; installing it can only pull existing dependencies but python3-distutils doesn't exist on D9. I'm unsure what to install in this case using pip3, either. Or if this package is needed at all on D9.

> a setup.py installation does not use `/etc/default/weewx`. the file `/etc/default/weewx` is used by `/etc/init.d/weewx` (and thus systemd), but that happens only on systems where weewx4 was installed from rpm or deb package.

Just to make a note, systemd only uses sysvinit scripts when they exist (and no native unit exists) and when they do, it generates a systemd unit for them that calls the sysvinit script. Only then is relevant what's inside a sysvinit script in presence of systemd. Native systemd units don't care, in this case they directly start `/home/weewx/bin/weewxd`.

> systemd is not used everywhere. however, we probably should refactor the 'run' section of the setup.py instructions so that there is one recipe for systemd, another for various init.d, another for bsd (and another for macos?)

Yes, everywhere is an exaggeration. But it's now available for all main Linux distros. I suggest where it's available it should be recommended or used as a native service by default.

kobuki, 2020-06-26:
@tkeffer I see that you've closed this, but I'm unable to assess the resolution. Was or are there any points addressing the package dependency issue or my suggestion of making systemd units default on supported systems? For those points, would you like me to open new issues?

tkeffer, 2020-06-26:
Maybe I'm misunderstanding things, but I closed the issue because, as I understand it, there is no fault.

Under Debian 9, the recommended procedure is to use Python 2, not Python 3. So `pip3` is not involved.

Making systemd the default is a recommendation, but the present system also works fine and, as @matthewwall noted, has some advantages.

Or, am I missing something?

kobuki, 2020-06-26:
> Under Debian 9, the recommended procedure is to use Python 2, not Python 3. So `pip3` is not involved.

Please see http://weewx.com/docs/setup.htm - Debian 9 tab. They're unmistakably Python 3 instructions, using P3 packages and in one case, pip3. Maybe these instructions can be removed completely from there. Issues were mentioned regarding this at the beginning of this issue thread. But, most likely, hardly anyone uses these instead of installing from packages or a repo.

> Making systemd the default is a recommendation, but the present system also works fine and, as @matthewwall noted, has some advantages.

It was just a suggestion, I don't mind either, since both works without problems. OTOH, I'm not aware of any advantages of using init.d scripts with systemd. As noted, systemd generates a unit for managing the init.d script, and we have little to no control over that. It's better to use a systemd unit directly, unless the startup is something really complicated, but in case of WeeWx, it's not the case.

tkeffer, 2020-06-27:
I think I see what's going on.

I just installed a brand-new Debian 9.12 system on a VM, using a net install. It came with a Python 3 version of `distutils` already installed.

The original test install was done on a version of Raspbian squeeze (Debian 9), which did not have a Python 3 version of `distutils` preinstalled. However, it could be installed using `python3-distutils`.

What distribution of Debian 9 did you use?

kobuki, 2020-06-27:
It was a Debian 9 amd64 in an LCX container. Sorry, can't remember the exact dot version, but it shouldn't make a difference. I've upgraded that CT to Buster since then.

FWIW, there are 2 packages of interest:

```
python3.7-distutils
python3.8-distutils
```

No `python-distutils`. Maybe that's the only thing to be fixed (the install docs).

tkeffer, 2020-06-27:
This may be as simple as changing the directions to

```shell
python3 -c "import distutils" || sudo apt install python3-distutils
```

kobuki, 2020-06-28:
Should work, I'd suggest a simple addition:

```
python3 -c "import distutils" 2>/dev/null || sudo apt install python3-distutils
```

tkeffer, 2020-06-29:
Fixed in commit https://github.com/weewx/weewx/commit/a3dca5aa6ca8fad928e7b37adaf56a279453f443.

closed 2020-06-29 by tkeffer