weewx/weewx #537: need to eliminate a version tag in extensions

matthewwall, 2020-05-07:
one pattern in the extension installer that should be addressed is the version tag.

right now it happens in 3 places:

- changelog
- install.py
- bin/user/xxx.py (the extension itself)

we risk skew between install.py and the extension itself.

i think we should reduce that to two:

- changelog
- bin/user/xxx.py

this should be fairly straightforward for drivers and services - make sure wee_extension uses the extension code, not install.py.  that might complicate installer housekeeping a bit, but better to trust the source - the extension itself - than something that talks about it - the install.py file.

but we should also consider how to use a version number within skins too.

tkeffer, 2020-05-07:
I agree. 

Alternatively, use tools that automatically increment version numbers.

tkeffer, 2020-10-30:
As far as I can tell, the extension installer does not use the version info provided by `install.py`. 

No reason to include it at all.

tkeffer, 2020-10-30:
Disregard previous message. Listing the installed extensions uses the extension version contained in `install.py`.

tkeffer, 2020-11-12:
I started playing around with this and have concluded that we can only make matters worse.

As you noted, we have no protocol for specifying skin versions.

For code, many extensions have more than one file in `bin/user`. Which one do we choose?

closed 2020-11-12 by tkeffer