weewx/weewx #539: WOW publish rate

jenolan, 2020-05-08:
I have my weather station archive rate set at 60 seconds, the Met Office (WOW) only allows publishing every 5 minutes at most. The code in 'rest.py' is set to publish a record for every archive period, so a bunch of requests 4/5 are being rejected 'WOW: Failed to publish record 2020-05-08 10:38:00 AEST (1588898280): Failed upload after 3 tries'.

I added report_timing = */5 * * * * to the [StdRESTful] -> [[WOW]] but obviously that has no effect. Will have a look at doing some code change to handle this but my suggestion would be that report_timing be available to the restful code.

jenolan, 2020-05-08:
My hack to do five minute sends line 784 of rest.py;
`    def new_archive_record(self, event):
        _ts = event.record['dateTime']
        if _ts % 300 == 0:
            self.archive_queue.put(event.record)
`

tkeffer, 2020-05-08:
It's poorly documented, but option `post_interval` already does this.

```ini
  [[WOW]]
    post_interval = 300
```

It really should be included in the documentation

closed 2020-05-08 by tkeffer

jenolan, 2020-05-09:
ug ... I checked the WOW code but that is obviously hidden in the event handler will change my stuff.

Thanks!!
Larry