weewx/weewx #542: Weewx TE923

ClunyABZ, 2020-05-09:
I was originally all setup and running fine on Wunderground with everything reporting as it should. I've recently got round to moving over to WeatherCloud but there's data missing, and not being uploaded to them. I'm a little confused as to where the problem lies. My Irox Pro2 Weatherstation, my Raspberry Pi setup, Weewx Software, or Weathercloud. I've had a post on Google Groups where I ran some debugging logs which appeared to say that the wrong information was being uploaded to WeatherCloud - https://groups.google.com/forum/m/#!topic/weewx-user/bsZ9rsNWqIc
My WeatherCloud page is - https://app.weathercloud.net/d9110687615

I'm wondering if anyone can help me figure out whats wrong, and what needs amended to solve my issues?
Thanks

tkeffer, 2020-05-10:
Please follow Gary's suggestion from his last post:

> You might want to run WeeWX directly so the loop packets (lines starting with LOOP:) and archive records (line starting with REC:) seen by WeeWX are displayed on the console. Look closely at the REC: archive records, do outTemp, outHumidity, dewpoint and heatindex exist and do they have valid data? In every archive record?

We need to first figure out whether data is arriving from your hardware.

To run WeeWX directly, see the section [*Running directly*](http://www.weewx.com/docs/usersguide.htm#Running_directly) in the User's Guide.

ClunyABZ, 2020-05-10:
Hi, Yes I think I can see where the problem lies but not how to resolve it. My external temperature is reported from t_3, but weewx is using t_1 to report the temperature to Weathercloud. The TE923 advises that you can have up to 5 temperature sensors reporting, but not how to stipulate which one to use as outdoor temperature.

tkeffer, 2020-05-10:
You should use a sensor map. In `weewx.conf`, you'll want something like this:

```ini
[TE923]

  # (as before)

  [[sensor_map]]
    outTemp = t_3
```
You may have to do the same thing for `outHumidity`.

In any case, this does not look like a bug in the driver, so I'm closing the issue.

closed 2020-05-10 by tkeffer