weewx/weewx #543: TypeError with weewx.drivers.ws28xx buster/weewx4

knackko, 2020-05-09:
After updating my raspbian from stretch to buster, the following errors occurs when connecting my WS2812:

```
May 09 22:59:24 knadock python3[10163]: weewx[10163] DEBUG weewx.drivers.ws28xx: handleNextAction: a3 (set time data)
May 09 22:59:24 knadock python3[10163]: weewx[10163] DEBUG weewx.drivers.ws28xx: buildTimeFrame: cs=04de
May 09 22:59:24 knadock python3[10163]: weewx[10163] DEBUG weewx.drivers.ws28xx: setResetMinMaxFlags: 0
May 09 22:59:24 knadock python3[10163]: weewx[10163] DEBUG weewx.drivers.ws28xx: SetTime/SetConfig data written
May 09 22:59:24 knadock python3[10163]: weewx[10163] DEBUG weewx.drivers.ws28xx: handleCurrentData: inBufCS of station does not match
May 09 22:59:24 knadock python3[10163]: weewx[10163] DEBUG weewx.drivers.ws28xx: handleConfig: sleep=0.42 first=0.3 next=0.01 count=13
May 09 22:59:25 knadock python3[10163]: weewx[10163] DEBUG weewx.drivers.ws28xx: handleHistoryData: time=2020-05-03 17:55:00 this=1714 (0x7a24) latest=1712 (0x7a00) nrec=1795
May 09 22:59:25 knadock python3[10163]: weewx[10163] INFO weewx.drivers.ws28xx: handleHistoryData: request records since 2020-05-08 11:50:00 CEST (1588931400)
May 09 22:59:25 knadock python3[10163]: weewx[10163] DEBUG weewx.drivers.ws28xx: handleHistoryData: start_index=1286.0 num_outstanding_records=426
May 09 22:59:25 knadock python3[10163]: weewx[10163] DEBUG weewx.drivers.ws28xx: handleHistoryData: next=1286.0
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx: exception in doRF: unsupported operand type(s) for >>: 'float' and 'int'
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx: Traceback (most recent call last):
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx:   File "/usr/share/weewx/weewx/drivers/ws28xx.py", line 3961, in doRF
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx:     self.doRFCommunication()
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx:   File "/usr/share/weewx/weewx/drivers/ws28xx.py", line 4005, in doRFCommunication
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx:     self.generateResponse(FrameBuffer, DataLength)
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx:   File "/usr/share/weewx/weewx/drivers/ws28xx.py", line 3748, in generateResponse
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx:     self.handleHistoryData(newBuffer, newLength)
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx:   File "/usr/share/weewx/weewx/drivers/ws28xx.py", line 3649, in handleHistoryData
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx:     newlen[0] = self.buildACKFrame(newbuf, EAction.aGetHistory, cs, nextIndex)
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx:   File "/usr/share/weewx/weewx/drivers/ws28xx.py", line 3460, in buildACKFrame
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx:     newBuffer[0][6] = (haddr >> 16) & 0x0F | 16 * (comInt & 0xF)
May 09 22:59:25 knadock python3[10163]: weewx[10163] ERROR weewx.drivers.ws28xx: TypeError: unsupported operand type(s) for >>: 'float' and 'int'
May 09 22:59:25 knadock python3[10163]: weewx[10163] DEBUG weewx.drivers.ws28xx: stopping rf communication
```

Here are infos from the system:

- weewx version
`
weewx/buster,now 4.0.0-1 all [installed]`

- python3 version

```
Python 3.7.3 (default, Dec 20 2019, 18:57:59) 
[GCC 8.3.0] on linux
```

- weewx config:
```
station_type = WS28xx

...
[WS28xx]
    # This section is for the La Crosse WS-2800 series of weather stations.
    # Radio frequency to use between USB transceiver and console: US or EU
    # US uses 915 MHz, EU uses 868.3 MHz.  Default is US.
    transceiver_frequency = EU
    # The station model, e.g., 'LaCrosse C86234' or 'TFA Primus'
    model = LaCrosse WS-2812IT
    # The driver to use:
    driver = weewx.drivers.ws28xx
...
```

knackko, 2020-05-09:
I did a cast on type to int:
   3460         newBuffer[0][6] = (int(haddr) >> 16) & 0x0F | 16 * (comInt & 0xF)
   3461         newBuffer[0][7] = (int(haddr) >> 8 ) & 0xFF
   3462         newBuffer[0][8] = (int(haddr) >> 0 ) & 0xFF

No error anymore, but dont know if it is right.

tkeffer, 2020-05-10:
Your version of the ws28xx.py driver does not match mine. For example, on the released version, line [3460](https://github.com/weewx/weewx/blob/v4.0.0/bin/weewx/drivers/ws28xx.py#L3460) looks like this:

    TMMUX            = 0x71

While your stack trace is showing an assignment to `newBuffer` on that line.

The released version of ws28xx.py shows `DRIVER_VERSION` of 0.50. What does yours show?

vsrods, 2020-05-10:
ws28 version at start of weewx:
```
May 10 16:16:28 knadock weewx[14083] INFO weewx.engine: Loading station type WS28xx (weewx.drivers.ws28xx)
May 10 16:16:28 knadock weewx[14083] INFO weewx.drivers.ws28xx: driver version is 0.50
```

tkeffer, 2020-05-10:
Odd. Somehow you ended up with a different version of the driver. Do you recall modifying it?

What happens if you use the "real" v0.50? You can get it here: https://raw.githubusercontent.com/weewx/weewx/v4.0.0/bin/weewx/drivers/ws28xx.py

tkeffer, 2020-05-10:
See [this thread](https://groups.google.com/forum/#!topic/weewx-user/BKRbFtagkQU). Exact same problem as you. Try the driver attached to the thread.

tkeffer, 2020-05-10:
Fixed in commit 4a73dc1. Still, I'd be interested in hearing what happened to your version of the driver.

closed 2020-05-10 by tkeffer

knackko, 2020-05-10:
I am using deb packages for weewx from:
http://weewx.com/apt/python3

Here is the list of drivers versions:

```
Module name              Driver name    Version  Status                   
  weewx.drivers.acurite    AcuRite        0.4                               
  weewx.drivers.cc3000     CC3000         0.30                              
  weewx.drivers.fousb      FineOffsetUSB  1.20                              
  weewx.drivers.simulator  Simulator      3.2                               
  weewx.drivers.te923      TE923          0.40                              
  weewx.drivers.ultimeter  Ultimeter      0.40                              
  weewx.drivers.vantage    Vantage        3.2.0                             
  weewx.drivers.wmr100     WMR100         3.5.0                             
  weewx.drivers.wmr200     WMR200         3.5.0                             
  weewx.drivers.wmr300     WMR300         0.33                              
  weewx.drivers.wmr9x8     WMR9x8         3.4.0                             
  weewx.drivers.ws1        WS1            0.40                              
  weewx.drivers.ws23xx     WS23xx         0.41                              
  weewx.drivers.ws28xx     WS28xx         0.50        
```