weewx/weewx #551: V4 weewx install using setup.py leaves /home/weewx owned by root

hawalter, 2020-05-12:
After installing weewx-4.0.0.tar.gz using setup.py, /home/weewx and all its sub directories are owned by root. This requires weewx.conf editors to be run as root which is a poor security practice.
Also, all the documentation for weewx extensions (and probably most other wee_ commands need to be changed to show the commands being run under sudo.

tkeffer, 2020-05-12:
Good points, although I don't think as bad as you make them out to be. After all, lots, maybe most, configuration files are owned by root.

Whether sudo is necessary when running a wee_ command depends on the details of how devices are mounted. Still, I get your point.

It would be useful to have a "Security" section in the User's Guide that discusses these tradeoffs. Care to put together a PR?

hawalter, 2020-05-12:
I will regretfully decline putting together a PR since my writing skills have never been very good.

A simple 
sudo chown -R pi:pi /home/weewx/ 
after ./setup.py install would be an improvement.

closed 2020-05-12 by tkeffer