weewx/weewx #552: Investigate more "pythonic" packaging

tkeffer, 2020-05-12:
This issue was provoked by a [long email thread](https://groups.google.com/d/msg/weewx-development/0va6QRKgG7E/EmNiEQw2BgAJ) about making WeeWX obey more standard python packaging rules. 

Here are the requirements:
1. Installation via `pip`. 
2. Ability to install into one directory, such as `/home/weewx`. It's OK if it takes some contorted setting of install directories in order to do this, but I like having all state, all libraries, all code, in one spot. It makes my life easier. I'm not real keen on having the libraries buried deep in this directory, but I'm willing to try it and see how it feels.
3. Ability to install in "standard" locations.
4. Skins, and the `user` directory cannot be touched by an upgrade.
5. `weewx.conf` gets automatically patched by an upgrade. This is a big feature of weewx over the standard package upgrades, which merely ask a user if they want to "merge" their old config file with an incoming, new config file.
6. A strategy for avoiding name clashes for modules `user` and `schema`.
7. Support for our "standard platforms," which generally includes the platforms listed in [Matthew's compatibility matrix](https://github.com/weewx/weewx/wiki/weewx4-compatibility-matrix).
8. Optionally, allow WeeWX extensions to be installed as standard pip/setup extensions.

tyll, 2020-05-19:
The strategy for user/schema seems to be a good first task since it seems to be a hard blocker. Are there any recommendations/steps how to test that it still works after a change? I guess it is related to installing extensions.

tkeffer, 2020-05-20:
Yes. After making your modification, I would test installing, using, then uninstalling a few extensions.

tkeffer, 2020-05-23:
> Did you consider upgrading the config when running weewx the first time
after upgrade instead of during the upgrade process? Could this be an
alternative? This could still be enabled/disabled in the config file.

That's a possibility, although that means WeeWX isn't really "installed" until you run something. It also means that, at least for a while, you have "old" weewx configuration files with "new" software, which sounds like a possible support problem.

tkeffer, 2020-05-30:
Till, have you thought any more about this?

vinceskahan, 2020-06-26:
FWIW - I stumbled across https://docs.fedoraproject.org/en-US/packaging-guidelines/Python/ today, which I know is how to RPMize python things (the opposite of this issue) but I thought I'd pass it along in case there's anything of value in there...

tyll, 2020-06-26:
> Till, have you thought any more about this?

Uh, just saw your message. I had to return the borrowed weather station and waiting for a new one. When I have it, I will take another look.

> FWIW - I stumbled across [docs.fedoraproject.org/en-US/packaging-guidelines/Python](https://docs.fedoraproject.org/en-US/packaging-guidelines/Python/) today, which I know is how to RPMize python things (the opposite of this issue) but I thought I'd pass it along in case there's anything of value in there...

Once weewx is pythonic, it will be very simple to RPMize it.

tkeffer, 2020-07-25:
Doesn't seem to be much outside interest in pursuing this, so I'm closing the issue. If anyone shows an interest, we can reopen.

closed 2020-07-25 by tkeffer