weewx/weewx #555: Error in Forecast.py under version 4

redcardan104, 2020-05-20:
File "/usr/share/weewx/user/forecast.py", line 1212, in update_forecast
May 20 09:00:21 weatherpi weewx[15068] CRITICAL __main__:     ****      elif time.time() - self.interval > self.last_ts:
May 20 09:00:21 weatherpi weewx[15068] CRITICAL __main__:     ****  TypeError: '>' not supported between instances of 'float' and 'NoneType'

tkeffer, 2020-05-20:
I'm sorry, but forecast is not part of WeeWX. You should post your note on the forecast website: https://github.com/weewx/weewx/wiki/forecasting

closed 2020-05-20 by tkeffer

redcardan104, 2020-05-20:
Thank you

 

Dan

 <mailto:dan@done-that.com> dan@done-that.com

When you have what you need, use the rest to bring joy into someone else’s life -Firoozeh Dumas

If you can afford it and the other can not, you pick up the check – Irv Freedman