weewx/weewx #556: String encoding issue when using Python 3 for WS1 driver

windcrusader, 2020-05-23:
https://github.com/weewx/weewx/blob/85525a230f538dff80713f200da668754a41a532/bin/weewx/drivers/ws1.py#L252

I tried to upgrade my system to Python 3 today and I can across what appears to be a simple encoding error in the WS1 driver when using Python 3. In Python 3 when the serial data is pulled in from the device it encodes it as a byte string e.g b"xxxxx". The checks for a valid string in the driver don't account for this. 

I managed to fix this in my version by changing the referenced line to:
`buf = self.get_readings().decode("utf-8")`

However, I feel that perhaps this is not a general fix, as it may break the driver when it is used in Python 2. I'm sure there is a better way to handle the byte strings that will be robust across both versions. Can someone please advise, then I'll submit a pull request with the fix.

windcrusader, 2020-05-23:
Actually I did a brief check on a terminal session of Python 2.17.16 and seems that calling .decode() won't have any negative consequences; at least I think so. 

If this is agreed, I'll make the pull request to fix the issue.

tkeffer, 2020-05-23:
Thanks, Brad, for finding this and, even better, figuring out a solution.

However, I would prefer changing the comparison at line [143](https://github.com/weewx/weewx/blob/85525a230f538dff80713f200da668754a41a532/bin/weewx/drivers/ws1.py#L143), rather than do a decode. The reason is that the data coming off the station really are bytearrays, not strings. So, the line becomes

            if buf[0:2] != b'!!':

There may be other lines with similar problems --- I didn't check. I can also see that the TCP version has similar problems.

See the Wiki article [*Supporting Python 3*](https://github.com/weewx/weewx/wiki/Python-3-strategy) for a general discussion.

Thanks again!

closed 2020-05-24 by windcrusader