weewx/weewx #558: There is no weewx-crt.tgz in master.zip file

G400, 2020-05-25:
No weewx-crt.tgz in master.zip file

From documentation:
Download file:
wget -O weewx-crt.zip https://github.com/matthewwall/weewx-crt/archive/master.zip

Then run the extension installer:
wee_extension --install weewx-crt.tgz

There is no weewx-crt.tgz

tkeffer, 2020-05-25:
Obviously a simple typo. The Wiki is correctable by anyone.

closed 2020-05-25 by tkeffer

G400, 2020-05-25:
Assuming you mean   weewx-crt.tgz should be weewx-crt.zip.
I tried this, but no success.

Request to install 'weewx-crt.zip'
Extracting from zip archive weewx-crt.zip
Traceback (most recent call last):
  File "/usr/share/weewx/wee_extension", line 88, in <module>
    main()
  File "/usr/share/weewx/wee_extension", line 80, in main
    ext.install_extension(options.install)
  File "/usr/share/weewx/weecfg/extension.py", line 113, in install_extension
    self.tmpdir, self.logger)
  File "/usr/share/weewx/weecfg/__init__.py", line 1760, in extract_zip
    zip_archive = zipfile.ZipFile(filename)
  File "/usr/lib/python3.7/zipfile.py", line 1222, in __init__
    self._RealGetContents()
  File "/usr/lib/python3.7/zipfile.py", line 1289, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file