weewx/weewx #560: Unable to install weewx-crt

G400, 2020-05-25:
I posted this question:

https://github.com/weewx/weewx/issues/558
Assuming a typo in the documentation.
From documentation:
Download file:
wget -O weewx-crt.zip https://github.com/matthewwall/weewx-crt/archive/master.zip

Guessing install install 'weewx-crt.tgz should be weewx-crt.zip. Inspecting header of file also confirms this is a zip file.

Still no luck:
sudo wee_extension --install weewx-crt.zip
Request to install 'weewx-crt.zip'
Extracting from zip archive weewx-crt.zip
Traceback (most recent call last):
File "/usr/share/weewx/wee_extension", line 88, in
main()
File "/usr/share/weewx/wee_extension", line 80, in main
ext.install_extension(options.install)
File "/usr/share/weewx/weecfg/extension.py", line 113, in install_extension
self.tmpdir, self.logger)
File "/usr/share/weewx/weecfg/init.py", line 1760, in extract_zip
zip_archive = zipfile.ZipFile(filename)
File "/usr/lib/python3.7/zipfile.py", line 1222, in init
self._RealGetContents()
File "/usr/lib/python3.7/zipfile.py", line 1289, in _RealGetContents
raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file

But the only answer I get is this:
tkeffer: crt is not part of WeeWX. You should post an issue on its site.

I', trying to install crt and crt install is failing. I can't understand this is an weewx issue?

tkeffer, 2020-05-25:
Just tried this on my machine and it worked perfectly. 

Try expanding `weewx-crt.zip`, then installing:

```shell
unzip weewx-crt.zip
sudo wee_extension --install=weewx-crt-master
```

I don't think this is a bug in WeeWX. Probably something got corrupted.

G400, 2020-05-25:
The missing "=" in readme and in Github caused the install issue
sudo wee_extension --install=weewx-crt-master
                                             ^
Thank you tkeffer!

tkeffer, 2020-05-25:
I don't think so. The `=` sign is optional. In any case, this is not a bug in WeeWX. Closing.

closed 2020-05-25 by tkeffer