weewx/weewx #562: Python3 conversion has completely forgotten about rainmachine.py

bigjools, 2020-05-28:
I've installed the debian buster packages on Ubuntu 20.04. It won't start.

Upon investigation, the interpreter was complaining about mixing tabs and spaces for indentation (!) and after I fixed that, it was apparent that the code is still using Python 2 imports (`import Queue` for example).

bigjools, 2020-05-28:
`/usr/share/weewx/user/rainmachine.py` is the bad file.

tkeffer, 2020-05-28:
`rainmachine` is not part of WeeWX. You should contact the author.

closed 2020-05-28 by tkeffer

bigjools, 2020-05-28:
Apologies, my bad! I installed it so long ago that I forgot.

tkeffer, 2020-05-29:
I had nothing better to do, so I forked it, then tried a quick port. Take a look at https://github.com/tkeffer/rainmachine-weewx.