weewx/weewx #564: centos 8 rpm requires non-existent 'python' package

vinceskahan, 2020-05-29:
The 4.1.0 rpm has some built-in dependencies that make no sense on a centos 8 system and that prevent the rpm from being installed on a minimal centos 8 system.

Most notably it 'requires' /usr/bin/python yet no centos 8 package provides that file, which doesn't exist on a centos 8 system.  The spec file needs some tweaking to be python3 only in the centos 8 rpm.

vinceskahan, 2020-05-29:
I've done some fiddling with the spec file and makefile and got a modified centos 8 rpm that removes the /usr/bin/python requirement, built in a centos 8 vagrant box, but I hit it with a chainsaw rather than doing more surgical tweaking.   I'll take a more controlled approach in seeing if there are easier fixes for this one.

vinceskahan, 2020-05-29:
This patch works for me for building on centos 8 (important) and builds both centos 7+8 rpms that seem to look ok for requires related to python.  Give it a try.

```
diff --git a/pkg/weewx.spec.in b/pkg/weewx.spec.in
index fe40f7bd..3bcf22db 100644
--- a/pkg/weewx.spec.in
+++ b/pkg/weewx.spec.in
@@ -37,7 +37,7 @@
 %define initdir %{_initrddir}
 # must install pyserial and pyusb using easy_install or pip
 %define deps python, python-setuptools, python-configobj, python-cheetah, python-imaging
-%define bld_deps python, python-configobj
+%define bld_deps python3, python3-configobj, platform-python-devel
 %define python python2
 %define html_root /var/www/html/weewx
 %endif
@@ -49,7 +49,7 @@
 %define initdir %{_initrddir}
 # must install cheetah from epel-release or using pip3
 %define deps python3, python3-setuptools, python3-configobj, python3-pillow, python3-pyserial, python3-pyusb
-%define bld_deps python, python-configobj
+%define bld_deps python3, python3-configobj, platform-python-devel
 %define python python3
 %define html_root /var/www/html/weewx
 %endif
@@ -86,6 +86,25 @@ PWSweather.com, or CWOP.
 %prep
 %setup -q

+#--- needed to build on CentOS 8 or Fedora equivalent ---
+# ref: https://fedoraproject.org/wiki/Changes/Make_ambiguous_python_shebangs_error
+#  buildrequire needs platform-python-devel
+#
+## Fix all Python shebangs recursively in .
+# -p preserves timestamps
+# -n prevents creating ~backup files
+# -i specifies the interpreter for the shebang
+# Need to list files that do not match ^[a-zA-Z0-9_]+\.py$ explicitly!
+pathfix.py -pni "%{__python3} %{py3_shbang_opts}" .  bin/wee_config
+pathfix.py -pni "%{__python3} %{py3_shbang_opts}" .  bin/wee_database
+pathfix.py -pni "%{__python3} %{py3_shbang_opts}" .  bin/wee_debug
+pathfix.py -pni "%{__python3} %{py3_shbang_opts}" .  bin/wee_device
+pathfix.py -pni "%{__python3} %{py3_shbang_opts}" .  bin/wee_extension
+pathfix.py -pni "%{__python3} %{py3_shbang_opts}" .  bin/wee_import
+pathfix.py -pni "%{__python3} %{py3_shbang_opts}" .  bin/wee_reports
+pathfix.py -pni "%{__python3} %{py3_shbang_opts}" .  bin/weewxd
+pathfix.py -pni "%{__python3} %{py3_shbang_opts}" .  bin/wunderfixer
+
 %build

 %install
```

matthewwall, 2020-05-30:
the problem is not the spec file.  the problem is that `wee_import` uses `/usr/bin/python` in its shebang.  apparently the tools that build an rpm scan all of the files to find the shebangs, and those go into the dependency list.

fix will appear in weewx 4.1.1

vinceskahan, 2020-05-30:
Unbelievable.  They couldn't just have done what was specified and trusted the packager (person).   Sigh.  Thanks.

matthewwall, 2020-05-30:
this appears to explain it:

https://stackoverflow.com/questions/33507865/rpmbuild-unexpected-dependencies

matthewwall, 2020-05-30:
yup, the AutoReqProv` fixed it (and i fixed wee_import).  not sure where the `/bin/sh` references are coming from.

```
$ rpm -qp --requires dist/weewx-4.1.1-1.el8.noarch.rpm
warning: dist/weewx-4.1.1-1.el8.noarch.rpm: Header V4 RSA/SHA1 Signature, key ID 15306a67: NOKEY
python3
python3-setuptools
python3-configobj
python3-pillow
python3-pyserial
python3-pyusb
/bin/sh
/bin/sh
/bin/sh
rpmlib(PartialHardlinkSets) <= 4.0.4-1
rpmlib(FileDigests) <= 4.6.0-1
rpmlib(PayloadFilesHavePrefix) <= 4.0-1
rpmlib(CompressedFileNames) <= 3.0.4-1
rpmlib(PayloadIsXz) <= 5.2-1

$ rpm -qp --requires dist/weewx-4.1.1-1.el7.noarch.rpm
warning: dist/weewx-4.1.1-1.el7.noarch.rpm: Header V4 RSA/SHA1 Signature, key ID 15306a67: NOKEY
python
python-setuptools
python-configobj
python-cheetah
python-imaging
/bin/sh
/bin/sh
/bin/sh
rpmlib(PartialHardlinkSets) <= 4.0.4-1
rpmlib(FileDigests) <= 4.6.0-1
rpmlib(PayloadFilesHavePrefix) <= 4.0-1
rpmlib(CompressedFileNames) <= 3.0.4-1
rpmlib(PayloadIsXz) <= 5.2-1
```

matthewwall, 2020-05-30:
also removed the `BuildRequires`, since we do that in a much easier to understand (and easier to override) way in the makefile.

vinceskahan, 2020-05-30:
Perfect.  It's a python program, it should only require python interpreters that match the runtime platform.   Glad to see perl is gone (as not everybody uses logwatch).  I was just going to mention that.

matthewwall, 2020-05-31:
so now the rpm package builds work fine on older centos, but the latest rpmbuild is facist about the shebangs:

```
+ /usr/lib/rpm/redhat/brp-mangle-shebangs
*** ERROR: ambiguous python shebang in /usr/share/weewx/wee_config: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/share/weewx/wee_database: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/share/weewx/wee_debug: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/share/weewx/wee_device: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/share/weewx/wee_extension: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/share/weewx/wee_import: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/share/weewx/wee_reports: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/share/weewx/weewxd: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/share/weewx/wunderfixer: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
error: Bad exit status from /var/tmp/rpm-tmp.PJSmta (%install)
```
in fact, we *want* the shebang to be `/usr/bin/env python`, not an explicit python invocation!

so either we have to special-case the redhat builds and do it the (latest) redhat-way, or figure out some way to turn off this stupid nanny behavior of rpmbuild.

sigh.

matthewwall, 2020-05-31:
it sounds like we can replace the shebangs for el8 package with `/usr/bin/python3`:

https://docs.fedoraproject.org/en-US/packaging-guidelines/Python/#Multiple_Python_Runtimes

then it is still possible to use `/etc/default/weewx` to force python2, without having to hack the shebangs in the actual install.

vinceskahan, 2020-05-31:
You can use the method I provided in %prep which seems to be their
preferred method to shut it up, but you'd need to also hack it back in
%post if you were adamant about your desired shebang. Kludgy but it should
work.

matthewwall, 2020-05-31:
we keep the `/usr/bin/env python` shebangs in the source.  that is the most reliable, cross-platform approach.  perhaps we change it to `usr/bin/env python3` since python3 is now the default supported environment for weewx (just as all of the source code uses `/home/weewx` as the assumed installation destination).

then we replace the shebangs on any platforms that have 'special needs'.

matthewwall, 2020-05-31:
now i have to figure out how to turn off this redhat facist crap, otherwise it is not possible to build a centos7 package on a centos8 system.  or we can just say "don't build on centos8 because the tools are too inflexible - stick with centos7 for the builds"

vinceskahan, 2020-05-31:
A few choices....

1. build both on centos7 and kick the can til it goes EOL in 2024
2. build every box on its native os one-by-one (again, Docker makes it
easier)
3. give a bit on the 'env' thing and do what the os upstream wants you to do
4. workaround the issue by post-processing back to your path in %post (did
you try that ?)

I'd be willing to publish Dockerfile(s) or Vagrantfile(s) for valid centos
build hosts if you want to do (2) above.   Let me know.

matthewwall, 2020-05-31:
thanks vince, but this is now resolved.  i used sed to replace the shebangs, with `/usr/bin/python2` for the centos7 package and `/usr/bin/python3` for the centos8 package.

and we work around redhat's restrictive worldview using the `/etc/default/weewx` with `/usr/bin/wee*` shell stubs.

closed 2020-05-31 by matthewwall