weewx/weewx #566: TypeError with last gauge-data.txt.tmpl

CdfCastronno, 2020-06-01:
Ciao,
From the first installation of Gauges Steel i've this error:

Jun  2 00:58:22 raspberrypi weewx[28959] DEBUG weewx.reportengine: Running report 'SteelSeries'
Jun  2 00:58:22 raspberrypi weewx[28959] DEBUG weewx.reportengine: Found configuration file /etc/weewx/skins/ss/skin.conf for report 'SteelSeries'
Jun  2 00:58:22 raspberrypi weewx[28959] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx/ss
Jun  2 00:58:22 raspberrypi weewx[28959] DEBUG weewx.cheetahgenerator: Using search list ['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Station', 'weewx.cheetahgenerator.Current', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInfo', 'weewx.cheetahgenerator.Extras']
Jun  2 00:58:22 raspberrypi weewx[28959] DEBUG weewx.manager: Daily summary version is 2.0
Jun  2 00:58:22 raspberrypi weewx[28959] ERROR weewx.cheetahgenerator: Generate failed with exception '<class 'TypeError'>'
Jun  2 00:58:22 raspberrypi weewx[28959] ERROR weewx.cheetahgenerator: **** Ignoring template /etc/weewx/skins/ss/gauge-data.txt.tmpl
Jun  2 00:58:22 raspberrypi weewx[28959] ERROR weewx.cheetahgenerator: **** Reason: unsupported operand type(s) for *: 'NoneType' and 'int'
Jun  2 00:58:22 raspberrypi weewx[28959] ERROR weewx.cheetahgenerator: ****  Traceback (most recent call last):
Jun  2 00:58:22 raspberrypi weewx[28959] ERROR weewx.cheetahgenerator: ****    File "/usr/share/weewx/weewx/cheetahgenerator.py", line 322, in generate
Jun  2 00:58:22 raspberrypi weewx[28959] ERROR weewx.cheetahgenerator: ****      unicode_string = compiled_template.respond()
Jun  2 00:58:22 raspberrypi weewx[28959] ERROR weewx.cheetahgenerator: ****    File "_etc_weewx_skins_ss_gauge_data_txt_tmpl.py", line 348, in respond
Jun  2 00:58:22 raspberrypi weewx[28959] ERROR weewx.cheetahgenerator: ****  TypeError: unsupported operand type(s) for *: 'NoneType' and 'int'
Jun  2 00:58:22 raspberrypi weewx[28959] INFO weewx.cheetahgenerator: Generated 1 files for report SteelSeries in 0.08 seconds

Could you help me to understand how i've forgot or wrong?
i use WH1080 with SDR antenna

http://cdfcastronno.altervista.org/WEEWX/
Thanks
Gianni

tkeffer, 2020-06-01:
Hello,

SteelSeries is not part of WeeWX. You should contact the author.

closed 2020-06-01 by tkeffer

gjr80, 2020-06-01:
@CdfCastronno. I have posted your problem [here](https://groups.google.com/d/topic/weewx-user/eWbsmA41ZL0/discussion) in the [weewx-user](https://groups.google.com/forum/#!forum/weewx-user) group. Please refer to that thread for further troubleshooting.

Gary