weewx/weewx #572: install latest forecast in weewx 4.1.1. Python3 stalls with `TypeError: '>' not supported between instances of 'float' and 'NoneType'`

peters77, 2020-06-15:
Hi, 

installing latest forecast extension with weewx 4.1.1., Puytjon3 and responsive theme and using simulator for testing gives this error. Any idea howto fix that or is this a problem with using the simulator?

Regards,

Christian 

The error:

Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__: Caught unrecoverable exception:
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  '>' not supported between instances of 'float' and 'NoneType'
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  Traceback (most recent call last):
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 195, in run
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      self.dispatchEvent(weewx.Event(weewx.CHECK_LOOP, packet=packet))
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 224, in dispatchEvent
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      callback(event)
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 578, in check_loop
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      raise BreakLoop
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  weewx.engine.BreakLoop
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  During handling of the above exception, another exception occurred:
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  Traceback (most recent call last):
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 594, in post_loop
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      self._catchup(self.engine.console.genArchiveRecords)
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 638, in _catchup
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      for record in generator(lastgood_ts):
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/drivers/__init__.py", line 30, in genArchiveRecords
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      raise NotImplementedError("Method 'genArchiveRecords' not implemented")
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  NotImplementedError: Method 'genArchiveRecords' not implemented
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  During handling of the above exception, another exception occurred:
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  Traceback (most recent call last):
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewxd", line 154, in main
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      engine.run()
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 202, in run
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      self.dispatchEvent(weewx.Event(weewx.POST_LOOP))
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 224, in dispatchEvent
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      callback(event)
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 596, in post_loop
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      self._software_catchup()
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 658, in _software_catchup
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      origin='software'))
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 224, in dispatchEvent
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      callback(event)
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****    File "/usr/share/weewx/user/forecast.py", line 1212, in update_forecast
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****      elif time.time() - self.interval > self.last_ts:
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  TypeError: '>' not supported between instances of 'float' and 'NoneType'
Jun 15 08:30:16 weewx weewx[3514] CRITICAL __main__:     ****  Exiting.

jordemort, 2020-06-19:
This also just happened to me with the interceptor driver; I had Zambretti and NWS enabled.

jordemort, 2020-06-19:
This fork seems to work: https://github.com/chaunceygardiner/weewx-forecast

closed 2020-06-26 by tkeffer