weewx/weewx #577: Different behavior now with xtypes.get_series()

poblabs, 2020-06-21:
As outlined by @gjr80 quite [nicely in this thread](https://groups.google.com/d/msg/weewx-user/UWSnqt7tY3Q/6BY0yiWvAAAJ), the weewx 4 `xtypes.get_series()` has some different behavior. 

Essentially what I'm seeing is:

The `archiveDaySpan` below does not return 288 results (based on a 5 minute archive interval)

`archive.getSqlVectors(TimeSpan(1592107200, 1592193600), outTemp, None, None)`

However, the `archiveYearSpan` returns 366 results (based on the full year span)

`archive.getSqlVectors(TimeSpan(1577854800, 1609477200), outTemp, max, 86400)`

This isn't consistent behavior given the time spans. It appears the aggregate is changing the behavior. 

That being said, I have users requesting this "full span" behavior where observations are populated with `None` values so their charts cover the full time span whether populated with data or not. 

So I wonder if we can either;

a) Modify all the `archive*Span`'s with this aggregate "bug", so even if aggregate is `None` it'll still populate the full timespan, or...
b) fix the aggregate issue within `archiveYearSpan`, but add a new function parameter to all the `archive*Span` to fill in the full span with `None`s.

Gary mentioned he would open a ticket on this. I haven't seen that yet. If he has, please let me know and we can close this as duplicate.

tkeffer, 2020-06-21:
Let me make sure I understand the problem.

Right now:
1. A vector from midnight to midnight produces points only up to the present time, which is generally something less than 288 results.
2. A vector from 1 Jan to 1 Jan, with no aggregations, produces points only up to the present time, which is generally something less than 365 * 288 results.
3. A vector from 1 Jan to 1 Jan, with aggregations over a day, always produces 365 results.

It sounds like you are satisfied with `#3` above, but want `#1` and `#2` to always produce 288 or 365*288 results, respectively. 

If that's the case, I don't see an easy way to do this because it requires knowledge about the archive interval in the future. Remember, in WeeWX, the archive interval can change.

Alternatively, if the goal is to bring `#3` into alignment with `#1` and `#2`, we can change `#3`, so it only produces data up to the present time. This would be the old `archive.getSqlVectors()` behavior.

poblabs, 2020-06-21:
Yep, I think you've got it exactly correct - at least how I understand it, too. 

I didn't want to implement something into my skin without knowing if this is expected behavior or not. 

I'm ok with whatever path forward. So if bringing `#3` into alignment with `#1` and `#2` is the preferred method forward, then I could probably add the `"full timespan"` option as a skin option. I would have to determine the interval as defined from the `config_dict`, and calculate that way possibly. 

If leaving everything alone because this is actually expected behavior, then that's fine too - I can still (I think) implement something in my skin. 

Thanks!

tkeffer, 2020-06-21:
The behavior has changed because how aggregates of a series are calculated has changed.

Before, the aggregates were calculated within `archive.getSqlVectors()`. Now, `get_series()`  breaks the series up into aggregation intervals, then uses `get_aggregation()` to calculate the aggregate of each interval. This makes it easier to introduce new aggregation types.

If the argument `timespan` to the function `get_series()` runs from the beginning of the year, to the present time, then you will get a series that covers just that time. If `timespan` runs from the beginning of the year to the end of the year, you will get a full year, with additional values beyond the present set to `None`. 

The stock ImageGenerator sets `timespan` according to the option `time_length` in `skin.conf`. This generally means the series runs from 12 months ago to the present. Changing that would require something other than ImageGenerator.

I'll readily admit that WeeWX plotting needs to be overhauled and made more flexible.

In summary: I don't see an easy way to change the present behavior.

poblabs, 2020-06-21:
I'm fine leaving everything alone since it's behavior is intended. Let me ask the question another way, how can I replicate this behavior to the other `archive*Span`'s? Is setting the [aggregate to `max`](https://groups.google.com/d/msg/weewx-user/UWSnqt7tY3Q/Gv9qdWTNAAAJ) as suggested the right idea (say in the case of `archiveDaySpan`)?

In that case, I don't need to alter my skin because the code already exists.

tkeffer, 2020-06-21:
I guess that would work, although `last` or `avg` might be better choices.

closed 2020-06-21 by tkeffer

poblabs, 2020-06-22:
@tkeffer I had another question regarding this. What if I wanted `archiveYearSpan` to just return Jan 1 to "today", while still maintaining aggregation? Is that possible in some method?

tkeffer, 2020-06-23:
If the `timespan` argument to `get_series()` runs from 1-Jan to today, that should work.

poblabs, 2020-06-23:
Which makes sense. Just getting used to the change in `get_series()` since previously using that with `archiveYearSpan` would have different results. I think I will work on adding some extra "specific" timespans to the skin to provide all scenarios. Thanks again