weewx/weewx #578: Wmr200: read_device() USB Error Reason:[Errno 110] Operation timed out

brachiur, 2020-06-22:
Hello all,
firstly thanks so much for your work!

As newbie, I'm having a problem running Weewx on Raspi 4 and Wmr200 as weather station.

I installed weewx 3 days ago, and already I have received the error (attached below) twice, that has blocked weewx.

I have solved rebooting the Raspi, but I hope there is a smarter solution.

Some infomation:
Raspbian version:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster

Weewx version:
Weewx version 4.1.1
Installed following http://www.weewx.com/docs/debian.htm

Thanks for any hints,
Alessandro

Jun 20 06:44:21 raspberrypi weewx[2902] DEBUG weewx.manager: Daily summary version is 2.0
Jun 20 06:44:22 raspberrypi weewx[2902] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 0.81 seconds
Jun 20 06:44:22 raspberrypi weewx[2902] DEBUG weewx.manager: Daily summary version is 2.0
Jun 20 06:44:22 raspberrypi weewx[2902] INFO weewx.imagegenerator: Generated 15 images for report SeasonsReport in 0.81 seconds
Jun 20 06:44:22 raspberrypi weewx[2902] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx
Jun 20 06:44:22 raspberrypi weewx[2902] DEBUG weewx.reportengine: Report 'SmartphoneReport' not enabled. Skipping.
Jun 20 06:44:22 raspberrypi weewx[2902] DEBUG weewx.reportengine: Report 'MobileReport' not enabled. Skipping.
Jun 20 06:44:22 raspberrypi weewx[2902] DEBUG weewx.reportengine: Report 'StandardReport' not enabled. Skipping.
Jun 20 06:44:22 raspberrypi weewx[2902] DEBUG weewx.reportengine: Report 'FTP' not enabled. Skipping.
Jun 20 06:44:22 raspberrypi weewx[2902] DEBUG weewx.reportengine: Report 'RSYNC' not enabled. Skipping.
Jun 20 06:44:30 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7252 live_queue_len:1
Jun 20 06:44:31 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8050
Jun 20 06:44:36 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7253 live_queue_len:1
Jun 20 06:44:37 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8051
Jun 20 06:44:38 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   adjust_rain rain_total:4.55 Rain.rain_total_last:4.55 rain_interval:0.00
Jun 20 06:44:38 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7254 live_queue_len:1
Jun 20 06:44:39 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8052
Jun 20 06:44:39 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7255 live_queue_len:1
Jun 20 06:44:40 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8053
Jun 20 06:44:49 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7256 live_queue_len:1
Jun 20 06:44:50 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8054
Jun 20 06:44:50 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7257 live_queue_len:1
Jun 20 06:44:51 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8055
Jun 20 06:44:57 raspberrypi weewx[2902] WARNING weewx.drivers.wmr200: Clock time unsynchronized
Jun 20 06:44:57 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7258 live_queue_len:1
Jun 20 06:44:58 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8056
Jun 20 06:44:59 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7259 live_queue_len:1
Jun 20 06:45:00 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8057
Jun 20 06:45:03 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7260 live_queue_len:1
Jun 20 06:45:04 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8058
Jun 20 06:45:04 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7261 live_queue_len:1
Jun 20 06:45:05 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8059
Jun 20 06:45:09 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7262 live_queue_len:1
Jun 20 06:45:10 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8060
Jun 20 06:45:12 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Acknowledged control packet rx:488
Jun 20 06:45:15 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:7263 live_queue_len:1
Jun 20 06:45:16 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:8062 
Jun 20 06:45:16 raspberrypi weewx[2902] INFO weewx.manager: Added record 2020-06-20 06:45:00 UTC (1592635500) to database 'weewx.sdb'
Jun 20 06:45:16 raspberrypi weewx[2902] INFO weewx.manager: Added record 2020-06-20 06:45:00 UTC (1592635500) to daily summary in 'weewx.sdb'
Jun 20 06:45:16 raspberrypi weewx[2902] DEBUG weewx.reportengine: Running reports for latest time in the database.
Jun 20 06:45:16 raspberrypi weewx[2902] DEBUG weewx.drivers.wmr200: genLoop() phase getting live packets
Jun 20 06:45:16 raspberrypi weewx[2902] WARNING weewx.drivers.wmr200: Skipped bytes before resync:3
Jun 20 06:45:16 raspberrypi weewx[2902] DEBUG weewx.reportengine: Running report 'SeasonsReport'
Jun 20 06:45:16 raspberrypi weewx[2902] DEBUG weewx.reportengine: Found configuration file /etc/weewx/skins/Seasons/skin.conf for report 'SeasonsReport'
Jun 20 06:45:16 raspberrypi weewx[2902] DEBUG weewx.cheetahgenerator: Using search list ['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Station', 'weewx.cheetahgenerator.Current', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInfo', 'weewx.cheetahgenerator.Extras']
Jun 20 06:45:16 raspberrypi weewx[2902] DEBUG weewx.manager: Daily summary version is 2.0
[... 33 lines cut ...]
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****      device._ctx.managed_claim_interface(device, interface)
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/core.py", line 102, in wrapper
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****      return f(self, *args, **kwargs)
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/core.py", line 167, in managed_claim_interface
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****      self.backend.claim_interface(self.handle, i)
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 811, in claim_interface
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****      _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 595, in _check
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****      raise USBError(_strerror(ret), ret, _libusb_errno[ret])
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****  usb.core.USBError: [Errno 16] Resource busy
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****  
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****  During handling of the above exception, another exception occurred:
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****  
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/engine.py", line 103, in setupStation
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****      self.console = loader_function(config_dict, self)
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 58, in loader
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****      return WMR200(**config_dict[DRIVER_NAME])
[... cut ...]
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 196, in open_device
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****      raise weewx.WakeupError(exception)
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL weewx.engine:     ****  weewx.WakeupError: [Errno 16] Resource busy
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL __main__: Unable to load driver: [Errno 16] Resource busy
Jun 20 06:46:54 raspberrypi weewx[2902] CRITICAL __main__:     ****  Exiting...

tkeffer, 2020-06-22:
Are you able to reproduce this error?

It could be caused by another program trying to claim the interface.

What do you get if you run

```shell
sudo lsusb -v -d 0x0fde:0xca01
```

You can also see if there is any recent kernel activity related to the device:
```shell
dmesg | grep ca01
```

brachiur, 2020-06-22:
Thanks a lot for your reply.

No, I can't reproduce the error, but it almost appears each 1.5/2 days.

The answer to "sudo lsusb -v -d 0x0fde:0xca01" is:
Bus 001 Device 004: ID 0fde:ca01 Oregon Scientific WMRS200 weather station
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0fde Oregon Scientific
  idProduct          0xca01 WMRS200 weather station
  bcdDevice            3.02
  iManufacturer           0 
  iProduct                1 Universal Bridge
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

To "dmesg | grep ca01" is:
[    1.435108] usb 1-1.4: New USB device found, idVendor=0fde, idProduct=ca01, bcdDevice= 3.02
[ 5350.976060] usb 1-1.3: New USB device found, idVendor=0fde, idProduct=ca01, bcdDevice= 3.02

Thanks again.

tkeffer, 2020-06-22:
I'm not an expert on this, but it looks to me like the device originally connected as `usb 1-1.4`, then, for some reason, connected again as `usb 1-1.3`, resulting in the resource busy error.

Are you sure no other process is competing for its attention?

This is going to take some sleuthing on your part, but you can find out which process is holding which usb port by using

```shell
dmesg | fgrep '/dev/bus/usb'
```

brachiur, 2020-06-22:
There is only weewx installed on Raspi, being reserved only to the weather station.
Answer to this "sudo lsof | fgrep '/dev/bus/usb'" is:

lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
python3   14194                       root   10u      CHR      189,3      0t0      36369 /dev/bus/usb/001/004
python3   14194 14199 python3         root   10u      CHR      189,3      0t0      36369 /dev/bus/usb/001/004
python3   14194 14200 python3         root   10u      CHR      189,3      0t0      36369 /dev/bus/usb/001/004
python3   14194 14201 python3         root   10u      CHR      189,3      0t0      36369 /dev/bus/usb/001/004

Thanks again

cmanton, 2020-06-22:
The observation that the USB id was assigned differently is interesting.  I'm not sure how USB id's are assigned/re-assigned, but it's curious that it would not be re-assigned the same one barring any other changes to the system.

My guess is that the wmr200 retry mechanism expects the same usb id, and/or libusb (which wmr200 uses to access the device) has a similar issue.

I have had poor experience with a raspberry pi and the wmr200 w.r.t. reliability.  Newer raspberry pi's maybe closed the usb connectivity issues I saw.  I think one solution was to ensure the raspberry pi had a sufficient power supply providing at least 2 amps.

Since re-starting the driver seems to fix it, there is likely an ability to reset state within the failing weewx instance, unless it falls into the libusb domain.

There may be a possibility of a software fix either in wmr200 driver releasing resources or libusb, but I would need to repro and start the "dig" into the failure.

Is there another non-raspberry pi solution you could try to see if the issues persist in another system ?  It's likely a system integration issue.

persson, 2020-06-22:
Probably completely unrelated to your issue... but after refusing to believe in the analog “ferrite bead” solution for my digital USB instability, in desperation I eventually added one to my USB cable for my Vantage Pro.  Sporadic USB failures connections to a low power Linux host (much less frequent than you are reporting, on the order of once every few months) disappeared completely.  Might be worth eliminating it as a possible factor if your cable doesn’t already have one.

brachiur, 2020-06-23:
Thanks you Chris, I see your point.
Just so you know, the power supply connected with Raspi is 3A.

> Is there another non-raspberry pi solution you could try to see if the issues persist in another system ? It's likely a system integration issue.
Right now, I maybe can try only creating a bootable USB stick with Ubuntu on Windows computer and installing weewx there.

brachiur, 2020-06-23:
Thanks for your hint Erik.
Actually, I have already tried to use a USB cable with ferrite bead (at this moment, I'm using a double ferrite bead) following some suggestions I found online.
Unfortunately without success, the problem seems to be uncorrelated with communication issues due to the wire.

cmanton, 2020-06-23:
I re-read and see it's a raspi4...probably fine, as the earlier issues with usb (I think) have since been fixed.

The only solution is to reboot ?  Meaning, after failure a re-launch of the weewx process doesn't work ?  If that's true, it suggests a libusb, kernel or hardware resource failure.

So the failing scenario is this:
1. failed read with a timeout (why?)
2. failed recovery (why?)

The wmr200 related code is this class.  This is the surface area interacting with libusb, and subsequently the device.
https://github.com/weewx/weewx/blob/effdfa5a0687da71b23934426ec1d4c97ccd66f7/bin/weewx/drivers/wmr200.py#L144

Jun 20 06:45:54 raspberrypi weewx[2902] ERROR weewx.drivers.wmr200: read_device() USB Error Reason:[Errno 110] Operation timed out

Would the failure occur if there is a powered usb hub in between the wmr200 and the rpi4 ?

brachiur, 2020-06-24:
Thanks again Chris.

> The only solution is to reboot ? Meaning, after failure a re-launch of the weewx process doesn't work ? If that's true, it suggests a libusb, kernel or hardware resource failure.
Actually, I was wrong. Last time I encountered the error (Sunday) I tried with a simple stop&start of weewx process, and it worked fine.

> Would the failure occur if there is a powered usb hub in between the wmr200 and the rpi4 ?
I'm sorry, but I have not a powered USB hub as a test at this time.

cmanton, 2020-06-24:
ok, so it sounds like there is a potential clean up solution w/out
reboot...but requires a new process, so there is some process state not
getting cleaned up between wmr restarts.

There is still the issue of why the device goes offline so
frequently...that is unexpected behavior.

brachiur, 2020-06-29:
Just giving you an update: the system is up and running since 6 days without any problem.
No changes have been made.

cmanton, 2020-06-29:
Good to hear.

There are 2 failures;
1. the initial device off-line from the USB link
2. the inability to recover from this failure.

weewx/wmr200 can only address (2), and then only if the external
libraries/kernel allow recovery.

Addressing (1) would be the best option, preventing initial failure, but
that is very difficult without a better understanding of the failure.

tfilo, 2020-11-03:
Hello, I see that this issue is closed but is there any working solution ? I used weewx for many years with older raspbian and it worked fine. Today I upgraded to Raspberry PI OS Buster, installed weewx based on python 3 and I got this error. I can reproduce it every boot. First I see in log that weewx is waiting 60 second for mysql to be ready. Than it shows this error and exits. If I start weewx manually, everything is working.

cmanton, 2020-11-03:
I hear 2 problems.
1. mysql access delay
2. USB error communicating with wmr200

The fact that the system *used* to work prior to the upgrade, and that if you start manually everything works lead me to believe that there isn't an immediate bug with the wmr200 driver.  (It does not mean there isn't a bug, just that there is a known working solution)

I'm not sure what it means to 'manually' start weewx (versus automated ?), but my first attack vector would be to investigate those differences.

The USB errors simply indicate a failing path to the device, which could be many things especially with the number of changes to your system.

tfilo, 2020-11-03:
By manually I meen when I do sudo /etc/init.d/weewx stop; sudo /etc/init.d/weewx start; versus automated when it starts at boot. Difference is that when it starts at boot it needs to wait for mysql. But you can see in log that wmr200 driver starts working `weewx.drivers.wmr200: USB polling device thread signaled to start`  than it waits for mysql. And after 60 seconds it can't connect again usb device. So I think that it somehow reserved usb device, than after waiting for mysql it maybe restarted process ???? And then when it tries again to connect to usb, usb is already busy.

tfilo, 2020-11-03:
Now I checked, process Id didn't change between waiting for mysql and wmr200 usb error.

● weewx.service - LSB: weewx weather system
   Loaded: loaded (/etc/init.d/weewx; generated)
   Active: active (running) since Tue 2020-11-03 18:11:12 CET; 38min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 553 ExecStart=/etc/init.d/weewx start (code=exited, status=0/SUCCESS)
    Tasks: 3 (limit: 2160)
   CGroup: /system.slice/weewx.service
           └─687 python3 /usr/share/weewx/weewxd --daemon --pidfile=/var/run/weewx.pid /etc/weewx/weewx.conf

nov 03 18:11:13 meteo python3[687]: weewx[687] INFO weewx.drivers.wmr200: Created USB polling thread to read block on device
nov 03 18:11:13 meteo python3[687]: weewx[687] INFO weewx.drivers.wmr200: USB polling device thread for live data launched
nov 03 18:11:13 meteo python3[687]: weewx[687] INFO weewx.drivers.wmr200: Reset console device
nov 03 18:11:14 meteo python3[687]: weewx[687] INFO weewx.drivers.wmr200: USB polling device thread signaled to start
nov 03 18:11:14 meteo python3[687]: weewx[687] INFO weewx.drivers.wmr200: Started watchdog thread live data
nov 03 18:11:14 meteo python3[687]: weewx[687] INFO weewx.engine: StdConvert target unit is 0x10
nov 03 18:11:14 meteo python3[687]: weewx[687] INFO weewx.drivers.wmr200: USB polling device thread exiting
nov 03 18:11:14 meteo python3[687]: weewx[687] INFO weewx.drivers.wmr200: USB polling thread expired
nov 03 18:11:14 meteo python3[687]: weewx[687] CRITICAL __main__: Database connection exception: (2002, "Can't connect to local MySQL server through socket '/
nov 03 18:11:14 meteo python3[687]: weewx[687] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...

● weewx.service - LSB: weewx weather system
   Loaded: loaded (/etc/init.d/weewx; generated)
   Active: active (running) since Tue 2020-11-03 18:11:12 CET; 38min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 553 ExecStart=/etc/init.d/weewx start (code=exited, status=0/SUCCESS)
    Tasks: 3 (limit: 2160)
   CGroup: /system.slice/weewx.service
           └─687 python3 /usr/share/weewx/weewxd --daemon --pidfile=/var/run/weewx.pid /etc/weewx/weewx.conf

nov 03 18:49:50 meteo python3[687]: weewx[687] CRITICAL weewx.engine:     ****      self.console = loader_function(config_dict, self)
nov 03 18:49:50 meteo python3[687]: weewx[687] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 58, in loader
nov 03 18:49:50 meteo python3[687]: weewx[687] CRITICAL weewx.engine:     ****      return WMR200(**config_dict[DRIVER_NAME])
nov 03 18:49:50 meteo python3[687]: weewx[687] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 1468, in __init__
nov 03 18:49:50 meteo python3[687]: weewx[687] CRITICAL weewx.engine:     ****      self.usb_device.open_device()
nov 03 18:49:50 meteo python3[687]: weewx[687] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 196, in open_device
nov 03 18:49:50 meteo python3[687]: weewx[687] CRITICAL weewx.engine:     ****      raise weewx.WakeupError(exception)
nov 03 18:49:50 meteo python3[687]: weewx[687] CRITICAL weewx.engine:     ****  weewx.WakeupError: [Errno 16] Resource busy
nov 03 18:49:50 meteo python3[687]: weewx[687] CRITICAL __main__: Unable to load driver: [Errno 16] Resource busy
nov 03 18:49:50 meteo python3[687]: weewx[687] CRITICAL __main__:     ****  Exiting...

cmanton, 2020-11-03:
When you start weewx manually, you are starting on a fully running system.
When it starts automatically, weewx starts in the init sequenced order provided by the OS init boot scripts.

There may be an init dependency difference between the two OS installations.

As a work around, you could replace the weewx init script with your own *after* system is booted to see if that works.

Something like this (may not fully apply to your installation)
https://tldp.org/HOWTO/HighQuality-Apps-HOWTO/boot.html

tfilo, 2020-11-03:
Yes this can work, and i will start weewx delayed but it is only workaround. I think it should be working out of the box with latest not modified debian/raspbian etc.

cmanton, 2020-11-03:
Agreed.

duane-nepa, 2020-11-03:
Do you have a rtc? Or are you using fakehwclock? As this is also a problem with rpi.

tfilo, 2020-11-03:
duane-nepa: I don't have rtc. And I don't use fakehwclock either. I configured ntp to update time after boot. And I modified /usr/share/weewx/user/extensions.py and added wait loop until datetime is higher than specified unixtimestamp. This will guarantees that it will not start saving records with wrong time. But can be this somehow related to this issue ? If this is issue I can buy and install RTC but it wasn't needed before and it worked for few year fine.

While trying to figure out best workaround, i disabled systemd-networkd-wait-online.service and i seems this helped. It most likely changed some order how process start at boot. Now after two reboots, both time weewx started fine. And wasn't waiting for mysql too.

And cmanton please don't get me wrong, I really appreciate work done by all weewx developers. I think it is great software and i use it for few years. I just tried to submit as much information as possible to figure this out and help maybe to other users with same issue.

Thank you both for answers.

cmanton, 2020-11-03:
tfilo: Please don't worry about when you have problems...that is why we try to help.  Thanks for your support !

duane-nepa, 2020-11-03:
I could be wrong. I used to have weewx wait until it had a reasonable time. But I don't know where this was done as I never looked for it.

I also don't know how the records are retrieved from the WMR200 as I don't have one.

I think however that weewx is now starting before some needed OS service is loaded.

Unless you specifically disabled fakehwclock it's used by default. Even if you use ntpd which seems slower at getting an initial sync (with iburst set) than timesyncd which is the default time sync demon.

 I use chronyd on my systems which is an updated ntpd. I also maintain my own ntp server (with chrony) on my LAN. It happens to be on the same hardware as weewx since it's always on.

Of course this could be unrelated to your problem.

tfilo, 2020-11-03:
I disabled and uninstaled fakehwclock so it is based only on ntp, when raspbian starts date is somewhere in 2019 and than it updates to actual date. I don’t know if this could be reason for this issue. But since i disabled wait for online, weewx works fine but it can be only good luck that some service starts sooner now. I will use it as it is. It this problem occueres again i will come back. But i don’t neex wait for online enabled so this is not any issue for me. Thanks for your time.

tfilo, 2020-11-05:
So tonight it happend again. Weewx stoped working with same error at 23:15 in night. Now at morning I did systemctl restart weewx and it is working fine again. But if it will repeat every few days it will be problem.

tfilo, 2020-11-26:
I enabled debug logs becasue it happens every few days. If somebody can help figure it out.

```
Nov 25 20:51:53 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32958 live_queue_len:1
Nov 25 20:51:54 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35158
Nov 25 20:52:03 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32959 live_queue_len:1
Nov 25 20:52:04 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35159
Nov 25 20:52:10 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32960 live_queue_len:1
Nov 25 20:52:10 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35160
Nov 25 20:52:10 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32961 live_queue_len:1
Nov 25 20:52:11 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35161
Nov 25 20:52:13 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32962 live_queue_len:1
Nov 25 20:52:14 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35162
Nov 25 20:52:15 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Acknowledged control packet rx:2197
Nov 25 20:52:23 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32963 live_queue_len:1
Nov 25 20:52:24 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35164
Nov 25 20:52:24 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32964 live_queue_len:1
Nov 25 20:52:25 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35165
Nov 25 20:52:26 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   adjust_rain rain_total:1.57 Rain.rain_total_last:1.57 rain_interval:0.00
Nov 25 20:52:26 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32965 live_queue_len:1
Nov 25 20:52:27 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35166
Nov 25 20:52:33 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32966 live_queue_len:1
Nov 25 20:52:34 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35167
Nov 25 20:52:38 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32967 live_queue_len:1
Nov 25 20:52:39 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35168
Nov 25 20:52:43 meteo weewx[5021] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:32968 live_queue_len:1
Nov 25 20:52:44 meteo weewx[5021] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:35169
Nov 25 20:52:58 meteo weewx[5021] ERROR weewx.drivers.wmr200: read_device() USB Error Reason:[Errno 110] Operation timed out
Nov 25 20:52:58 meteo weewx[5021] ERROR weewx.drivers.wmr200: USB device read error
Nov 25 20:52:58 meteo weewx[5021] ERROR weewx.drivers.wmr200: USB polling thread unexpectedly terminated
Nov 25 20:52:58 meteo weewx[5021] INFO weewx.engine: Main loop exiting. Shutting engine down.
Nov 25 20:52:58 meteo weewx[5021] INFO weewx.engine: Shutting down StdReport thread
Nov 25 20:52:58 meteo weewx[5021] DEBUG weewx.engine: StdReport thread has been terminated
Nov 25 20:52:58 meteo weewx[5021] DEBUG weewx.restx: Shut down StationRegistry thread.
Nov 25 20:52:58 meteo weewx[5021] INFO weewx.drivers.wmr200: USB polling thread expired
Nov 25 20:52:58 meteo weewx[5021] CRITICAL __main__: Caught WeeWxIOError: USB polling thread unexpectedly terminated
Nov 25 20:52:58 meteo weewx[5021] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...
Nov 25 20:53:49 meteo kernel: [199701.334274] [UFW BLOCK] IN=enxb827eb4d09c7 OUT= MAC=01:00:5e:00:00:01:84:26:15:dd:dc:53:08:00:46:c0:00:20:40:82:00:00:01:02:41:eb SRC=192.168.1.1 DST=224.0.0.1 LEN=32 TOS=0x00 PREC=0xC0 TTL=1 ID=16514 PROTO=2 
Nov 25 20:53:57 meteo kernel: [199709.312367] [UFW BLOCK] IN=enxb827eb4d09c7 OUT= MAC=01:00:5e:00:00:fb:20:17:42:ef:1e:69:08:00:46:c0:00:20:00:00:40:00:01:02:41:62 SRC=192.168.1.18 DST=224.0.0.251 LEN=32 TOS=0x00 PREC=0xC0 TTL=1 ID=0 DF PROTO=2 
Nov 25 20:53:58 meteo weewx[5021] INFO __main__: retrying...
Nov 25 20:53:58 meteo weewx[5021] INFO __main__: Using configuration file /etc/weewx/weewx.conf
Nov 25 20:53:58 meteo weewx[5021] INFO __main__: Debug is 1
Nov 25 20:53:58 meteo weewx[5021] DEBUG __main__: Initializing engine
Nov 25 20:53:58 meteo weewx[5021] INFO weewx.engine: Loading station type WMR200 (weewx.drivers.wmr200)
Nov 25 20:53:58 meteo weewx[5021] INFO weewx.drivers.wmr200: driver version is 3.5.0
Nov 25 20:53:58 meteo weewx[5021] INFO weewx.drivers.wmr200: sensor map is {'altimeter': 'altimeter', 'pressure': 'pressure', 'windSpeed': 'wind_speed', 'windDir': 'wind_dir', 'windGust': 'wind_gust', 'windBatteryStatus': 'battery_status_wind', 'inTemp': 'temperature_0', 'outTemp': 'temperature_1', 'extraTemp1': 'temperature_2', 'extraTemp2': 'temperature_3', 'extraTemp3': 'temperature_4', 'extraTemp4': 'temperature_5', 'extraTemp5': 'temperature_6', 'extraTemp6': 'temperature_7', 'extraTemp7': 'temperature_8', 'inHumidity': 'humidity_0', 'outHumidity': 'humidity_1', 'extraHumid1': 'humidity_2', 'extraHumid2': 'humidity_3', 'extraHumid3': 'humidity_4', 'extraHumid4': 'humidity_5', 'extraHumid5': 'humidity_6', 'extraHumid6': 'humidity_7', 'extraHumid7': 'humidity_8', 'inHeatindex': 'heatindex_0', 'heatindex [... cut ...]
Nov 25 20:53:58 meteo weewx[5021] CRITICAL weewx.engine:     ****      raise weewx.WakeupError(exception)
Nov 25 20:53:58 meteo weewx[5021] CRITICAL weewx.engine:     ****  weewx.WakeupError: [Errno 16] Resource busy
Nov 25 20:53:58 meteo weewx[5021] CRITICAL __main__: Unable to load driver: [Errno 16] Resource busy
Nov 25 20:53:58 meteo weewx[5021] CRITICAL __main__:     ****  Exiting...
```

I understand that WMR200 will be no longer supported starting from weewx v. 4.3 but I would like to solve this so I can stay on v4.2 forever. Thanks in advance for any help.

tfilo, 2020-12-15:
Hello, i noticed that you opened this thread again. Since I was loosing lot of data becasue of this issue. I had to do something. This bug to me occured since I updated from Raspbian 8 to RaspberyPI OS 10 a there was switch from python 2 to python 3 too. I took a risk and installed weewx again but on python 2 in Raspberry PI OS 10 (based on debian 10). Now it is few days, but it seems that weewx works well under python 2. No occurrence of this error anymore. But I am on old not recommended python 2.

tkeffer, 2020-12-15:
Thank you, Tomáš! That is good to know.

I would be interested in hearing if switching to Python 2 works for other people with "Resource busy" problems.

lylavoie, 2020-12-28:
Hello All,

I'm seeing exactly the same error, after upgrading to python3, weewx 4.2.0.  This is on a Raspberry Pi Zero, running Rasp OS 10 (buster), but I also saw this on OS 9 (stretch) as well, as I had hoped maybe upgrading to the latest OS and libraries would help things.  

Couple of interesting points, my log file looks just like the above, USB timeout, closes that USB thread, causes the main thread to also close (see below).  I'm starting weewx using systemd.  What's interesting is, systemd service should see the main thread exit and restart it, but it also seems like the main thread doesn't actually completely exit (process is still there), so systemd doesn't actually see this as a failed service and handle the restart.

I'm going to try downgrading to python 2, and see if that also works for me (I'll report back).

Cheers,
Lincoln

`Dec 28 13:21:38 raspberrypi weewxd[311]: weewx[311] INFO weewx.imagegenerator: Generated 14 images for report SeasonsReport in 5.09 seconds
Dec 28 13:21:38 raspberrypi weewxd[311]: weewx[311] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx
Dec 28 13:21:46 raspberrypi weewxd[311]: weewx[311] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:260 live_queue_len:1
Dec 28 13:21:47 raspberrypi weewxd[311]: weewx[311] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:323
Dec 28 13:21:47 raspberrypi weewxd[311]: weewx[311] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:261 live_queue_len:1
Dec 28 13:21:48 raspberrypi weewxd[311]: weewx[311] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:324
Dec 28 13:21:56 raspberrypi weewxd[311]: weewx[311] DEBUG weewx.drivers.wmr200:   Queuing live packet rx:262 live_queue_len:1
Dec 28 13:21:57 raspberrypi weewxd[311]: weewx[311] DEBUG weewx.drivers.wmr200: genLoop() Yielding live queued packet id:325
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] ERROR weewx.drivers.wmr200: write_device() Unable to send USB control message [Errno 110] Operation timed out
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] INFO weewx.engine: Main loop exiting. Shutting engine down.
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] INFO weewx.engine: Shutting down StdReport thread
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] DEBUG weewx.engine: StdReport thread has been terminated
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] DEBUG weewx.restx: Shut down PWSWeather thread.
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] DEBUG weewx.restx: Shut down Wunderground-PWS thread.
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] DEBUG weewx.restx: Shut down StationRegistry thread.
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] INFO weewx.drivers.wmr200: USB polling device thread exiting
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] INFO weewx.drivers.wmr200: USB polling thread expired
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] CRITICAL __main__: Caught WeeWxIOError: [Errno 110] Operation timed out
Dec 28 13:22:02 raspberrypi weewxd[311]: weewx[311] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] INFO __main__: retrying...
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] INFO __main__: Using configuration file /usr/local/weewx/weewx.conf
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] INFO __main__: Debug is 1
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] DEBUG __main__: Initializing engine
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] INFO weewx.engine: Loading station type WMR200 (weewx.drivers.wmr200)
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] INFO weewx.drivers.wmr200: driver version is 3.5.0
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] INFO weewx.drivers.wmr200: sensor map is {'altimeter': 'altimeter', 'pressure': 'pressure', 'windSpeed': 'wind_speed', 'windDir':
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.drivers.wmr200: open_device() Unable to claim USB interface. Reason: [Errno 16] Resource busy
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] ERROR weewx.engine: Import of driver failed: [Errno 16] Resource busy (<class 'weewx.WakeupError'>)
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****    File "/usr/local/weewx/bin/weewx/drivers/wmr200.py", line 193, in open_device
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****      self.handle.claimInterface(self.interface)
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/legacy.py", line 230, in claimInterface
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****      util.claim_interface(self.dev, interface)
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/util.py", line 205, in claim_interface
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****      device._ctx.managed_claim_interface(device, interface)
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/core.py", line 102, in wrapper
[... cut ...]
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****    File "/usr/local/weewx/bin/weewx/drivers/wmr200.py", line 196, in open_device
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****      raise weewx.WakeupError(exception)
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL weewx.engine:     ****  weewx.WakeupError: [Errno 16] Resource busy
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL __main__: Unable to load driver: [Errno 16] Resource busy
Dec 28 13:23:02 raspberrypi weewxd[311]: weewx[311] CRITICAL __main__:     ****  Exiting...
lines 1025-1050/1050 (END)

`

lylavoie, 2020-12-29:
Update, running on python 2.7, there are still USB timeout issues, however, it seems like the restarts are able to occur successfully, so the system recovers and keeps going.  Last restart count 4, over about 4 to 5 hours.

lylavoie, 2020-12-30:
Final run was 1 day, 4 hours, 25 minutes, then it hit the same error and failed to recover / restart and stopped.

tfilo, 2020-12-30:
For me weewx is working for more than 15 days with python2. Before with python3 it failed every few days and was unable to recover.

No I checked logs and there seems to be still same errors, but it can recover from this error by itself.

```
Dec 30 01:31:28 meteo weewx[666] ERROR weewx.drivers.wmr200: read_device() USB Error Reason:[Errno 110] Operation timed out
Dec 30 01:31:28 meteo weewx[666] ERROR weewx.drivers.wmr200: USB device read error
Dec 30 01:31:29 meteo weewx[666] ERROR weewx.drivers.wmr200: USB polling thread unexpectedly terminated
Dec 30 01:31:29 meteo weewx[666] INFO weewx.engine: Main loop exiting. Shutting engine down.
Dec 30 01:31:29 meteo weewx[666] INFO weewx.engine: Shutting down StdReport thread
Dec 30 01:31:29 meteo weewx[666] INFO weewx.drivers.wmr200: USB polling thread expired
Dec 30 01:31:29 meteo weewx[666] INFO weewx.drivers.wmr200: Watchdog received shutdown
Dec 30 01:31:29 meteo weewx[666] INFO weewx.drivers.wmr200: Watchdog thread exiting
Dec 30 01:31:29 meteo weewx[666] INFO weewx.drivers.wmr200: Watchdog thread expired
Dec 30 01:31:29 meteo weewx[666] INFO weewx.drivers.wmr200: Received packet count live:24544 archive:5 control:1623
Dec 30 01:31:29 meteo weewx[666] INFO weewx.drivers.wmr200: Received bytes:1831912 sent bytes:38832
Dec 30 01:31:29 meteo weewx[666] INFO weewx.drivers.wmr200: Packet archive queue len:5 live queue len:0
Dec 30 01:31:29 meteo weewx[666] WARNING weewx.drivers.wmr200: Exiting with packets still in archive queue cnt:5
Dec 30 01:31:29 meteo weewx[666] INFO weewx.drivers.wmr200: Driver gracefully exiting
Dec 30 01:31:29 meteo weewx[666] CRITICAL __main__: Caught WeeWxIOError: USB polling thread unexpectedly terminated
Dec 30 01:31:29 meteo weewx[666] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...
Dec 30 01:32:15 meteo kernel: [96845.549813] [UFW BLOCK] IN=enxb827eb4d09c7 OUT= MAC=01:00:5e:00:00:01:84:26:15:dd:dc:53:08:00:46:c0:00:20:ac:0e:00:00:01:02:d6:5e SRC=192.168.1.1 DST=224.0.0.1 LEN=32 TOS=0x00 PREC=0xC0 TTL=1 ID=44046 PROTO=2 
Dec 30 01:32:29 meteo weewx[666] INFO __main__: retrying...
Dec 30 01:32:29 meteo weewx[666] INFO __main__: Using configuration file /etc/weewx/weewx.conf
Dec 30 01:32:29 meteo weewx[666] INFO __main__: Debug is 0
Dec 30 01:32:29 meteo weewx[666] INFO weewx.engine: Loading station type WMR200 (weewx.drivers.wmr200)
Dec 30 01:32:29 meteo weewx[666] INFO weewx.drivers.wmr200: driver version is 3.5.0
```

closed 2021-01-30 by tkeffer

CameronD73, 2021-01-31:
A few observations.  I tried a USB cable unplug/reconnect with my WMR300 and weewx recovers well (Debian 10, Python3.7, weewx 4.3.0.)

Now, the drivers are very different overall between WMR200 and WMR300 models, but initialisation is similar and I noticed that the WRM300 version calls `self.handle.reset()` before trying detachKernelDriver().
Comments in the source suggest the reset call isn't doing much, but perhaps it is more important in the Python3 Pyusb code. It would be a trivial test if  somebody with a wmr200 wanted to try it out.

In my experience, if USB devices are disconnected and reconnected under Linux, they are never allocated the same device number - always a new one. However  that should not be a problem because the device's open() code does a search for the device.

When weewx uses mysql on the same system, the better approach is to create a systemd unit file and instruct it to wait for the mysql service.  There is an example in the wiki.  Continuing to use  SysV init files seems to give random results - I don't know how systemd combines the startup order from init files with unit file specs.  In any case, recent weewx recovers well from the "mysql server has gone away" issue, so long as mysqld actually does return.

zoomx, 2021-03-31:
I have the same issue in different hardware and OS
FITPC2i with Ubuntu 18.04.5 LTS bionic, CPU is Intel(R) Atom(TM) CPU Z530   @ 1.60GHz
It is not a console hardware or cable fault since it works uner Windows (on toher hardware) with old Wx Data Logger.
Power source is the one that is selled with FITPC2i.
WeeWX version 	4.4.0
Python 3.6.9

mr-bryn, 2021-08-09:
I've recently encontered the same "[Errno 16] Resource busy" error.

I changed my setup to run the USB connection over 10m cat6 ethernet with an adapter. For extra complexity weewx is running within a VM on proxmox 7 and the WMR200 is assigned to the VM. With the cat6, adapters, USB passthrough, etc, I suspect I could have a phyiscal connectivity issue with my WMR200.

Previously I've been running plugged in directly to a raspberry pi 2 model b for years and it's been ok (in my experience the pi itself isn't the most stable platform, hence experimenting with a different setup).

I've tried both python 3 and 2 versions of weewx. Under python 3 I used WMR200 driver v3.5.2, and under python 2 I used v3.3.5 (because it's what I used on the pi). It didn't seem to make a difference.

Is there any way I can determine if I have a USB connection issue to the WMR, or some other gremlin?
I'm willing to test a beta driver if anyone is able to look into improving the driver's ability to handle connectivity issues.

superhet70, 2022-05-09:
I'm also running Raspbian 11 (bullseye) with Python 3. This is running on a Raspberry Pi 2 Model B version 1.1. 
Weewx version 4.8.0. Weather station is WMR200.
I had issues with under voltage alarms so I purchased a genuine Raspberry Pi power supply and those errors have disappeared.
However I get the _Unable to send USB control message [Errno 110] Operation timed out_ failure almost daily. It waits for 60 seconds and tries to restart the driver but this fails with _CRITICAL user.wmr200: open_device() Unable to claim USB interface. Reason: [Errno 16] Resource busy_. It then tries to exit, _CRITICAL __main__:     ****  Exiting..._ but it doesn't exit and I have to stop weewx and restart it. 
Has anyone had success reverting to Python 2 on bullseye?
 Below it an extract of syslog
[weewx-syslog-failure.txt](https://github.com/weewx/weewx/files/8649368/weewx-syslog-failure.txt)
.

tfilo, 2022-05-09:
@superhet70 I suggested workaround in this issue thread https://github.com/weewx/weewx-wmr200/issues/3 since WMR200 was removed and placed to separate repository. It works for me for few months now.

superhet70, 2022-05-10:
@tfilo Thanks for the script. I have created a crontab and syslog show it running the script every 5 minutes. 
Currently there is no restart_log.txt file. Fingers crossed! Given the failure rate I should know in a couple of days if this works.

Thanks again.

superhet70, 2022-05-12:
I solved the missing restart_log.txt. (root doesn't have a $HOME directory.) Weewx has been running now for 3 days and has restarted vi the script twice in the last 24 hours. I have now commented the writing of 'ok' to the log file to reduce the writing to the SD card.

tfilo, 2022-05-12:
@superhet70 Great, i am glad it works for you. It is really good to disable 'ok' output. In my case it was mainly for debugging purpose. And since i use overlay fs in ram. I don't care about writes too much until RAM while there is enough free ram.