weewx/weewx #579: weedb.NoDatabaseError: Attempt to open a non-existent database %(WEEWX_ROOT)s/archive/forecast.sdb

tkeffer, 2020-06-29:
Discovered by @chaunceygardiner. His theory:

The problem is that `deep_copy` temporarily turns off interpolation. A context switch happens. Another service attempts to use the dictionary, but gets the wrong value because of the inactive interpolation.

See the comments under this commit https://github.com/weewx/weewx/commit/b6d2424482c2a7dc6b46e7f542c644e13383baea

tkeffer, 2020-06-29:
Commit https://github.com/weewx/weewx/commit/a3dca5aa6ca8fad928e7b37adaf56a279453f443 should fix this. Needs to be tested

tkeffer, 2020-06-29:
[Comment from @chaunceygardiner](https://github.com/weewx/weewx/commit/b6d2424482c2a7dc6b46e7f542c644e13383baea#commitcomment-40242444):

> WeeWX doesn’t start with this change:
> 
> Jun 29 09:22:01 judy weewx[21280] CRITICAL __main__: Database OperationalError exception: Attempt to open a non-existent database %(WEEWX_ROOT)s/archive/weewx.sdb
> Jun 29 09:22:01 judy weewx[21280] CRITICAL __main__: **** Waiting 2 minutes then retrying...
> 
> I have a meeting now, so I can’t investigate further; would you see if it starts up for you?
> 

Works fine for me.

How about some details:
1. what version of Python
2. what version of configobj:
    ```
    python -c "import configobj;print(configobj.__version__)"
    ```

chaunceygardiner, 2020-06-29:
$ python3 --version
Python 3.7.3

$ python3 -c "import configobj;print(configobj.__version__)"
5.0.6

python3/stable,now 3.7.3-1 armhf [installed,automatic]
python3-configobj/stable,now 5.0.6-3 all [installed]

tkeffer, 2020-06-29:
Hang on. I am able to replicate the error. It was not on startup, but when the reports are run.

chaunceygardiner, 2020-06-29:
That makes sense.  I have an extension that needs to read from the database at startup.

I added a line to dump the stack.  You probably don’t need that now, but since it’s on my clipboard:

Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__: Database OperationalError exception: Attempt to open a non-existent database %(WEEWX_ROOT)s/archive/weewx.sdb 
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****  Waiting 2 minutes then retrying...                                                                  
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****  Traceback (most recent call last):                                                                  
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/weewxd", line 148, in main                                                  
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      engine = weewx.engine.StdEngine(config_dict)                                                    
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/weewx/engine.py", line 76, in __init__                                      
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      self.loadServices(config_dict)                                                                  
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/weewx/engine.py", line 141, in loadServices                                 
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      obj = weeutil.weeutil.get_object(svc)(self,config_dict_copy)                                    
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/user/loopdata.py", line 272, in __init__                                    
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      dbm = db_binder.get_manager(default_binding)                                                    
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/weewx/manager.py", line 530, in get_manager                                 
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      self.manager_cache[data_binding] = open_manager(manager_dict, initialize)                       
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/weewx/manager.py", line 683, in open_manager                                
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      manager_dict['table_name'])                                                                     
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/weewx/manager.py", line 128, in open                                        
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      connection = weedb.connect(database_dict)                                                       
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/weedb/__init__.py", line 92, in connect                                     
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      return driver_mod.connect(**db_dict)                                                            
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/weedb/sqlite.py", line 52, in connect                                       
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      return Connection(database_name=database_name, SQLITE_ROOT=SQLITE_ROOT, **argv)                 
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/weedb/sqlite.py", line 29, in guarded_fn                                    
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      return fn(*args, **kwargs)                                                                      
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****    File "/home/weewx/bin/weedb/sqlite.py", line 120, in __init__                                     
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****      raise weedb.NoDatabaseError("Attempt to open a non-existent database %s" % self.file_path)      
Jun 29 14:46:09 judygirldog weewx[10832] CRITICAL __main__:     ****  weedb.NoDatabaseError: Attempt to open a non-existent database %(WEEWX_ROOT)s/archive/weewx.sdb

tkeffer, 2020-06-29:
Try it now. The problem was actually unrelated to thread contention. Instead, it was a deep copy that failed to copy the interpolation setting.

In any case, I changed to strategy option 2 --- the deep copy is now done without touching the interpolation property

chaunceygardiner, 2020-06-29:
Now I make it to report generation; but get the following error.  Alas, I can’t look further at present as I have meeting in about to start (unfortunately, that seems to be what I do all day long).

Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine: Caught unrecoverable exception in generator 'weewx.cheetahgenerator.CheetahGenerator'
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****  Parsing failed with several errors.#012First error at line 271.
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****  Traceback (most recent call last):
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/reportengine.py", line 196, in run
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****      obj.start()
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/reportengine.py", line 281, in start
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****      self.run()
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/cheetahgenerator.py", line 131, in run
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****      gen_dict = deep_copy(self.skin_dict)
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weeutil/config.py", line 242, in deep_copy
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****      interpolation=old_dict.interpolation)
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****    File "/usr/lib/python3/dist-packages/configobj.py", line 1229, in __init__
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****      self._load(infile, configspec)
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****    File "/usr/lib/python3/dist-packages/configobj.py", line 1318, in _load
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****      raise error
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****    File "<string>", line None
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****  configobj.ConfigObjError: Parsing failed with several errors.
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****  First error at line 271.
Jun 29 15:50:06 judy weewx[2601] ERROR weewx.reportengine:         ****  Generator terminated

tkeffer, 2020-06-29:
What's at line 271 in your copy of `weewx.conf`?

I'm amazed you have any time at all during the day. You must be working from home...

chaunceygardiner, 2020-06-29:
Yep, working from home.  For me, that has meant nearly non-stop video conferences with little time to get any work done.

Line 271 is pretty innocuous:

cached_radar_filename = radar.gif

If you want to see it in context, it’s the 11th line below.

    [[LiveSeasonsReport]]                                                                                                                                                 
        skin = LiveSeasons                                                                                                                                                
        enable = true                                                                                                                                                     
        [[[CheetahGenerator]]]                                                                                                                                            
            search_list_extensions = user.forecast.ForecastVariables, user.nws.NWSForecastVariables                                                                       
            [[[[ToDate]]]]                                                                                                                                                
                [[[[[forecast]]]]]                                                                                                                                        
                    template = forecast.html.tmpl                                                                                                                         
        [[[Extras]]]                                                                                                                                                      
            radar_img = https://radar.weather.gov/ridge/lite/N0R/MUX_loop.gif                                                                                             
            cached_radar_filename = radar.gif                                                                                                                             
            radar_url = https://radar.weather.gov/ridge/radar_lite.php?product=NCR&rid=MUX&loop=yes                                                                       
            satellite_img = https://cdn.star.nesdis.noaa.gov/GOES17/ABI/SECTOR/psw/GEOCOLOR/GOES17-PSW-GEOCOLOR-600x600.gif                                               
            cached_satellite_filename = satellite.gif                                                                                                                     
            satellite_url = https://www.star.nesdis.noaa.gov/GOES/sector_band.php?sat=G17&sector=wus&band=GEOCOLOR&length=12

tkeffer, 2020-06-30:
Can you try reading it directly? It would look something like

```shell
python -c "import configobj;c=configobj.ConfigObj('/path/to/weewx.conf')"
```

chaunceygardiner, 2020-06-30:
Hi Tom,

Reading it directly works in python2 and python3.  I added a print(c) to be sure.

I next retried running weewx at head.  Same error.  I commented out line 271, and got the same error, but reported at line 270.

I ran test_config.py in py2 and py3 with no issues.

jkline@judy:~/software/weewx $ PYTHONPATH=/home/weewx/bin python3 bin/weeutil/tests/test_config.py 
..
----------------------------------------------------------------------
Ran 2 tests in 0.004s

OK
jkline@judy:~/software/weewx $ PYTHONPATH=/home/weewx/bin python bin/weeutil/tests/test_config.py 
..
----------------------------------------------------------------------
Ran 2 tests in 0.004s

OK

chaunceygardiner, 2020-06-30:
More data.  Although I can’t get weewx at head to survive an archive period on the machine above (Parsing failed with several errors.#012First error at line 271),  I DID get it to run on another machine.

I’ve been running WeeWX at head since it since ~7 PM last night—with weewx-forecast checked out before I switched database operations to the main thread.  Alas, I got the original error this morning ~9am:
   Jun 30 09:00:04 judygirldog weewx[14222] ERROR user.forecast: ZambrettiThread: Zambretti: forecast failure: Attempt to open a non-existent database %(WEEWX_ROOT)s/archive/forecast.sdb, dbm_dict: {'manager': 'weewx.manager.Manager', 'schema':—-lots-of-stuff-deleted—{'database_name': 'forecast.sdb', 'driver': 'weedb.sqlite', 'SQLITE_ROOT': '%(WEEWX_ROOT)s/archive'}}

So, in addition to it not parsing my weewx.conf on one machine, it looks like the changes don’t address the original problem.

 ---------------------- weewx End -------------------------

tkeffer, 2020-07-01:
Unfortunately, I am able to replicate none of this.

Would you be willing to send me your `weewx.conf` file? You can obfuscate any passwords first --- I won't be needing them.

chaunceygardiner, 2020-07-01:
I *think* I have all the passwords X’d out, but I’d rather send weewx.conf to you personally.  Would you send a private email to john@johnkline.com?  I will reply with the file.

chaunceygardiner, 2020-07-01:
My theory that it was the liveseasons.py service causing the config obj error didn’t pan out.  I removed it and same error.  I should have known that since the error is at reporting time, not startup time.

If you want to see the skin, I can give you access to the github repository.

tkeffer, 2020-07-01:
Silly me. Now that I look at your [stack trace](https://github.com/weewx/weewx/issues/579#issuecomment-651411434) more closely, I see that the parsing error is in one of your (many) `skin.conf` files. Can you check line 271 in these?

chaunceygardiner, 2020-07-01:
Surely it’s LiveSeasons skin.conf.  I’ve mailed it to you.

On the machine that was working with WeeWX at head, I installed LiveSeasons and enabled the report.  I get the same error.  Unfortunately, line 271 is a comment.

tkeffer, 2020-07-01:
When I try a deep copy of your `skin.conf` file it works fine. Perhaps the error is in another `skin.conf`?

I'm sorry, John, but I'm unable to replicate this, and you have so many custom services and skins, it's hard for me to do any remote debugging. You're going to have to simplify it down. In particular, the file `liveseasons.py`.

chaunceygardiner, 2020-07-01:
You can just remove liveseasons.py; but the bigger issue here is that you’ve spent a lot of time on this; and no one else is reporting problems.

I worked around the original issue and my workaround (doing the work on the main thread) seems quite solid in eliminating the problem.

I would be happy if you just reverted the two attempts at fixing my issue and called it a day.  We could wait to see if anyone else reports the issue.  I’m happy closing this for now.

Again, thanks for your time on this.

Cheers,
John

tkeffer, 2020-07-01:
The only change left behind was in the function `deep_copy()`. It no longer turns off interpolation first, and that's probably a good thing.

Closing.

closed 2020-07-01 by tkeffer