weewx/weewx #584: windDir missing from archive records

weswitt, 2020-07-15:
i have an issue where windDir values are not being included in archive records. i can see that the windDir values are being received by weewx loop packets. i can run weewxd and see the windDir values in the loop packets and then a value of "None" in the archive records.

why are the values not included in the archive records?

tkeffer, 2020-07-15:
See the documentation: http://weewx.com/docs/usersguide.htm#zero_windspeed

closed 2020-07-15 by tkeffer

weswitt, 2020-07-15:
that doc does not directly apply.

my windspeed values are in fact NOT zero. after reading the weewx code i discovered that how the wind values are presented in the loop packets matters. i was creating separate loop packets for windspeed and winddir, which i think caused weewx to believe that the speed is zero when processing the winddir. i changed my driver to deliver a single loop packet that includes both windSpeed and windDir.  this seems to have resolved the issue as  i now see windDir values in my archive records.