weewx/weewx #589: Turning off logging for the database service?

leomeyer, 2020-08-06:
This is for weewx 4.1.1 on a freshly installed Raspberry PI OS (based on Debian Buster).

I get lots of messages in /var/log/syslog like:
Aug  6 15:22:18 server weewx[11547] INFO weewx.manager: Added record 2020-08-06 15:22:00 BST (1596723720) to daily summary in 'weewx.sdb'

This fills up my ramdisk on /var/log and I want to run this system without rebooting for as long as possible. I have not found a way to turn off this log message. log_success = false is set on the top level of weewx.conf.

Is there any way to turn this off? If I disable the StdArchive service it doesn't generate skins any more, so, alternatively, is there a way to generate skins without writing to the database? I don't really need the database.

Thanks, Leo

tkeffer, 2020-08-06:
The issues list is for bug reporting and tracking. Your questions are best answered on the [user's group forum](https://groups.google.com/forum/#!forum/weewx-user).

closed 2020-08-06 by tkeffer

leomeyer, 2020-08-06:
OK, thanks.