weewx/weewx #593: OWMThread addRecord() got an unexpected keyword argument 'log_level'

SylvainGa, 2020-08-14:
Hi,

Just added the forecast module to my one month old installation of weewx on a Raspberry Pi 3 running a current version of Rasbian.

Running manually 

http://api.openweathermap.org/data/2.5/forecast?APPID=XXXXXXXXXXXXXXXXXXXXXXXXXXXX7041&lat=XXXX&lon=XXXX'

responses with a json packet so OWM with my API key is working.

However, in syslog, I have this when weewx builds its data and I can't get a forecast in weewx:

```
Aug 13 22:26:00 raspberrypi weewxd: forecast: OWMThread: OWM: download forecast from 'http://api.openweathermap.org/data/2.5/forecast?APPID=XXXXXXXXXXXXXXXXXXXXXXXXXXXX7041&lat=XXXX&lon=XXXX'
Aug 13 22:26:01 raspberrypi weewxd: forecast: OWMThread: OWM: got 40 forecast records
Aug 13 22:26:01 raspberrypi weewxd: forecast: OWMThread: OWM: saving 40 forecast records
Aug 13 22:26:01 raspberrypi weewxd: forecast: OWMThread: OWM: forecast failure: addRecord() got an unexpected keyword argument 'log_level'
Aug 13 22:26:01 raspberrypi weewxd: forecast: OWMThread: OWM: terminating thread
```

What can I do to fix this?

Thanks.

SylvainGa, 2020-08-14:
So modifying /usr/share/weewx/user/forecast.py as such fixed the error and I'm now generating forecasts.

```
                #dbm.addRecord(records, log_level=syslog.LOG_DEBUG)
                dbm.addRecord(records)
```

But why do I need to do this?

tkeffer, 2020-08-14:
The issues list is for bug reporting and tracking of WeeWX. The forecast module is not part of WeeWX. 

You should either contact the author of the extension, or post to [weewx-user](https://groups.google.com/forum/#!forum/weewx-user).

closed 2020-08-14 by tkeffer