weewx/weewx #598: Problem: Install the Bootstrap skin

DaZimmi, 2020-08-30:
Raspberry Pi 3+
system = Buster
Weewx 4.1.1

Can somebody help me? I tryed to install the Bootstrap skin and got this traceback.

[image]

DaZimmi, 2020-08-31:
> 
> 
> Raspberry Pi 3+
> system = Buster
> Weewx 4.1.1
> 
> Can somebody help me? I tryed to install the Bootstrap skin and got this traceback.
> 
> [image]

This error also occurs with wee_extension --list

:~ $ wee_extension --list
Traceback (most recent call last):
  File "/usr/share/weewx/wee_extension", line 88, in <module>
    main()
  File "/usr/share/weewx/wee_extension", line 67, in main
    config_path, config_dict = weecfg.read_config(options.config, _args)
  File "/usr/share/weewx/weecfg/__init__.py", line 179, in read_config
    default_encoding='utf-8')
  File "/usr/lib/python3/dist-packages/configobj.py", line 1229, in __init__
    self._load(infile, configspec)
  File "/usr/lib/python3/dist-packages/configobj.py", line 1318, in _load
    raise error
configobj.ConfigObjError: Parsing failed with several errors.
First error at line 235.

tkeffer, 2020-08-31:
The issues list is for reporting and tracking bugs in WeeWX. For support questions, please use [weewx-user](https://groups.google.com/forum/#!forum/weewx-user).

closed 2020-08-31 by tkeffer

Hofei90, 2020-09-01:
Granted, I am confused. 
The recommendation to submit a report here came from me.
The user gets an error message after using the documented tool wee_extension --list.
The usage is documented in the manual, if this results in a traceback it is an error in my opinion.
Granted, it wasn't helpful that @DaZimmi mixed the error message of wee_extension --list with the installation of the bootstrap skin.

However, when using wee_extension --list, the error remains with @DaZimmi.

```
:~ $ wee_extension --list
Traceback (most recent call last):
File "/usr/share/weewx/wee_extension", line 88, in
main()
File "/usr/share/weewx/wee_extension", line 67, in main
config_path, config_dict = weecfg.read_config(options.config, _args)
File "/usr/share/weewx/weecfg/init.py", line 179, in read_config
default_encoding='utf-8')
File "/usr/lib/python3/dist-packages/configobj.py", line 1229, in init
self._load(infile, configspec)
File "/usr/lib/python3/dist-packages/configobj.py", line 1318, in _load
raise error
configobj.ConfigObjError: Parsing failed with several errors.
First error at line 235.
```

tkeffer, 2020-09-01:
Not all errors are bugs! In this case, the user probably made a typo while editing the configuration file `/etc/weewx/weewx.conf`. Check line 235.