weewx/weewx #599: Wetter Problem

jh537, 2020-09-03:
Hello, the transmission to Wetter.com only works when starting the server briefly then the error occurs. Can you please help me? Thanks

`ERROR weewx.restx: Wetter: Unexpected exception of type <class 'TypeError'>
Sep  3 11:15:27 shop weewx[3745] ERROR weewx.restx: *** Traceback (most recent call last):
Sep  3 11:15:27 shop weewx[3745] ERROR weewx.restx: ***   File "/usr/share/weewx/weewx/restx.py", line 378, in run_loop
Sep  3 11:15:27 shop weewx[3745] ERROR weewx.restx: ***     self.process_record(_record, dbmanager)
Sep  3 11:15:27 shop weewx[3745] ERROR weewx.restx: ***   File "/usr/share/weewx/weewx/restx.py", line 442, in process_record
Sep  3 11:15:27 shop weewx[3745] ERROR weewx.restx: ***     self.post_with_retries(_request, data)
Sep  3 11:15:27 shop weewx[3745] ERROR weewx.restx: ***   File "/usr/share/weewx/weewx/restx.py", line 484, in post_with_retries
Sep  3 11:15:27 shop weewx[3745] ERROR weewx.restx: ***     self.check_response(_response)
Sep  3 11:15:27 shop weewx[3745] ERROR weewx.restx: ***   File "/usr/share/weewx/user/wetter.py", line 143, in check_response
Sep  3 11:15:27 shop weewx[3745] ERROR weewx.restx: ***     if txt.find('"errorcode":"100"') != -1 or \
Sep  3 11:15:27 shop weewx[3745] ERROR weewx.restx: *** TypeError: argument should be integer or bytes-like object, not 'str'
Sep  3 11:15:27 shop weewx[3745] CRITICAL weewx.restx: Wetter: Thread terminating. Reason: argument should be integer or bytes-like object, not 'str'
`

tkeffer, 2020-09-03:
This is a bug in the extension `weewx-wetter`, not WeeWX. Nevertheless, it's a simple fix. in the file `/usr/share/weewx/user/wetter.py`, change this

        txt = response.read().decode().lower()

to this

        txt = response.read().decode().lower().decode()

tkeffer, 2020-09-03:
Sorry. I gave some bad advice (which I've since deleted). My 'fix' won't work.

What version of `wetter.py` are you using, and what version of Python? My guess is that you are trying to use an old version of `wetter.py` that has not been ported to Python 3.

To find the version of `wetter.py`:

```shell
grep VERSION /usr/share/weewx/user/wetter.py
```

c0by85, 2020-09-03:
I need this fix too and get the same error. My weewx run with python 3.

```
grep VERSION /usr/share/weewx/user/wetter.py
VERSION = "0.6"
API_VERSION = "5.0.2 - 2015/06/01"
        loginf("service version is %s" % VERSION)
        loginf("wetter API version is %s" % API_VERSION)
        print("wetter uploader version %s" % VERSION)
```

tkeffer, 2020-09-03:
@c0by85 You need to upgrade your version of `wetter`. Current version is `0.7`

c0by85, 2020-09-03:
Ok, where can I find the link? My version was from the original site.

magnum61, 2020-09-03:
Same here. I just downloaded and installed it a few hours ago. So where is a newer version?

jh537, 2020-09-03:
VERSION = "0.6"
API_VERSION = "5.0.2 - 2015/06/01"
        loginf("service version is %s" % VERSION)
        loginf("wetter API version is %s" % API_VERSION)
        print("wetter uploader version %s" % VERSION)

c0by85, 2020-09-03:
[wetter.py](https://github.com/matthewwall/weewx-wetter/blob/master/bin/user/wetter.py)

```
VERSION = "0.6"
API_VERSION = "5.0.2 - 2015/06/01"
```

c0by85, 2020-09-03:
Maybe i found the version 0.7. But it was not pulled...

[https://github.com/matthewwall/weewx-wetter/pull/2/commits/2aadcfdc4aa0eb491e7f4ee7c160262841d8c8fd](https://github.com/matthewwall/weewx-wetter/pull/2/commits/2aadcfdc4aa0eb491e7f4ee7c160262841d8c8fd)

magnum61, 2020-09-03:
Could be, but we will need to wait for a confirmed archive

c0by85, 2020-09-03:
i will test with the `modded` file... first run look good.

```
Sep  3 15:25:04 pi2 weewx[27916] INFO user.wetter: service version is 0.6
Sep  3 15:25:04 pi2 weewx[27916] INFO user.wetter: wetter API version is 5.0.2 - 2015/06/01
Sep  3 15:25:04 pi2 weewx[27916] INFO user.wetter: Data will be uploaded for station id blub
```

chaunceygardiner, 2020-09-03:
It looks like Matthew Wall never merged Tom’s PR.

You can grab it from here:
https://github.com/tkeffer/weewx-wetter

chaunceygardiner, 2020-09-03:
> No, that's the V 0.6

NO, that’s the V 0.7.

jh537, 2020-09-03:
> > > You can grab it from here: https://github.com/tkeffer/weewx-wetter
> > 
> > 
> > No, that's the V0.6
> 
> change the one line and it will work...
> 
> [matthewwall/weewx-wetter@2aadcfd](https://github.com/matthewwall/weewx-wetter/commit/2aadcfdc4aa0eb491e7f4ee7c160262841d8c8fd)
> 
> ```
> -        txt = response.read().lower()
> +       txt = response.read().decode().lower()
> ```

no, work not. The error is only gone but the transmission seems not to work.

magnum61, 2020-09-03:
Cionfirmed, it works:

[image]

c0by85, 2020-09-03:
i am waiting for the update from wetter.com... there was only an update every x minutes. not so timely...

c0by85, 2020-09-03:
[image]

confirmed, it works... ;-) change the lines and go...

c0by85, 2020-09-03:
> > i am waiting for the update from wetter.com... there was only an update every x minutes. not so timely...
> 
> Hm, the update is made once the upload is done. This was for the initial time right after restarting weewx. Maybe the next updates are taking longer
> 
> But i've seen that the report is one hour "behind" the real time. My Raspberry is set to correct time. Must be something with wetter.com

hmm nope? mine is fine... no delay.

tkeffer, 2020-09-03:
It does appear that Matthew has not merged my PR. In the meantime, you can access the fixed version here https://github.com/tkeffer/weewx-wetter

magnum61, 2020-09-03:
> It does appear that Matthew has not merged my PR. In the meantime, you can access the fixed version here https://github.com/tkeffer/weewx-wetter

Good job, thanks!

magnum61, 2020-09-03:
> @magnum61 hmm nope? mine is fine... no delay.

Meanwhile mine has decided to show the correct time as well. Only the initial upload was one hour ahead.

jh537, 2020-09-03:
Yes thank you very much, it really works.