weewx/weewx #601: Modernize heat index

tkeffer, 2020-09-05:
Heat index was originally based on a chart from Steadman (1979), but there are now more modern ways of computing it.  See the formula offered by the NWS

https://www.wpc.ncep.noaa.gov/html/heatindex_equation.shtml

PeterQuinn925, 2020-09-06:
This does look easy. Just to make sure I'm not missing anything - It's updating the heatindexF and heatindexC functions in wxfomrulas.py to use the equations in the link rather than what's there now. Right?

tkeffer, 2020-09-06:
That's it!

The link describes a two-step process, which I think is important to follow. If we're going to do it right, let's do it right!

When you're done, make sure you double check the tests for heat index (in `bin/weewx/tests/test_wxformulas.py`). Because the algorithm changed, the results will probably change. Make sure you're getting the correct result.

PeterQuinn925, 2020-09-06:
Ok. Done it. I don't know how to test the tests, but I've updated the code and the test results.

I have no intuitive understanding of heat index. I don't live in an area where it's a meaningful stat. I followed the recipe and tested it and I think it's right.

tkeffer, 2020-09-13:
Fixed in commit https://github.com/weewx/weewx/commit/4a0a9c1b488da5ac952735ddddfc25097aec02f4

tkeffer, 2020-10-23:
I'm a little suspicious of the results. Is it proper for heat index to be less than the sensible temperature? Take today. Outside temperature is a bit over 40, humidity about 80%. Here is today's plot of wind chill and heat index.

[image]

The Wind Chill line is effectively the outside temperature (`outTemp`), so heat index is well below `outTemp` for long periods of time.

If you look at the [NOAA heat index calculator](https://www.wpc.ncep.noaa.gov/html/heatindex.shtml), it shows the same results.  But, the [heat index chart](https://www.weather.gov/safety/heat-index) does not go below 80F.

PeterQuinn925, 2020-10-24:
I guess I didn't do it right.

tkeffer, 2020-10-24:
No, you did good! The implementation is correct. Remember, NOAA's own heat index calculator gives the same results.

I'm just wondering if its appropriate for temperatures below 80°F.

duane-nepa, 2020-10-24:
I don't think it's appropriate for temperatures below 80F. Looking for a reference other than the NOAA graph and memory.

duane-nepa, 2020-10-24:
From Part 1.
[image]

Below 61F.

tkeffer, 2020-10-24:
Good find on Steadman's paper. His final table:

[image]

There are some cells where the apparent temperature is below the dry bulb temperature, so I guess that's to be expected.

His table cuts off at 20°C. We should probably cut off there as well.

tkeffer, 2020-10-24:
Looking behind the scenes, here is the Javascript used by the [NWS heat index calculator](https://www.wpc.ncep.noaa.gov/html/heatindex.shtml):

```javascript
function heat1(HeatIndex){
    if(HeatIndex.RHumidity.value > 100){
        alert("Relative humidity cannot exceed 100%.");
    }
    else if (HeatIndex.RHumidity.value < 0) {
        alert("Relative humidity cannot be less than 0%.");
    }
    else if (HeatIndex.tempair.value <= 40.0) {
        var hi = HeatIndex.tempair.value;
    }
    else {
        var hitemp = 61.0+((HeatIndex.tempair.value-68.0)*1.2)+(HeatIndex.RHumidity.value*0.094);
        var fptemp = parseFloat(HeatIndex.tempair.value);
        var hifinal = 0.5*(fptemp+hitemp);

        if(hifinal > 79.0){
            var hi = -42.379+2.04901523*HeatIndex.tempair.value+10.14333127*HeatIndex.RHumidity.value-0.22475541*HeatIndex.tempair.value*HeatIndex.RHumidity.value-6.83783*(Math.pow(10, -3))*(Math.pow(HeatIndex.tempair.value, 2))-5.481717*(Math.pow(10, -2))*(Math.pow(HeatIndex.RHumidity.value, 2))+1.22874*(Math.pow(10, -3))*(Math.pow(HeatIndex.tempair.value, 2))*HeatIndex.RHumidity.value+8.5282*(Math.pow(10, -4))*HeatIndex.tempair.value*(Math.pow(HeatIndex.RHumidity.value, 2))-1.99*(Math.pow(10, -6))*(Math.pow(HeatIndex.tempair.value, 2))*(Math.pow(HeatIndex.RHumidity.value,2));
            if((HeatIndex.RHumidity.value <= 13) && (HeatIndex.tempair.value >= 80.0) && (HeatIndex.tempair.value <= 112.0)) {
                var adj1 = (13.0-HeatIndex.RHumidity.value)/4.0;
                var adj2 = Math.sqrt((17.0-Math.abs(HeatIndex.tempair.value-95.0))/17.0);
                var adj = adj1 * adj2;
                var hi = hi - adj;
            }
            else if ((HeatIndex.RHumidity.value > 85.0) && (HeatIndex.tempair.value >= 80.0) && (HeatIndex.tempair.value <= 87.0)) {
                var adj1 = (HeatIndex.RHumidity.value-85.0)/10.0;
                var adj2 = (87.0-HeatIndex.tempair.value)/5.0;
                var adj = adj1 * adj2;
                var hi = hi + adj;
            }
        }
        else{
            var hi = hifinal;
        }
    }
```

This is exactly what we are doing. It has a cutoff of 40°F, exactly the same as ours.

So I guess we've got it right.

closed 2020-10-24 by tkeffer

duane-nepa, 2020-10-24:
That was really good work and a great idea.

The thing that lead me to Steadman was the link, [How do we calculate the heat index?](https://www.wpc.ncep.noaa.gov/html/heatindex_equation.shtml), "NWS heat index calculator page". Steadman's work is mentioned a few times and he's also referenced on the wikipedia page. Surprised to see that NWS didn't follow what they said as far as a lower cutoff.

We might need to do [humindex](https://en.wikipedia.org/wiki/Humidex) if there are enough people in Canada using weewx. It's a little different than the heat index. Unless the Canadians are happy with our south of the border heat index as the two weren't harmonized.

Maybe Gary knows what they are using in Europe and if it's different from what the Americans are doing.