weewx/weewx #606: Crash with WMR200 on Weewx 4.1.1

Adri1G, 2020-09-13:
I got issues like this after a long time of running. Weewx is not customized by plugins.
After crash, I restard Weewx and it working normaly... to the next crash (next day to next week).

```log
Sep  4 20:46:31 raspberrypi weewx[713] WARNING weewx.drivers.wmr200: Clock time unsynchronized
Sep  4 20:47:21 raspberrypi weewx[713] INFO weewx.manager: Added record 2020-09-04 20:47:00 CEST (1599245220) to database 'weewx.sdb'
Sep  4 20:47:21 raspberrypi weewx[713] INFO weewx.manager: Added record 2020-09-04 20:47:00 CEST (1599245220) to daily summary in 'weewx.sdb'
Sep  4 20:47:21 raspberrypi weewx[713] WARNING weewx.drivers.wmr200: Skipped bytes before resync:9
Sep  4 20:47:21 raspberrypi weewx[713] ERROR weewx.drivers.wmr200: Checksum miscompare act:0x0000 exp:0x02d3 packet:d3 02 
Sep  4 20:47:21 raspberrypi weewx[713] INFO weewx.engine: Main loop exiting. Shutting engine down.
Sep  4 20:47:21 raspberrypi weewx[713] INFO weewx.engine: Shutting down StdReport thread
Sep  4 20:47:25 raspberrypi weewx[713] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 3.80 seconds
Sep  4 20:47:28 raspberrypi weewx[713] INFO weewx.imagegenerator: Generated 15 images for report SeasonsReport in 2.96 seconds
Sep  4 20:47:28 raspberrypi weewx[713] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx
Sep  4 20:47:28 raspberrypi weewx[713] INFO weewx.drivers.wmr200: USB polling device thread exiting
Sep  4 20:47:28 raspberrypi weewx[713] INFO weewx.drivers.wmr200: USB polling thread expired
Sep  4 20:47:28 raspberrypi weewx[713] CRITICAL __main__: Caught WeeWxIOError: Checksum miscompare act:0x0000 exp:0x02d3
Sep  4 20:47:28 raspberrypi weewx[713] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...
Sep  4 20:48:28 raspberrypi weewx[713] INFO __main__: retrying...
Sep  4 20:48:28 raspberrypi weewx[713] INFO __main__: Using configuration file /etc/weewx/weewx.conf
Sep  4 20:48:28 raspberrypi weewx[713] INFO __main__: Debug is 0
Sep  4 20:48:28 raspberrypi weewx[713] INFO weewx.engine: Loading station type WMR200 (weewx.drivers.wmr200)
Sep  4 20:48:28 raspberrypi weewx[713] INFO weewx.drivers.wmr200: driver version is 3.5.0
Sep  4 20:48:28 raspberrypi weewx[713] INFO weewx.drivers.wmr200: sensor map is {'altimeter': 'altimeter', 'pressure': 'pressure', 'windSpeed': 'wind_speed', 'windDir': 'wind_dir', 'windGust': 'wind_gust', 'windBatteryStatus': 'battery_status_wind', 'inTemp': 'temperature_0', 'outTemp': 'temperature_1', 'extraTemp1': 'temperature_2', 'extraTemp2': 'temperature_3', 'extraTemp3': 'temperature_4', 'extraTemp4': 'temperature_5', 'extraTemp5': 'temperature_6', 'extraTemp6': 'temperature_7', 'extraTemp7': 'temperature_8', 'inHumidity': 'humidity_0', 'outHumidity': 'humidity_1', 'extraHumid1': 'humidity_2', 'extraHumid2': 'humidity_3', 'extraHumid3': 'humidity_4', 'extraHumid4': 'humidity_5', 'extraHumid5': 'humidity_6', 'extraHumid6': 'humidity_7', 'extraHumid7': 'humidity_8', 'inHeatindex': 'heatindex_0', 'heatindex': 'heatindex_1', 'heatindex1': 'heatindex_2', 'heatindex2': 'heatindex_3', 'heatindex3': 'heatindex_4', 'heatindex4': 'heatindex_5', 'heatindex5': 'heatindex_6', 'heatindex6': 'heatindex_7', 'heatindex7': 'heatindex_8', 'outTempBatteryStatus': 'battery_status_out', 'rain': 'rain', 'rainTotal': 'rain_total', 'rainRate': 'rain_rate', 'hourRain': 'rain_hour', 'rain24': 'rain_24', 'rainBatteryStatus': 'battery_status_rain', 'UV': 'uv', 'uvBatteryStatus': 'battery_status_uv', 'windchill': 'windchill', 'forecastIcon': 'forecast_icon', 'outTempFault': 'out_fault', 'windFault': 'wind_fault', 'uvFault': 'uv_fault', 'rainFault': 'rain_fault', 'clockUnsynchronized': 'clock_unsynchronized'}
Sep  4 20:48:28 raspberrypi weewx[713] CRITICAL weewx.drivers.wmr200: open_device() Unable to claim USB interface. Reason: [Errno 16] Resource busy
Sep  4 20:48:28 raspberrypi weewx[713] ERROR weewx.engine: Import of driver failed: [Errno 16] Resource busy (<class 'weewx.WakeupError'>)
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 193, in open_device
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      self.handle.claimInterface(self.interface)
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/legacy.py", line 230, in claimInterface
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      util.claim_interface(self.dev, interface)
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/util.py", line 205, in claim_interface
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      device._ctx.managed_claim_interface(device, interface)
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/core.py", line 102, in wrapper
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      return f(self, *args, **kwargs)
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/core.py", line 167, in managed_claim_interface
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      self.backend.claim_interface(self.handle, i)
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 811, in claim_interface
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 595, in _check
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      raise USBError(_strerror(ret), ret, _libusb_errno[ret])
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****  usb.core.USBError: [Errno 16] Resource busy
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****  
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****  During handling of the above exception, another exception occurred:
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****  
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/engine.py", line 103, in setupStation
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      self.console = loader_function(config_dict, self)
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 58, in loader
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      return WMR200(**config_dict[DRIVER_NAME])
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 1466, in __init__
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      self.usb_device.open_device()
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 196, in open_device
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****      raise weewx.WakeupError(exception)
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL weewx.engine:     ****  weewx.WakeupError: [Errno 16] Resource busy
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL __main__: Unable to load driver: [Errno 16] Resource busy
Sep  4 20:48:29 raspberrypi weewx[713] CRITICAL __main__:     ****  Exiting...
```

tkeffer, 2020-09-13:
Could you try setting the [`loop_on_init` option](http://www.weewx.com/docs/usersguide.htm#General)? That would at least give weewxd a chance to restart.

tkeffer, 2020-09-16:
@Adri1G ?

Adri1G, 2020-09-16:
> weewxd

Ok, thank you! :-)
I currently do not have access to my server. I'll let you know the results when I implement your suggestion. Maybe in a month's time.

Adri1G, 2020-09-22:
@tkeffer I changed like has you said, but the crash remains. I had to restart weewx with `sudo /etc/init.d/weewx restart`

tkeffer, 2020-09-22:
`loop_on_init` will not fix the crash, but it does cause `weewxd` to keep trying, rather than exit after the first failure.

So, the question is: does `weewxd` now recover from the error?

Adri1G, 2020-09-22:
Yes I understood that it is not a fix. And no, Weewx do not recover.  I have gotten this log:

```log
Sep 22 00:01:21 raspberrypi weewx[5556] INFO __main__: retrying...
Sep 22 00:01:21 raspberrypi weewx[5556] INFO __main__: Using configuration file /etc/weewx/weewx.conf
Sep 22 00:01:21 raspberrypi weewx[5556] INFO __main__: Debug is 0
Sep 22 00:01:21 raspberrypi weewx[5556] INFO weewx.engine: Loading station type WMR200 (weewx.drivers.wmr200)
Sep 22 00:01:21 raspberrypi weewx[5556] INFO weewx.drivers.wmr200: driver version is 3.5.0
Sep 22 00:01:21 raspberrypi weewx[5556] INFO weewx.drivers.wmr200: sensor map is {'altimeter': 'altimeter', 'pressure': 'pressure', 'windSpeed': 'wind_speed', 'windDir': 'wind_dir', 'windGust': 'wind_gust', 'windBatteryStatus': 'battery_status_wind', 'inTemp': 'temperature_0', 'outTemp': 'temperature_1', 'extraTemp1': 'temperature_2', 'extraTemp2': 'temperature_3', 'extraTemp3': 'temperature_4', 'extraTemp4': 'temperature_5', 'extraTemp5': 'temperature_6', 'extraTemp6': 'temperature_7', 'extraTemp7': 'temperature_8', 'inHumidity': 'humidity_0', 'outHumidity': 'humidity_1', 'extraHumid1': 'humidity_2', 'extraHumid2': 'humidity_3', 'extraHumid3': 'humidity_4', 'extraHumid4': 'humidity_5', 'extraHumid5': 'humidity_6', 'extraHumid6': 'humidity_7', 'extraHumid7': 'humidity_8', 'inHeatindex': 'heatindex_0', 'heatindex': 'heatindex_1', 'heatindex1': 'heatindex_2', 'heatindex2': 'heatindex_3', 'heatindex3': 'heatindex_4', 'heatindex4': 'heatindex_5', 'heatindex5': 'heatindex_6', 'heatindex6': 'heatindex_7', 'heatindex7': 'heatindex_8', 'outTempBatteryStatus': 'battery_status_out', 'rain': 'rain', 'rainTotal': 'rain_total', 'rainRate': 'rain_rate', 'hourRain': 'rain_hour', 'rain24': 'rain_24', 'rainBatteryStatus': 'battery_status_rain', 'UV': 'uv', 'uvBatteryStatus': 'battery_status_uv', 'windchill': 'windchill', 'forecastIcon': 'forecast_icon', 'outTempFault': 'out_fault', 'windFault': 'wind_fault', 'uvFault': 'uv_fault', 'rainFault': 'rain_fault', 'clockUnsynchronized': 'clock_unsynchronized'}
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL weewx.drivers.wmr200: Unable to find device with VendorID=0fde ProductID=ca01
Sep 22 00:01:21 raspberrypi weewx[5556] ERROR weewx.engine: Import of driver failed: Unable to find USB device (<class 'weewx.WeeWxIOError'>)
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/engine.py", line 103, in setupStation
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL weewx.engine:     ****      self.console = loader_function(config_dict, self)
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 58, in loader
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL weewx.engine:     ****      return WMR200(**config_dict[DRIVER_NAME])
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 1463, in __init__
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL weewx.engine:     ****      raise weewx.WeeWxIOError("Unable to find USB device")
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL weewx.engine:     ****  weewx.WeeWxIOError: Unable to find USB device
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL __main__: Unable to load driver: Unable to find USB device
Sep 22 00:01:21 raspberrypi weewx[5556] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...
```

tkeffer, 2020-09-22:
That's a different error than before. It looks like the driver is unable to locate the device on the bus at all.

What do you get if you use the `lsusb` command?

Adri1G, 2020-09-23:
> 
> 
> That's a different error than before. It looks like the driver is unable to locate the device on the bus at all.
> 
> What do you get if you use the `lsusb` command?

```
Bus 001 Device 005: ID 0fde:ca01 Oregon Scientific WMRS200 weather station
Bus 001 Device 004: ID 0781:5571 SanDisk Corp. Cruzer Fit
Bus 001 Device 006: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

tkeffer, 2020-09-23:
Looks like the device is there. Wish I had an explanation for why the driver is unable to find it, but I don't.

Adri1G, 2020-09-23:
I do not touch the hardware. This crash occurs occasionnaly, from few day to few week. I have to restart all Weewx. I do not investigate more, it is maybe a kind of overflow in Weewx, or an error on my part in the configuration of my raspberry (unlikely, there is only weewx to use usb, the rest is network.).

Adri1G, 2020-09-30:
When I execute ``` sudo /etc/init.d/weewx restart```, I got this:

```log
Sep 30 08:46:10 raspberrypi weewx[20159]: Stopping weewx weather system: weewx........................ failed!
Sep 30 08:46:10 raspberrypi systemd[1]: weewx.service: Control process exited, code=exited, status=1/FAILURE
Sep 30 08:46:10 raspberrypi systemd[1]: weewx.service: Failed with result 'exit-code'.
Sep 30 08:46:10 raspberrypi systemd[1]: Stopped LSB: weewx weather system.
Sep 30 08:46:10 raspberrypi systemd[1]: weewx.service: Found left-over process 24878 (python3) in control group while starting unit. Ignoring.
Sep 30 08:46:10 raspberrypi systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Sep 30 08:46:10 raspberrypi systemd[1]: Starting LSB: weewx weather system...
Sep 30 08:46:10 raspberrypi weewx[20410]: Starting weewx weather system: weewx already running....
Sep 30 08:46:10 raspberrypi systemd[1]: Started LSB: weewx weather system.
```

I will reboot my raspberry.

closed 2020-10-02 by tkeffer