weewx/weewx #609: Label on right side of graphs

brentfraser, 2020-09-26:
I find it inconvenient that the graph's y-axis labels are on the left but the latest data is on the far right side, so my eye has to follow the grid lines over to the left to estimate the values.   A simple solution is put labels on the right side as well, like so:

[image]

I've changed `genplot.py` to widen the right margin:
```
        self.rmargin = int(4.0 * self.axis_label_font_size)
```

and to label the right y-axis:

```
                sdraw.draw.text( (self.lmargin - axis_label_size[0] - 2, ypos - axis_label_size[1]/2),
                                ylabel, fill=self.axis_label_font_color, font=axis_label_font)
                # label right 
                sdraw.draw.text( (self.image_width - self.rmargin + 4, ypos - axis_label_size[1]/2),
                                ylabel, fill=self.axis_label_font_color, font=axis_label_font)
```

But I suppose, in preparation of a pull request, I should add code to make it optional.  As a newbie, I'm not sure the best way to do that.  An option in the skin.conf like:

`  y_label = left, right`

with the default `= left`

Opinions?

richardh9935, 2020-09-26:
Good idea. Australian Bureau of Statistics swapped to the right-sided Y-axis in the 1980's for exactly that reason. 
Richard.

tkeffer, 2020-09-26:
That seems reasonable: `left`, `right`, or `both`

brentfraser, 2020-09-27:
I had to add a little bit of code to move the units label
[image]

And change the parameter name:

`		y_label_side = both
`

closed 2020-09-28 by tkeffer