weewx/weewx #613: Not all new types in the extended schema are in obs_group_dict.

tkeffer, 2020-10-12:
`snowDepth`, `snowMoisture`, and `snowRate`, and others.

bellrichm, 2020-10-12:
I’d be happy to take this on. My first step will be to do a diff to find out if there are any others.
Would it be ok to do a first commit of sorting obs_group_dict. This would make it easier to check in the future. Then a second commit with the additions, so it is easy to see the actual additions.
rich

tkeffer, 2020-10-12:
Yes, that's fine.

tkeffer, 2020-10-24:
Here are the types that I found missing, and my candidate for which group they should belong to.
```
"co" : "group_concentration"
"co2" : "group_concentration"
"lightning_energy" : "group_energy"
"luminosity" :
"nh3" : "group_concentration"
"no2" : "group_concentration"
"noise" : 
"o3" : "group_concentration"
"pb" : "group_concentration"
"snowDepth" : "group_rain"
"snowMoisture" : "group_percent"
"snowRate" : "group_rainrate"
"so2" : "group_concentration"
```
I've ignored all of the "status" types because they are usually an index into an unknown table.

I'm not sure what group `luminosity` should belong to. Perhaps a new group, `group_luminosity`?

The type `noise` should probably belong to a new group, `group_db`.

Some of these are problematic. For example, `group_concentration` is usually in micrograms per meter cubed. But, that doesn't make sense for something like CO2 or O3. Perhaps we need something measured in ppm?

Similar issue with `lightning_energy`. I'm not sure exactly what it is, but it's likely to be measured in megaJoules, or even gigaJoules. Certainly not watt seconds!

tkeffer, 2020-10-24:
Propose adding the following groups, and their associated unit

```
group_db : dB
group_fraction: ppm
group_bigenergy: MJ
```

tkeffer, 2020-10-24:
Doing a bit of research, a lightning strike has roughly 1 GJ of energy, which is 277778 Watt-hours. So, storing that energy in units of Watt-hours is not unreasonable. For plotting purposes, we probably want conversions from Watt-hours to kW-hours and MJ.

So, `group_bigenergy` is not necessary.

tkeffer, 2020-10-24:
Fixed in commit https://github.com/weewx/weewx/commit/2efa6b73982624722b60c7a4ea568779e71ccf92

tkeffer, 2020-10-25:
In postscript, here's what I went with.

__New types__

```
"co" : "group_concentration"
"co2" : "group_concentration"
"luminosity" : "group_illuminence"
"nh3" : "group_concentration"
"no2" : "group_concentration"
"noise" : "group_db"
"o3" : "group_concentration"
"pb" : "group_concentration"
"snowDepth" : "group_rain"
"snowMoisture" : "group_percent"
"snowRate" : "group_rainrate"
"so2" : "group_concentration"
```

The type `lightning_energy` is treated as dimensionless, as it is a creation by the as3935 sensor.

__New groups__
```
group_db : dB
group_fraction : ppm
group_illuminence : lux
```

__New units__
```
dB
ppm
lux
mega_joule
```

gjr80, 2020-10-25:
‘illuminence’ or ‘illuminance’, I thought it was the latter?

tkeffer, 2020-10-25:
It is the latter. Thanks, Gary.

I am concerned that the schema contains `luminosity`, when what the sensors measure is `illuminance`. In fact, I don't think any of the sensors measure `luminosity`. The documentation is also sloppy, and uses `luminosity` and `illuminance` interchangeably. 

It's too late to change the schema, but not too late to change the units. We should probably have *both* a `group_luminosity` *and* a `group_illuminance`.

tkeffer, 2020-10-25:
I should add that with the introduction of the observation type `illuminance`, measured in `lux`, we have an opportunity to add it to the fousb sensor map.

duane-nepa, 2020-10-26:
I was wondering where the documentation for this is? I'd love to see it. It would help me better understand.

Have you seen this page, https://madis.ncep.noaa.gov/sfc_mesonet_variable_list.shtml? and this one, https://madis.ncep.noaa.gov/snow_variable_list.shtml?

I agree and doubt any user is measuring luminosity unless they are working in an astronomical observatory or are very advanced amature astronomers, which is possible. There is a good wiki article on luminosity and how it's measured. Most of us measure illuminance if we have that sensor.

I'm considering trying to build my own road temperature and road snow depth sensors over the winter. I haven't started researching it yet and will see if someone already made one first. In which case I'll make one or buy it if not too expensive.

duane-nepa, 2020-10-26:
This is from the wiki page https://en.wikipedia.org/wiki/Lumen_(unit)#SI_photometry_units. It's got all the units. I didn't try to trace back to the original sources.

[image]

closed 2020-10-29 by tkeffer