weewx/weewx #614: Null winddir causes misleading "Unavailable" status in wunderground

proxygoat, 2020-10-12:
When windSpeed is 0, weewx sends a null winddir value to wunderground. According to wunderground support, this causes the "Wind" display in "PWS Current Conditions" to show "Unavailable." This is misleading. When an anemometer is present and working and windSpeed is 0, I think the transmitted winddir should remain at the value it had at the last previous non-zero windSpeed; the null value should be transmitted only when an anemometer is actually electrically unavailable.

tkeffer, 2020-10-12:
Can you give me an example of where this has happened?

In any case, you can retain wind direction by using option [`ignore_zero_wind`](http://www.weewx.com/docs/usersguide.htm#StdWXCalculate)

proxygoat, 2020-10-13:
Here's what wunderground support said:
    
      Hi, Phil!
        
        This is what your station is sending in:
        
        
            "observations": [
                {
                    "stationID": "KWASEQUI26",
                    "obsTimeUtc": "2020-10-12T21:30:00Z",
                    "obsTimeLocal": "2020-10-12 14:30:00",
                    "neighborhood": "Blyn/Gardiner",
                    "softwareType": "weewx-3.9.2",
                    "country": "US",
                    "solarRadiation": null,
                    "lon": -122.974495,
                    "realtimeFrequency": null,
                    "epoch": 1602538200,
                    "lat": 48.041691,
                    "uv": null,
                    "winddir": null,
                    "humidity": 73,
                    "qcStatus": 1,
                    "imperial": {
                        "temp": 57,
                        "heatIndex": 57,
                        "dewpt": 49,
                        "windChill": 57,
                        "windSpeed": 0,
                        "windGust": 5,
                        "pressure": 30.19,
                        "precipRate": 0.00,
                        "precipTotal": 0.01,
                        "elev": 247
        
        As you can see, wind direction is not 0, not 360, but null. 
        This is what is causing the "unavailable" message.  
        
        You might want to see if the sending software can be tweaked to
        send in something for the direction when it's calm.  You
        already have a windspeed of zero coming in.  I think even if
        it's zero, you will see a difference.
        
        Let me know if you have any questions.  Thank you for being part
        of the Weather Underground community!
        
        --Victoria
    
    I changed my configuration per your suggestion. Unfortunately,
      we're in the midst of a typical fall Pacific Northwest windstorm
      right now, so it'll be a while before I have a verdict on that.
    
    On 10/12/20 4:03 PM, Tom Keffer wrote:
    
    
      
      Can you give me an example of where this has happened?
      In any case, you can retain wind direction by using option ignore_zero_wind
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/weewx/weewx/issues/614#issuecomment-707387540",
"url": "https://github.com/weewx/weewx/issues/614#issuecomment-707387540",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]
    
      What was that
        about a rising tide lifting all boats? What if you don't have a
        boat?
           — Eugene Robinson

proxygoat, 2020-10-13:
1. I don't know. I'd guess it's what they generate from weewx
      input and hand over to their GUI.
    2. I modded my config file and restarted weewx last night, so
      ignore_zero_wind was False at 0930 this morning. 
    
    On 10/13/20 10:01 AM, Tom Keffer wrote:
    
    
      
      Some questions:
      
        This looks like some kind of JSON data structure, rather
          than what is actually sent to the WU by WeeWX. Do you know
          where it came from?
        The question is not whether a null value is sent, but what
          the WU's reaction to it is. Do you have an example of that? My
          station regularly sends null values for wind direction with no
          ill effect. I also see that today you have a case where windSpeed
          is zero (at 0930) and it seems to have tolerated it.
        In any case, it sounds like they think the appropriate
          reaction is to claim the station is unavailable. Typical WU
          response: rather than fix their software, they blame the
          poster! Nevertheless, if you want to do it their way, try the
          option ignore_zero_wind.
      
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/weewx/weewx/issues/614#issuecomment-707880677",
"url": "https://github.com/weewx/weewx/issues/614#issuecomment-707880677",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]
    
      What was that
        about a rising tide lifting all boats? What if you don't have a
        boat?
           — Eugene Robinson

tkeffer, 2020-10-13:
Again, do you have an example of where the WU went offline for you when wind direction was zero? A screenshot, cut-and-paste, link,... anything will do.

tkeffer, 2020-10-18:
Anything new on this? Do you have an example?

duane-nepa, 2020-10-18:
I think it depends on how long the wind is at 0 speed. I've seen it many times. I can try to get a better example than the one in the pic if you want.

But you are correct, when there is no wind speed there is no direction.

[image]

tkeffer, 2020-10-18:
Setting `ignore_zero_wind` will cause weewx to report whatever the hardware is reporting. 

To send the last value would require some work in `restx.py`. Using cached values is not something we like to do because it
complicates code, and it is "making up" data. 

This is far from the first irritating bug in the WU.

In any case, there is a work around.

duane-nepa, 2020-10-18:
You might need to reopen this. But maybe it's something I did.

I figured you'd still get this. I edited weewx.conf
[image]

Restarted. Had to go out for a couple of hours and when I came back i had the same unavailable on WU.

So I set debug = 2 and log_success = True. Restarted.

[image]

"Oct 18 18:44:15 weewx weewx[11705] DEBUG weewx.restx: Ambient: url: https://weatherstation.wunderground.com/weatherstation/updateweatherstation.php?action=updateraw&ID=KPASTROU8&PASSWORD=XXX&softwaretype=weewx-4.1.1&baromin=30.252&dateutc=2020-10-18%2022%3A44%3A00&dailyrainin=0.00&dewptf=37.8&rainin=0.00&leafwetness=000&humidity=055&tempf=53.6&solarradiation=0.00&soilmoisture=001&soiltempf=41.0&UV=0.00&windgustmph=0.0&windspeedmph=0.0"

I'm not seeing a 'windDir=' in that string. Also it's not reporting a direction in the windDir column in the db.  The last windDir value I have in the db corresponds to the value that shows on WU.

Maybe I didn't do this correctly?

tkeffer, 2020-10-19:
Perhaps the name `ignore_zero_wind` isn't the best. It's supposed to mean that when it's true, do not enforce setting wind direction to null when wind speed is zero. That is, ignore the instances of zero wind speed.

So, you want to set it to `True`, not `False`.

duane-nepa, 2020-10-19:
Ok.. I'll change it and let you know. But I thought I followed the user's guide.

duane-nepa, 2020-10-19:
Still not in there. Do I have to wait for the wind to have speed before I'll see it? Maybe that's the issue??? If so I'll wait for a puff.

duane-nepa, 2020-10-19:
I had a puff. While there was wind speed it sent the direction, but when there wasn't it didn't. Does capitalization matter with this one, I think I used a "T" and an "F" but not certain? Doesn't seem to matter elsewhere reason I'm asking. That was with ignore_zero_wind set to both True and False. Do you want log files? or snips from them?

tkeffer, 2020-10-19:
My apologies. You're right! There are so many damn options, even I lose track of them!

You want `ignore_zero_wind` to be `False`.

As for why it didn't work before... Can you run weewxd directly from the command line? It will show you all LOOP packets and archive records. What value do they show for `windDir`?

Definitely prefer log file snippets. Screen shots are hard to read, waste space, and don't give you much information.

tkeffer, 2020-10-19:
Capitalization does not matter. `FALSE`, `False`, `false`, and even `no` or `0` all evaluate to false.

duane-nepa, 2020-10-19:
I thought so. Just wanted to be certain. I don't know how to bet the gray background or I'd use it. I'm going to assume TRUE, True, true, yes, or 1 all evaluate to true as well going forward?

It's erratic at best.

This is a good time to test it here. Because the night time inversion I often get is starting, I get intermittent flows of cold air off the mountain. Sort of like just before the sea breeze starts with an offshore prevailing wind like we see at the Jersey shore all the time.

duane-nepa, 2020-10-19:
Ok.. I'm setting to False. I'll run from command line and report. I have an ssh connection. So easy to do.

duane-nepa, 2020-10-19:
I'll send you a file of the output. just restarted it and sending to a file now.

duane-nepa, 2020-10-19:
Here you are. I had debug set to 0, log_success = false. If you want other settings let me know.

[weewxdouput.txt](https://github.com/weewx/weewx/files/5398701/weewxdouput.txt)

gjr80, 2020-10-19:
I have a theory. You wouldn’t happen to be using software record generation (ie `record_generation = software` under `[StdArchive]` in `weewx.conf`) would you? If so I believe the process used to extract wind data from the accumulator can effectively negate the effect of `ignore_zero_wind`when an archive record is extracted from the accumulator and all `windSpeed` values in the accumulator are 0.
If this is the case, and my theory is correct, a simple change to hardware record generation (with ‘ignore_zero_wind = false`) would solve the problem.

tkeffer, 2020-10-19:
Even better, how about the full system log with debug=2, from startup, through the first record generation?

duane-nepa, 2020-10-19:
Give me some time, I'll get to it before I go to bed as I have a few things in the hopper for today.

Let me report this first so I know if I have to do on this. Yesterday when I had debug=2 and was tailing I copied the syslog but I don't know what's what in it to just get it from there.

I have record_generation = hardware, never changed it. Under both settings for ignore_zero_wind when all wind speeds went to 0 it did not send winddir to WU.

I don't know how to manually rotate the syslog, I'll do a quick search and see if I can figure it out. If I can't I'll just extract the parts of the log you need for each case.

I'll do this if you tell me to proceed. Set debug = 2 and use both settings for record_generation = [hardware, software] and for ignore_zero_wind = [true, false]. So these are the cases I'll test and provide logs for:

- [ ] record_generation = hardware,  ignore_zero_wind = true

- [ ] record_generation = hardware,  ignore_zero_wind = false  

- [ ] record_generation = software,  ignore_zero_wind = true

- [ ] record_generation = software,  ignore_zero_wind = false

Will that do it? Or something else you might want?

gjr80, 2020-10-19:
Don’t worry about record_generation, leave it as hardware.

duane-nepa, 2020-10-19:
ok.. here you are.
[syslog record_generation = hardware and ignore_zero_wind=False.txt](https://github.com/weewx/weewx/files/5403791/syslog.record_generation.hardware.and.ignore_zero_wind.False.txt)

tkeffer, 2020-10-19:
Testing zero wind speeds is not easy for me to do because I live in a very windy place (the Columbia River Gorge). I tied the wind cups in my anemometer so they wouldn't turn, then watched it for a few minutes.

It turns out that the VP2 emits records with null wind direction when the wind speed is zero. LOOP packets still have non-null values, but the archive records have null values.

It does not help that you are using a fairly short archive interval. If there is any non-zero wind speed in the archive interval, the hardware will emit something for wind direction, but if there is none, then you get a null wind direction. With a short interval, odds increase that you might not see any wind during that period.

You have two choices:
1. Modify `restx.py` so it caches the last non-None value, then uses that when there is no wind speed.
2. Live with it.

duane-nepa, 2020-10-19:
Here is the problem I have.

1.  I agree wind is a vector. No wind, no direction.
2. WU doesn't mark the station for data quality when we intermittently report wind. This causes the station to be demoted when displayed on the WU map. Not all that big a deal to me, but some of the local TV stations have shown my station during reports in the past. Don't get that if I'm not on the map.

I'd rather only fudge what goes to WU. Guessing weewx.restx.StdWunderground?

I also wonder what would happen if I just reported winddir=null when wind speed is 0? Would that would satisfy them. I'll try to find the WU documentation and see what's allowed in winddir.

If this is open again I'll report back here? Should be by Oct. 23-26. I also have a wiki page to write. Got a placeholder done for that. I'd welcome input as I write that as well. I'm going to benchmark SQLite on a few systems with different OS's and memory and try to repeat for MySQL as well. So it'll be more comprehensive than what I've done to date. But I'll use the same queries.

tkeffer, 2020-10-19:
Try this version of `restx.py`: https://raw.githubusercontent.com/weewx/weewx/development/bin/weewx/restx.py

It includes an option `force_direction` that will cache the last non-None wind direction and use that instead.

```ini
...
  [[Wunderground]]
    force_direction = true
```

duane-nepa, 2020-10-20:
Thanks so much. I think that did it.

I renamed it to restx_forcedir.py and changed the weewx.conf file to make it work. 

[image]

tkeffer, 2020-10-20:
To appear in v4.2

closed 2020-10-20 by tkeffer

duane-nepa, 2020-10-21:
I'm sorry for being a pia.

I found one more condition today that we need to account for.

I stopped weewx. I started weewx. I'm in the middle of a calm period that will last for a while. 

In restx.py
self.last_direction = none

Still not reporting winddir to WU and it doesn't take wind out of unavailable. I read what I could find on the format for this. And of course it says you don't have to report everything.

It could be worse. I think just initializing self.last_direction = 0 cures this problem. I'll try it and let you know.

duane-nepa, 2020-10-21:
Ok. All good now.

tkeffer, 2020-10-21:
Changed in commit https://github.com/weewx/weewx/commit/7a522d8707add6e373374f0cdc0d17fa41ccbb41.

Kills me that the WU is forcing us to make up data.

duane-nepa, 2020-10-21:
I totally agree. It seems they had people with limited experience in the physical sciences and probably math write this. ie. they don't understand a vector with no scalar can't have a direction because it's a point. 

Could we try sending them null and see what happens? i.e. winddir=null? i'll see if if I can do it if you want.

tkeffer, 2020-10-21:
AFAIK, they don't have a way of signaling "null". Instead, you leave the value off the URL.

Here's the protocol: https://support.weather.com/s/article/PWS-Upload-Protocol?language=en_US

duane-nepa, 2020-10-21:
That's what I read last night. I'll try to send them a message and see what they say regarding zero wind. I'll let you know. Since we have it working I think this can wait till 4.3.

duane-nepa, 2020-10-21:
Opps setting self.last_direction = 0 causes a type error.

I guess it's "N" instead.

duane-nepa, 2020-10-21:
I don't know what I did. 

duane@weewx:/usr/share/weewx/weewx$ sudo systemctl status weewx
● weewx.service - LSB: weewx weather system
     Loaded: loaded (/etc/init.d/weewx; generated)
     Active: active (exited) since Wed 2020-10-21 10:43:36 EDT; 25min ago
       Docs: man:systemd-sysv-generator(8)
    Process: 3480 ExecStart=/etc/init.d/weewx start (code=exited, status=0/SUCCESS)

Oct 21 10:43:36 weewx python3[3496]: weewx[3496] CRITICAL __main__:     ****    File "/usr/share/weewx/weewxd", line 148, in main
Oct 21 10:43:36 weewx python3[3496]: weewx[3496] CRITICAL __main__:     ****      engine = weewx.engine.StdEngine(config_dict)
Oct 21 10:43:36 weewx python3[3496]: weewx[3496] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 75, in __init__
Oct 21 10:43:36 weewx python3[3496]: weewx[3496] CRITICAL __main__:     ****      self.loadServices(config_dict)
Oct 21 10:43:36 weewx python3[3496]: weewx[3496] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 138, in loadServices
Oct 21 10:43:36 weewx python3[3496]: weewx[3496] CRITICAL __main__:     ****      obj = weeutil.weeutil.get_object(svc)(self,config_dict)
Oct 21 10:43:36 weewx python3[3496]: weewx[3496] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/restx_forcedir.py", line 638, in __init__
Oct 21 10:43:36 weewx python3[3496]: weewx[3496] CRITICAL __main__:     ****      self.loop_thread = AmbientLoopThread(
Oct 21 10:43:36 weewx python3[3496]: weewx[3496] CRITICAL __main__:     ****  TypeError: __init__() got an unexpected keyword argument 'force_direction'
Oct 21 10:43:36 weewx python3[3496]: weewx[3496] CRITICAL __main__:     ****  Exiting.

I replaced restx.py again from the link and it's still doing this yet it was working before.

duane-nepa, 2020-10-21:
I didn't stop weewx. I'll try setting self.last_direction again after I get back tonight.

duane-nepa, 2020-10-21:
We'll see what happens. my message...

Hi,
We are working on an issue with "Wind" showing "UNAVAILABLE" when it's calm, both "windspeedmph" and "windgustmph" are 0.
I'm sure you're aware that "winddir" and "windgustdir" would become undefined or null when there is no speed as they are the direction of a vector. Vectors with zero scalers by definition don't have a direction, they are points.
What do we do so that your software will not put wind as "UNAVAILABLE" in this case? 
We have also noticed that this affects your QC routines and demotes the station on the map and nearby station lists.
Thank you in advance for your quick reply.
VR,
-dk

[image]

duane-nepa, 2020-10-22:
Seems we broke it again and with force_direction = true it's not sending wind direction again. I just replaced the file. 

It works when rapidfire is set to false. But when rapidfire = true you get this errors. I'll get them tomorrow.

duane-nepa, 2020-10-22:
I got the errors for you. rapidfire = true, force_direction = true. It will run when rapidfire = false It's also not forcing the direction as best I can tell.

Oct 22 15:11:08 weewx systemd[1]: Starting LSB: weewx weather system...
Oct 22 15:11:08 weewx weewx[6579]:  * Starting weewx weather system weewx
Oct 22 15:11:08 weewx weewx[6591] INFO __main__: Initializing weewx version 4.1.1
Oct 22 15:11:08 weewx weewx[6591] INFO __main__: Using Python 3.8.5 (default, Jul 28 2020, 12:59:40) #012[GCC 9.3.0]
Oct 22 15:11:08 weewx weewx[6591] INFO __main__: Platform Linux-5.4.0-1022-raspi-aarch64-with-glibc2.29
Oct 22 15:11:08 weewx weewx[6591] INFO __main__: Locale is 'C.UTF-8'
Oct 22 15:11:08 weewx weewx[6591] INFO __main__: PID file is /var/run/weewx.pid
Oct 22 15:11:08 weewx weewx[6594] INFO __main__: Using configuration file /etc/weewx/weewx.conf
Oct 22 15:11:08 weewx weewx[6594] INFO __main__: Debug is 0
Oct 22 15:11:08 weewx weewx[6594] INFO weewx.engine: Loading station type Vantage (weewx.drivers.vantage)
Oct 22 15:11:08 weewx weewx[6579]:    ...done.
Oct 22 15:11:08 weewx systemd[1]: Started LSB: weewx weather system.
Oct 22 15:11:08 weewx /weewxd: TED Pro: probing http://192.168.0.20:80/api/DashData.xml?T=0&D=0&M=1
Oct 22 15:11:08 weewx weewx[6594] INFO weewx.engine: StdConvert target unit is 0x1
Oct 22 15:11:08 weewx weewx[6594] INFO weewx.wxservices: The following values will be calculated: pressure=prefer_hardware, barometer=prefer_hardware, altimeter=prefer_hardware, windchill=prefer_hardware, heatindex=prefer_hardware, dewpoint=prefer_hardware, inDewpoint=prefer_hardware, rainRate=prefer_hardware, maxSolarRad=prefer_hardware, cloudbase=prefer_hardware, humidex=prefer_hardware, appTemp=prefer_hardware, ET=prefer_hardware, windrun=prefer_hardware
Oct 22 15:11:08 weewx weewx[6594] INFO weewx.wxservices: The following algorithms will be used for calculations: altimeter=aaASOS, maxSolarRad=RS
Oct 22 15:11:08 weewx weewx[6594] INFO weewx.engine: Archive will use data binding wx_binding
Oct 22 15:11:08 weewx weewx[6594] INFO weewx.engine: Record generation will be attempted in 'hardware'
Oct 22 15:11:08 weewx weewx[6594] INFO weewx.engine: Using archive interval of 60 seconds (specified by hardware)
Oct 22 15:11:08 weewx weewx[6594] INFO weewx.restx_forcedir: StationRegistry: Station will be registered.
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__: Caught unrecoverable exception:
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****  __init__() got an unexpected keyword argument 'force_direction'
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****  Traceback (most recent call last):
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****    File "/usr/share/weewx/weewxd", line 148, in main
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****      engine = weewx.engine.StdEngine(config_dict)
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 75, in __init__
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****      self.loadServices(config_dict)
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 138, in loadServices
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****      obj = weeutil.weeutil.get_object(svc)(self,config_dict)
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/restx_forcedir.py", line 638, in __init__
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****      self.loop_thread = AmbientLoopThread(
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****  TypeError: __init__() got an unexpected keyword argument 'force_direction'
Oct 22 15:11:08 weewx weewx[6594] CRITICAL __main__:     ****  Exiting.

duane-nepa, 2020-10-23:
I fixed the rapidfire section. put the date 2020.10.22 in a comment near the changes.

http://duanek.name/restx/restx_forcedir.txt

tkeffer, 2020-10-23:
Try this version: https://raw.githubusercontent.com/weewx/weewx/development/bin/weewx/restx.py

I pretty much did what you did, except I refactored a bit and did not make any changes for WOW. Do they suffer from the same bug?

duane-nepa, 2020-10-23:
I'll give it a try shortly. I copied what you did and just added it into the rapidfire section. Did I change WOW, wasn't trying to.

I'm not uploading to WOW right now. I think I have an account, if not I'll get one made and give them a try. It's not calm right now however, probably later it will be.

WU hasn't answered me yet. If I don't hear by Tuesday I'll contact again. It would be nice to get them to fix the software.

duane-nepa, 2020-10-23:
Works fine on WU, both rapidfire and standard.

I'm getting this from WOW. 
Oct 23 18:44:26 weewx weewx[5813] DEBUG weewx.restx_test: WOW: Failed upload attempt 3: HTTP Error 400: Bad Request

I just made my Authentication Key, I was going to set this up but never did. Maybe it needs some time for that to propagate?