weewx/weewx #617: Weewx stop with WeatherLinkLive

ddperso, 2020-10-22:
In weewx.conf I have the driver WeatherLinkLiveJSON:

##############################################################################

[WeatherLinkLiveJSON]
    driver = user.WeatherLinkLiveJSON
    max_tries = 10
    retry_wait = 5
    poll_interval = 2.5
    url = http://192.168.1.201:80/v1/current_conditions

##############################################################################
Yesterday I have a brief link interruption between my server and the WeatherLinkLive box, then weewx service has stopped.

/var/log/syslog.1:Oct 21 22:47:15 meteo weewx[26403] INFO user.WeatherLinkLiveJSON: failure to get data http://192.168.1.201:80/v1/current_conditions - try 1 - (HTTPConnectionPool(host='192.168.1.201', port=80): Max retries exceeded with url: /v1/current_conditions (Caused by NewConnectionError("<urllib3.connection.HTTPConnection object at 0x7efc0cdede48>: Failed to establish a new connection: [Errno 110] Connexion terminée par expiration du délai d'attente")))
/var/log/syslog.1:Oct 21 22:47:17 meteo weewx[26403] CRITICAL weewx.engine: Internal error. Packet loop has exited.
/var/log/syslog.1:Oct 21 22:47:18 meteo weewx[26403] INFO weewx.engine: Main loop exiting. Shutting engine down.

In syslog I can see that weewx stops with try = 1, but in weewx.conf I have max_tries = 10

Is it normal?

Thank's a lot

tkeffer, 2020-10-22:
This issues list is for tracking bugs in WeeWX. WeatherLinkLive is not part of WeeWX.

Please post your question to [weewx-user](https://groups.google.com/forum/#!forum/weewx-user). You are more likely to get a response there.

closed 2020-10-22 by tkeffer

ddperso, 2020-10-22:
Sorry, 
thank's