weewx/weewx #618: Ubuntu 20.04 breaks dependencies in weewx deb packaging

urilabob, 2020-10-24:
I recently upgraded a Ubuntu installation to 20.04; weewx was previously working perfectly, but has been removed in the upgrade and can't be reinstalled because:
 weewx : Depends: python-configobj (>= 4.5) which is a virtual package and is not provided by any available package
               Depends: python-serial (>= 2.3) which is a virtual package and is not provided by any available package
python3-config and python3-serial are installable (but of course, don't satisfy the dependencies). I think the renaming decision in ubuntu is pretty much a hard-and-fast decision, I can't see them changing it. Is there any workaround in the works (I note that on the web, some people suggest using setup.py, but this still seems to require installing python-configobj and python-serial directly, and this also fails).

Looking at the deb file documentation, it looks like fixing this is complex, but I don't understand the deb unpacking process well enough. If the package specifies alternate dependencies (e.g. python3-configobj | python-configobj), and the package manager chose, say, python3-configobj, is this choice visible to trigger scripts? If so, it seems it might be possible to run sed in a trigger script to replace occurrences of python-configobj by python3-configobj during unpacking, and similarly for python3-serial. If that could work, and if someone with deb packaging expertise is willing to work with me, I'd be happy to do the low-level grunt work writing and testing scripts (or more generally, help in any way I can).

duane-nepa, 2020-10-24:
I run weewx 4.1.1 on Ubuntu server 20.04 and python3 with no issues.

I had to install pip3 for something but it was over a month ago and don't remember why. 
$ sudo apt update
$ sudo apt install python3-pip

I think it was for a missing dependency probably the ones you mentioned.

gjr80, 2020-10-24:
You might want search [weewx-user](https://groups.google.com/g/weewx-user), this has come up a couple of times now I believe.

urilabob, 2020-10-24:
Thank you Duane, gjr, your trouble is much appreciated, and got me on the right track. 

Actually, I followed all listed instructions and the problem remained. However I then went through all the steps again, but replacing install with reinstall. Yeeha, it worked. 

I know the remove script that was invoked by the upgrade to Ubuntu 20.04 didn't actually remove everything (for example, I know weewx.conf was left in place - fortunately, as it saved me a lot of reconfiguration). I'm suspecting that something left behind was what was blocking it, and the reinstalls cleaned up whatever the problem was. So problem solved. 

Now I just have to  mark this as closed (I'll put it as notabug, though actually I think there is a bug, not in the install script but in either the new or old remove script; but there's very little chance that I can now find what went wrong, as fixing it has probably removed all the evidence...). Anyway, I've now learnt quite a bit about how deb packages work, so I guess that'll come in handy sometime...
    Thanks and Best Wishes
    Bob

urilabob, 2020-10-24:
Ooops, sorry, my misunderstanding of the close process. Doesn't look like I get to mark the kind of resolution. I think there is no bug in the install process. However I think there probably is a bug in the remove process (that was activated by a ubuntu upgrade because it couldn't find the python2 dependencies). That left the whole shebang in an unstable state that prevented install by the process described in http://weewx.com/docs/debian.htm. However replacing 'install' by 'reinstall' at every step cleaned it up enough for the installation to work.

closed 2020-10-24 by urilabob