weewx/weewx #619: wee_extension and other utilities should warn about insufficient permissions

michael-slx, 2020-10-30:
### Issue

When using `wee_extension` as well as some of the other utilities they do not check for proper file permissions or only do so after the interactive prompts are completed.

### Suggested behavior

Utilities should check for permissions on relevant paths (e.g. weewx.conf, extension directory, skin directory, database files), print an error and terminate.

Being an interactive utility, `wee_config` should do this before prompting the user in order to avoid having to re-type all answers.

### Examples

#### `wee_extension`

`wee_extension` doesn't mention anything permission-related. This makes it an hard-to-trace issue especially for less experienced users.

```sh
vagrant@vm-weewx-wll-test:~$ wee_extension --install=weewx-weatherlinklive-driver.tar.xz /etc/weewx/weewx.conf
Request to install 'weewx-weatherlinklive-driver.tar.xz'
Extracting from tar archive weewx-weatherlinklive-driver.tar.xz
Traceback (most recent call last):
  File "/usr/share/weewx/wee_extension", line 88, in <module>
    main()
  File "/usr/share/weewx/wee_extension", line 80, in main
    ext.install_extension(options.install)
  File "/usr/share/weewx/weecfg/extension.py", line 123, in install_extension
    self.install_from_dir(extension_dir)
  File "/usr/share/weewx/weecfg/extension.py", line 181, in install_from_dir
    shutil.copy(source_path, destination_path)
  File "/usr/lib/python3.7/shutil.py", line 245, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.7/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/share/weewx/user/weatherlink_live/__init__.py'
```

#### `wee_config`

`wee_config `fails due to lacking permissions only after interactive prompts are completed., making it a nuisance having to re-enter all answers.

```sh
vagrant@vm-weewx-wll-test:~$ wee_config --reconfigure /etc/weewx/weewx.conf
Using configuration file /etc/weewx/weewx.conf
Enter a brief description of the station, such as its location.  For example:
Santa's Workshop, North Pole
description [Santa's Workshop, North Pole]:

Specify altitude, with units 'foot' or 'meter'.  For example:
35, foot
12, meter
altitude [0, meter]:

Specify latitude in decimal degrees, negative for south.
latitude [90.000]:
Specify longitude in decimal degrees, negative for west.
longitude [0.000]:

You can register your station on weewx.com, where it will be included
in a map. You will need a unique URL to identify your station (such as a
website, or WeatherUnderground link).
Include station in the station registry (y/n)? [n]:

Indicate the preferred units for display: ['us', 'metric']
units [us]:

Installed drivers include:
  0) AcuRite         (weewx.drivers.acurite)
  1) CC3000          (weewx.drivers.cc3000)
  2) FineOffsetUSB   (weewx.drivers.fousb)
  3) Simulator       (weewx.drivers.simulator)
  4) TE923           (weewx.drivers.te923)
  5) Ultimeter       (weewx.drivers.ultimeter)
  6) Vantage         (weewx.drivers.vantage)
  7) WMR100          (weewx.drivers.wmr100)
  8) WMR200          (weewx.drivers.wmr200)
  9) WMR300          (weewx.drivers.wmr300)
 10) WMR9x8          (weewx.drivers.wmr9x8)
 11) WS1             (weewx.drivers.ws1)
 12) WS23xx          (weewx.drivers.ws23xx)
 13) WS28xx          (weewx.drivers.ws28xx)
choose a driver [3]: Traceback (most recent call last):
  File "/usr/lib/python3.7/shutil.py", line 563, in move
    os.rename(src, real_dst)
PermissionError: [Errno 13] Permission denied: '/etc/weewx/weewx.conf' -> '/etc/weewx/weewx.conf.20201030090526'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/weewx/wee_config", line 123, in <module>
    main()
  File "/usr/share/weewx/wee_config", line 117, in main
    config_mgr.run(args, options)
  File "/usr/share/weewx/weecfg/config.py", line 126, in run
    backup_path = weecfg.save(config_dict, output_path, not options.no_backup)
  File "/usr/share/weewx/weecfg/__init__.py", line 200, in save
    backup_path = weeutil.weeutil.move_with_timestamp(config_path)
  File "/usr/share/weewx/weeutil/weeutil.py", line 1309, in move_with_timestamp
    shutil.move(filepath, newpath)
  File "/usr/lib/python3.7/shutil.py", line 577, in move
    copy_function(src, real_dst)
  File "/usr/lib/python3.7/shutil.py", line 263, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.7/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: '/etc/weewx/weewx.conf.20201030090526'
```

tkeffer, 2020-10-30:
Good points. Shouldn't be too hard to do.

tkeffer, 2020-11-03:
Sorry, Michael. I started to implement this, and it was becoming way more complicated than the benefits warrant. For example, there are downstream implications for the package installers.

closed 2020-11-03 by tkeffer

michael-slx, 2020-11-03:
Thanks, @tkeffer for your efforts, anyway.