weewx/weewx #624: Seasons Report Exposes Inside Temperature

JeremyCaron, 2020-12-09:
Hi - I just upgraded from an AcuRite 5-in-1 to an Ambient WS-2000, which means my station now sends weewx indoor temp data when it didn't before.  I've noticed that the Seasons report exposes graphs for inside temp by default, when weewx's approach generally seems to be to respect privacy in this area.  See the comment from the User Guide about indoor temp not being sent to Weather Underground by default: "In the interest of respecting your privacy, WeeWX does not post indoor temperature or humidity to the Weather Underground unless you set this option to True. Default is False."

[Screenshot](https://i.imgur.com/mm5CJov.png)

Could we get an option to include/exclude these graphs for this skin?  If not, could they just be removed?  All the other built-in skins seem to suppress this by default (smartphone and mobile at the very least do).

Thanks for your consideration on this topic, and for this AWESOME piece of software :)

tkeffer, 2020-12-09:
The WU upload is different from the skins. With the former, the user has little control over what gets sent, and s/he has to rely on the option `post_indoor_observations` to protect privacy.

However, with the skins, you have total control. Simply comment out what you don't want. For example, change this

```ini
        [[[daytempfeel]]]
            [[[[windchill]]]]
            [[[[heatindex]]]]

        [[[daytempin]]]
            [[[[inTemp]]]]

        [[[daytemp]]]
            yscale = None, None, 0.5
            [[[[extraTemp1]]]]
            [[[[extraTemp2]]]]
            [[[[extraTemp3]]]]
```

to this

```ini
        [[[daytempfeel]]]
            [[[[windchill]]]]
            [[[[heatindex]]]]

#        [[[daytempin]]]
#            [[[[inTemp]]]]

        [[[daytemp]]]
            yscale = None, None, 0.5
            [[[[extraTemp1]]]]
            [[[[extraTemp2]]]]
            [[[[extraTemp3]]]]
```
And so on.

closed 2020-12-09 by tkeffer

JeremyCaron, 2020-12-09:
Will the commented code be overwritten the next time I upgrade weewx?

tkeffer, 2020-12-09:
Yes, you will have to comment out (or delete) things in several places. Whatever you don't want.

Everything in the `skins` subdirectory is preserved across upgrades.

JeremyCaron, 2020-12-09:
Got it, thank you.  If I were to do the dev work to add such a switch, would you consider merging the PR or are you opposed to the idea regardless of who builds it?

tkeffer, 2020-12-09:
Of course I'd consider it, but it's harder than you think. There's no meta-language for `ConfigObj` files such as `skin.conf`, so to prevent generating unnecessary images, you'd still have to tell the user to comment out the unneeded images in `skin.conf`. 

You'd also need to update the docs with whatever option you choose.

sastorsl, 2022-08-29:
I wanted to add inTemp to the Smartphone report and even considered a PR, so I'm glad I found this issue which brings some light into why inTemp was removed.

For those interested here's what I did to add inTemp manually.

```bash
# Double check the install directory for your OS / packet manager
cd /etc/weewx/skins/Smartphone

# Add inTemp before (insert) outTemp in the home page template file
# index.html.tmpl
        <li><a href="tempin.html">$obs.label.inTemp: $current.inTemp [$day.inTemp.min, $day.inTemp.max]</a></li>' 

# Add MobileTempIn to skin.conf under CheetahGenerator
        [[[MobileTempIn]]]
            template = tempin.html.tmpl

# Add localization of "inTemp" to the appropriate lang/NN.conf file
inTemp = Intra temperatus
```

Wait for a new report to be created or run `wee_reports` if you are impatient.