weewx/weewx #625: can't use launchctl to start on M1 mac with python 3.9.1

geoffsharris, 2020-12-21:
Installed python 3.9.1 onto M1 mac and can get weewx to start as a native daemon from native terminal, but can't get to start using launchctl. Using pyenv with global and .zprofile path for the python install so wondering if this is some sort of path issue related to launchctl,  a weewx issue or something else. 

Dec 21 10:58:10 Mac-mini com.apple.xpc.launchd[1] (com.weewx.weewxd[893]): Service could not initialize: 20C69: xpcproxy + 23696 [839][D05D0576-CE1E-3530-B74C-0EB266FFBFDD]: 0x2
Dec 21 10:58:10 Mac-mini com.apple.xpc.launchd[1] (com.weewx.weewxd[893]): Service exited with abnormal code: 78

using python3 ./bin/weewxd weewx.conf
weewx[1269] INFO __main__: Initializing weewx version 4.2.0
weewx[1269] INFO __main__: Using Python 3.9.1 (default, Dec 21 2020, 07:33:29) 
[Clang 12.0.0 (clang-1200.0.32.28)]
weewx[1269] INFO __main__: Platform macOS-11.1-arm64-arm-64bit
weewx[1269] INFO __main__: Locale is 'en_US.UTF-8'
weewx[1269] INFO __main__: PID file is /var/run/weewx.pid
weewx[1288] INFO __main__: Using configuration file weewx.conf
weewx[1288] INFO __main__: Debug is 0

tkeffer, 2020-12-21:
No idea, and I do not have access to a Mac with Apple Silicon.

You'd probably be better off posting your question to [weewx-user](https://groups.google.com/forum/#!forum/weewx-user). You might attract the eye of a MacExpert there.

tkeffer, 2020-12-21:
Looks like you already did so.

closed 2020-12-21 by tkeffer

geoffsharris, 2020-12-22:
It looks to be a an issue with the pointer to the python3 path. I'm using pyenv with python 3.9.1. I have pyenv set correctly and it spits out python3 -V 3.9.1 and which python at the .pyenv location. My path points to the .pyenv/shim location as well. When I do /usr/bin/env python it says 3.9.1. When setup.py is run with 3.9.1 I end up with the shebang at the top of weewxd it point to:
#!/Library/Frameworks/Python.framework/Versions/3.9/bin/python3
Which is a folder that does not exit. 
If i build using system in pyenv the top of weewxd shebang is:
#!/Library/Developer/CommandLineTools/usr/bin/python3
Not sure how the reference is created for /Library/Frameworks/Python.framework/Versions/3.9/bin/python3, but it clearly doesn't exist on my system and can't be found when launchctl tries to start. Weewx works fine when I start as a daemon from the terminal. Any thoughts on why weewx would be populated with the /Library/Frameworks/Python.framework/Versions/3.9/bin/python3 reference?

geoffsharris, 2020-12-22:
Had previously installed python 3.9.1 as a package installer which after removal had this dead symlink. Grrrr. Good news is that if you build python 3.9.1 from source, get libjpeg and freetype via a the experimental native version of Homebrew and then build PILLOW from source all in the native Terminal, weewx works fine.

tkeffer, 2020-12-22:
Good to know.

I'm thinking of getting a new MacBook to replace my 6 month old Intel machine. Apple is offering a very generous trade in allowance. But, I'm uncertain how much memory to get. Does yours have 8GB, or 16GB?

geoffsharris, 2020-12-23:
I went with 16gb. Actually isn't spec'd all that much higher than the 10 year old mini it is replacing.