weewx/weewx #629: Switching to Beaufort scale

uajqq, 2020-12-27:
I'm trying to display my wind speed in Beaufort scale. If I make the appropriate adjustments in weewx.conf, the values switch over and look correct. However, I'm getting the following error tracing that I can't make sense of:

```
weewx stopped!                                                                                                                                                   
weewx restarted!                                                                                                                                                 
Using configuration file /Users/Shared/weewx/weewx.conf                                                                                                          
Generating for all time                                                                                                                                          
Traceback (most recent call last):                                                                                                                               
  File "/Users/Shared/weewx/bin/weewx/units.py", line 1322, in convert                                                                                           
    new_val = list([conversion_func(x) if x is not None else None for x in val_t[0]])                                                                            
  File "/Users/Shared/weewx/bin/weewx/units.py", line 1322, in <listcomp>                                                                                        
    new_val = list([conversion_func(x) if x is not None else None for x in val_t[0]])
  File "/Users/Shared/weewx/bin/weewx/wxformulas.py", line 589, in <lambda>
    weewx.units.conversionDict['mile_per_hour']['beaufort'] = lambda x : beaufort(mph_to_knot(x))
  File "/Users/Shared/weewx/bin/weewx/wxformulas.py", line 562, in beaufort
    elif ws_kts < 1:
TypeError: '<' not supported between instances of 'complex' and 'int'
  
During handling of the above exception, another exception occurred:
  
Traceback (most recent call last):
  File "/Users/Shared/weewx/bin/weewx/reportengine.py", line 196, in run
    obj.start()
  File "/Users/Shared/weewx/bin/weewx/reportengine.py", line 281, in start
    self.run()
  File "/Users/Shared/weewx/bin/weewx/imagegenerator.py", line 41, in run
    self.genImages(self.gen_ts)
  File "/Users/Shared/weewx/bin/weewx/imagegenerator.py", line 195, in genImages 
    new_data_vec_t = self.converter.convert(data_vec_t)
  File "/Users/Shared/weewx/bin/weewx/units.py", line 966, in convert
    new_val_t = convert(val_t, new_unit_type)
  File "/Users/Shared/weewx/bin/weewx/units.py", line 1324, in convert
    new_val = conversion_func(val_t[0]) if val_t[0] is not None else None
File "/Users/Shared/weewx/bin/weewx/wxformulas.py", line 589, in <lambda>
    weewx.units.conversionDict['mile_per_hour']['beaufort'] = lambda x : beaufort(mph_to_knot(x))
  File "/Users/Shared/weewx/bin/weewx/units.py", line 66, in mph_to_knot
    return x * 0.868976242
TypeError: can't multiply sequence by non-int of type 'float'
Traceback (most recent call last):
  File "/Users/Shared/weewx/bin/weewx/reportengine.py", line 196, in run
    obj.start()
  File "/Users/Shared/weewx/bin/weewx/reportengine.py", line 281, in start
    self.run()
  File "/Users/Shared/weewx/bin/user/gaugeengine.py", line 121, in run
    self.gen_gauges()
  File "/Users/Shared/weewx/bin/user/gaugeengine.py", line 171, in gen_gauges
    ret = self.gen_gauge(gauge, plot_options, img_file)
  File "/Users/Shared/weewx/bin/user/gaugeengine.py", line 355, in gen_gauge
    gauge.add_history(history_list, num_buckets, windspeed_history_list, rings, ring_colors)
  File "/Users/Shared/weewx/bin/user/gauges.py", line 519, in add_history
    raise Exception("Number of ring vals (%d) does not match the number of list vals (%d)." % (len(ring_vals), len(list_vals)))
Exception: Number of ring vals (0) does not match the number of list vals (283).
```

When I added some debugging reports to units.py, it looks like the issue is that beaufort(mile_to_knot(x)) is being called on a long array of complex numbers. Here's a sample:
```
Failed converting this: [(-11.075065230904336+10.535770571918707j), (-11.570320074638877+9.071309977839318j), (-11.391080869179204+9.558251756038839j), 
(-6.681337263369387+7.710325157056584j), (-12.919528669270413+6.596441353624022j), (-0.4738855040745599+7.941874371269544j), (-12.192653371078269+8.9181515900165
08j), (-11.125189750738684+3.5970327705797387j),
...
```

I can't make heads or tails of this. Is this a bug or do I have a configuration setting wrong? Like I said, the values look like they are being calculated and graphed correctly on all of my skins.

tkeffer, 2020-12-28:
The gauge graphics that you are using are not part of WeeWX. You'll have to ask the extension author why s/he is trying to do a conversion on a list, instead of a scalar.

uajqq, 2020-12-28:
Ah yes—I had installed that extension so long ago I forgot it wasn’t part of weewx.

However, after cleaning out every unused extension except for Belchertown, the first two errors are still being thrown. Adding an exception to units.py shows it’s still trying to convert a large array of complex numbers. Tracing now:
```
Using configuration file /Users/Shared/weewx/weewx.conf
Generating for all time
Traceback (most recent call last):
  File "/Users/Shared/weewx/bin/weewx/units.py", line 1322, in convert
    new_val = list([conversion_func(x) if x is not None else None for x in val_t[0]])
  File "/Users/Shared/weewx/bin/weewx/units.py", line 1322, in <listcomp>
    new_val = list([conversion_func(x) if x is not None else None for x in val_t[0]])
  File "/Users/Shared/weewx/bin/weewx/wxformulas.py", line 589, in <lambda>
    weewx.units.conversionDict['mile_per_hour']['beaufort'] = lambda x : beaufort(mph_to_knot(x))
  File "/Users/Shared/weewx/bin/weewx/wxformulas.py", line 562, in beaufort
    elif ws_kts < 1:
TypeError: '<' not supported between instances of 'complex' and 'int'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/Shared/weewx/bin/weewx/reportengine.py", line 196, in run
    obj.start()
  File "/Users/Shared/weewx/bin/weewx/reportengine.py", line 281, in start
    self.run()
  File "/Users/Shared/weewx/bin/weewx/imagegenerator.py", line 41, in run
    self.genImages(self.gen_ts)
  File "/Users/Shared/weewx/bin/weewx/imagegenerator.py", line 195, in genImages
    new_data_vec_t = self.converter.convert(data_vec_t)
  File "/Users/Shared/weewx/bin/weewx/units.py", line 966, in convert
    new_val_t = convert(val_t, new_unit_type)
  File "/Users/Shared/weewx/bin/weewx/units.py", line 1324, in convert
    new_val = conversion_func(val_t[0]) if val_t[0] is not None else None
  File "/Users/Shared/weewx/bin/weewx/wxformulas.py", line 589, in <lambda>
    weewx.units.conversionDict['mile_per_hour']['beaufort'] = lambda x : beaufort(mph_to_knot(x))
  File "/Users/Shared/weewx/bin/weewx/units.py", line 66, in mph_to_knot
    return x * 0.868976242
TypeError: can't multiply sequence by non-int of type 'float'
```

uajqq, 2020-12-28:
Not sure if it's relevant, but I cleared out the public_html folder, and noticed that the wind vector chart is the only one not rendering. Is it possible wind vectors are getting converted somehow?

tkeffer, 2020-12-28:
Two things:

1. I think we're going to have to see the log in order to understand the context. Set `debug=1`, restart weewxd, let it run through the error, post everything.
2. Where are you asking for the beaufort conversion?

uajqq, 2020-12-28:
1. Here's the entire startup log:
```
weewx[92027] INFO __main__: Initializing weewx version 4.3.0b3
weewx[92027] INFO __main__: Using Python 3.8.3 (default, Jul 22 2020, 21:10:41)
[Clang 11.0.3 (clang-1103.0.32.29)]
wee_reports[92040] INFO user.davishealthapi: service version is 0.10
wee_reports[92040] INFO user.davishealthapi: archive interval is 360
wee_reports[92040] DEBUG weewx.manager: Daily summary version is 3.0
wee_reports[92040] DEBUG weewx.engine: Finished loading service user.davishealthapi.DavisHealthAPI
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.engine.StdConvert
wee_reports[92040] INFO weewx.engine: StdConvert target unit is 0x1
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.engine.StdConvert
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.engine.StdCalibrate
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.engine.StdCalibrate
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.engine.StdQC
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.engine.StdQC
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.wxservices.StdWXCalculate
wee_reports[92040] DEBUG weewx.manager: Daily summary version is 3.0
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.wxservices.StdWXCalculate
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdWXXTypes
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdWXXTypes
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdPressureCooker
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdPressureCooker
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdRainRater
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdRainRater
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdDelta
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdDelta
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.engine.StdArchive
wee_reports[92040] INFO weewx.engine: Archive will use data binding wx_binding
wee_reports[92040] INFO weewx.engine: Record generation will be attempted in 'hardware'
wee_reports[92040] INFO weewx.engine: Using archive interval of 300 seconds (specified by hardware)
wee_reports[92040] DEBUG weewx.engine: Use LOOP data in hi/low calculations: 1
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.engine.StdArchive
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.restx.StdStationRegistry
wee_reports[92040] INFO weewx.restx: StationRegistry: Station will be registered.
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.restx.StdStationRegistry
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.restx.StdWunderground
wee_reports[92040] DEBUG weewx.restx: WU essentials: {}
wee_reports[92040] INFO weewx.restx: Wunderground-PWS: Data for station KCTNEWHA41 will be posted
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.restx.StdWunderground
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.restx.StdPWSweather
wee_reports[92040] INFO weewx.restx: PWSWeather: Data for station FW5820 will be posted
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.restx.StdPWSweather
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.restx.StdCWOP
wee_reports[92040] INFO weewx.restx: CWOP: Data for station FW5820 will be posted
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.restx.StdCWOP
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.restx.StdWOW
wee_reports[92040] INFO weewx.restx: WOW: Posting not enabled.
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.restx.StdWOW
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.restx.StdAWEKAS
wee_reports[92040] INFO weewx.restx: AWEKAS: Posting not enabled.
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.restx.StdAWEKAS
wee_reports[92040] DEBUG weewx.engine: Loading service user.mqtt.MQTT
wee_reports[92040] DEBUG weewx.manager: Daily summary version is 3.0
weewx[92027] INFO __main__: Platform macOS-10.16-x86_64-i386-64bit
weewx[92027] INFO __main__: Locale is 'en_US/UTF-8/en_US/en_US/en_US/en_US'
wee_reports[92040] INFO user.mqtt: service version is 0.22
wee_reports[92040] INFO user.mqtt: binding to ['archive', 'loop']
wee_reports[92040] DEBUG user.mqtt: TLS parameters: {'ca_certs': '/Users/Shared/weewx/cacert.pem'}
wee_reports[92040] INFO user.mqtt: topic is weather
wee_reports[92040] INFO user.mqtt: desired unit system is US
wee_reports[92040] INFO user.mqtt: data will be uploaded to mqtt://xxx:8883
wee_reports[92040] INFO user.mqtt: network encryption/authentication will be attempted
wee_reports[92040] DEBUG weewx.engine: Finished loading service user.mqtt.MQTT
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.engine.StdPrint
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.engine.StdPrint
wee_reports[92040] DEBUG weewx.engine: Loading service weewx.engine.StdReport
wee_reports[92040] DEBUG weewx.engine: Finished loading service weewx.engine.StdReport
wee_reports[92040] DEBUG weewx.manager: Daily summary version is 3.0
wee_reports[92040] DEBUG weewx.manager: Daily summary version is 3.0
[... 157 lines cut ...]
[... cut ...]
wee_reports[92040] ERROR weewx.reportengine:         ****  Generator terminated
```

2. I requested the conversion in weewx.conf by setting:
```
[StdReport]
    [[Defaults]]
        [[[Units]]]
            [[[[Groups]]]]
                group_speed = beaufort
            [[[[StringFormats]]]]
                beaufort = %.0f
            [[[[Labels]]]]
                beaufort = " Beaufort"
```

I see in the logs its warning that type Beaufort has been depreciated in favor of unit Beaufort. That should be set as part of group_speed, though, right?

tkeffer, 2020-12-28:
OK. Now I see what you're doing. This should work. I'll check it out.

tkeffer, 2020-12-28:
Not sure why you'd want to do this, but it is now possible to have progressive wind vector plots using units of `beaufort`. 

Commit https://github.com/weewx/weewx/commit/19790b2786ff709492759de0491a827fbe07c54d, to appear in V4.3

closed 2020-12-28 by tkeffer