weewx/weewx #636: Netatmo station - help needed for read some observation.

RebeckaJohansson, 2021-01-12:
I have Netamo station indoor modul and outdoor module with Wind Anemometer and Rain Gauge.
I installed it with the Netatmo extension from Matthewwall via an forked updated extension from Bricebou.
I also use Belchertown skin and public site via MQTT broker and so far all works and the loop info reads all the observation in command line on my Raspberry pi 4. 

But i dont know how to do to publish the wifi, Rf and battery sensors readings. 

Here is a copy/paste from my loop and as you can see it reads all information for weewx but in the main season skin it only says low battery everywhere and i cant get it on Belchertown either. 

`Jan 12 11:37:57 Raspi /weewxd: netatmo: netatmo-client: next update in 300 seconds
Jan 12 11:37:57 Raspi /weewxd: netatmo: MainThread: data: {'70:ee:50:65:3d:c0.NAMai
n.time_utc': 1610447313, '70:ee:50:65:3d:c0.NAMain.wifi_status': 57, '70:ee:50:65:3
d:c0.NAMain.co2_calibrating': True, '70:ee:50:65:3d:c0.NAMain._id': '70:ee:50:65:3d
:c0', '70:ee:50:65:3d:c0.NAMain.module_name': 'MyNetatmo', '70:ee:50:65:3d:c0.NAMai
n.last_status_store': 1610447319, '70:ee:50:65:3d:c0.NAMain.firmware': 179, '70:ee:
50:65:3d:c0.NAMain.last_setup': 1605898056, '70:ee:50:65:3d:c0.NAMain.date_setup':
1605898056, '70:ee:50:65:3d:c0.NAMain.Temperature': 23.1, '70:ee:50:65:3d:c0.NAMain
.Humidity': 40, '70:ee:50:65:3d:c0.NAMain.AbsolutePressure': 994.5, '70:ee:50:65:3d
:c0.NAMain.Pressure': 995.5, '70:ee:50:65:3d:c0.NAMain.CO2': 672, '70:ee:50:65:3d:c
0.NAMain.Noise': 44, '02:00:00:65:02:32.NAModule1.time_utc': 1610447288, '02:00:00:
65:02:32.NAModule1.rf_status': 60, '02:00:00:65:02:32.NAModule1.battery_vp': 5874,
'02:00:00:65:02:32.NAModule1._id': '02:00:00:65:02:32', '02:00:00:65:02:32.NAModule
1.module_name': 'Outdoor', '02:00:00:65:02:32.NAModule1.last_seen': 1610447288, '02
:00:00:65:02:32.NAModule1.battery_percent': 93, '02:00:00:65:02:32.NAModule1.firmwa
re': 50, '02:00:00:65:02:32.NAModule1.last_setup': 1605898013, '02:00:00:65:02:32.N
AModule1.Temperature': 3, '02:00:00:65:02:32.NAModule1.Humidity': 75, '05:00:00:07:
f3:1e.NAModule3.time_utc': 1610447308, '05:00:00:07:f3:1e.NAModule3.rf_status': 63,
 '05:00:00:07:f3:1e.NAModule3.battery_vp': 6046, '05:00:00:07:f3:1e.NAModule3._id':
 '05:00:00:07:f3:1e', '05:00:00:07:f3:1e.NAModule3.module_name': 'Outdoor Rain', '0
5:00:00:07:f3:1e.NAModule3.last_seen': 1610447308, '05:00:00:07:f3:1e.NAModule3.bat
tery_percent': 100, '05:00:00:07:f3:1e.NAModule3.firmware': 12, '05:00:00:07:f3:1e.
NAModule3.last_setup': 1605982034, '05:00:00:07:f3:1e.NAModule3.Rain': 0.0, '05:00:
00:07:f3:1e.NAModule3.sum_rain_24': 0.06, '05:00:00:07:f3:1e.NAModule3.sum_rain_1':
 0.0, '06:00:00:04:d3:ec.NAModule2.time_utc': 1610447289, '06:00:00:04:d3:ec.NAModu
le2.rf_status': 86, '06:00:00:04:d3:ec.NAModule2.battery_vp': 5907, '06:00:00:04:d3
:ec.NAModule2._id': '06:00:00:04:d3:ec', '06:00:00:04:d3:ec.NAModule2.module_name':
 'Outdoor Wind', '06:00:00:04:d3:ec.NAModule2.last_seen': 1610447314, '06:00:00:04:
d3:ec.NAModule2.battery_percent': 95, '06:00:00:04:d3:ec.NAModule2.firmware': 25, '
06:00:00:04:d3:ec.NAModule2.last_setup': 1605982333, '06:00:00:04:d3:ec.NAModule2.W
indStrength': 2, '06:00:00:04:d3:ec.NAModule2.WindAngle': 237, '06:00:00:04:d3:ec.N
AModule2.GustStrength': 5, '06:00:00:04:d3:ec.NAModule2.GustAngle': 237}
Jan 12 11:37:57 Raspi /weewxd: netatmo: MainThread: packet: {'dateTime': 1610447878
, 'usUnits': 16, 'pressure': 994.5, 'inTemp': 23.1, 'inHumidity': 40, 'co2': 672, '
noise': 44, 'wifi_status': 57, 'outTemp': 3, 'outHumidity': 75, 'out_rf_status': 60
, 'out_battery_vp': 5874, 'outTempBatteryStatus': 93, 'windSpeed': 2, 'windDir': 23
7, 'windGust': 5, 'windGustDir': 237, 'wind_rf_status': 86, 'wind_battery_vp': 5907
, 'windBatteryStatus': 95, 'rain': 0.0, 'rain_total': 0.06, 'rain_rf_status': 63, '
rain_battery_vp': 6046, 'rainBatteryStatus': 100}
Jan 12 11:37:57 Raspi weewx[25741] INFO weewx.restx: MQTT: Published record 2021-01
-12 11:37:58 CET (1610447878)
Jan 12 11:37:57 Raspi weewx[25741] INFO weewx.manager: Added record 2021-01-12 11:3
5:00 CET (1610447700) to database 'weewx.sdb'
Jan 12 11:37:58 Raspi weewx[25741] INFO weewx.manager: Added record 2021-01-12 11:3
5:00 CET (1610447700) to daily summary in 'weewx.sdb'
Jan 12 11:37:58 Raspi weewx[25741] DEBUG weewx.reportengine: Running reports for la
test time in the database.
Jan 12 11:37:58 Raspi weewx[25741] DEBUG weewx.reportengine: Running report 'Season
sReport'
Jan 12 11:37:58 Raspi weewx[25741] DEBUG weewx.restx: StationRegistry: wait interva
l (1800 < 604800) has not passed for record 2021-01-12 11:35:00 CET (1610447700)
Jan 12 11:37:58 Raspi weewx[25741] DEBUG weewx.reportengine: Found configuration fi
le /etc/weewx/skins/Seasons/skin.conf for report 'SeasonsReport'
Jan 12 11:37:58 Raspi weewx[25741] DEBUG weewx.cheetahgenerator: Using search list
['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Station', 'weewx.cheetah
generator.Current', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInf
o', 'weewx.cheetahgenerator.Extras']
Jan 12 11:37:58 Raspi weewx[25741] INFO weewx.restx: MQTT: Published record 2021-01
-12 11:35:00 CET (1610447700)
Jan 12 11:37:58 Raspi weewx[25741] DEBUG weewx.manager: Daily summary version is 3.
0
Jan 12 11:37:58 Raspi weewx[25741] INFO weewx.restx: PWSWeather: Published record 2
021-01-12 11:35:00 CET (1610447700)
Jan 12 11:37:59 Raspi weewx[25741] INFO weewx.restx: Wunderground-PWS: Published re
cord 2021-01-12 11:35:00 CET (1610447700)
Jan 12 11:37:59 Raspi weewx[25741] INFO weewx.cheetahgenerator: Generated 8 files f
or report SeasonsReport in 0.84 seconds
Jan 12 11:37:59 Raspi weewx[25741] DEBUG weewx.manager: Daily summary version is 3.
0
Jan 12 11:37:59 Raspi weewx[25741] INFO weewx.imagegenerator: Generated 15 images f
or report SeasonsReport in 0.57 seconds
Jan 12 11:37:59 Raspi weewx[25741] INFO weewx.reportengine: Copied 0 files to /var/
www/html/weewx
Jan 12 11:37:59 Raspi weewx[25741] DEBUG weewx.reportengine: Report 'SmartphoneRepo
rt' not enabled. Skipping.
Jan 12 11:37:59 Raspi weewx[25741] DEBUG weewx.reportengine: Report 'MobileReport'
not enabled. Skipping.
Jan 12 11:37:59 Raspi weewx[25741] DEBUG weewx.reportengine: Report 'StandardReport
' not enabled. Skipping.
Jan 12 11:37:59 Raspi weewx[25741] DEBUG weewx.reportengine: Running report 'Belche
rtown'
Jan 12 11:37:59 Raspi weewx[25741] DEBUG weewx.reportengine: Found configuration fi
le /etc/weewx/skins/Belchertown/skin.conf for report 'Belchertown'
Jan 12 11:38:00 Raspi weewx[25741] DEBUG weewx.cheetahgenerator: Using search list
['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Station', 'weewx.cheetah
generator.Current', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInf
o', 'weewx.cheetahgenerator.Extras', 'user.belchertown.getData']
Jan 12 11:38:00 Raspi weewx[25741] DEBUG weewx.manager: Daily summary version is 3.
0
Jan 12 11:38:00 Raspi weewx[25741] INFO weewx.cheetahgenerator: Generated 11 files
for report Belchertown in 0.97 seconds
Jan 12 11:38:00 Raspi weewx[25741] INFO weewx.reportengine: Copied 2 files to /var/
www/html
Jan 12 11:38:01 Raspi weewx[25741] DEBUG weewx.manager: Daily summary version is 3.
0
Jan 12 11:38:02 Raspi weewx[25741] DEBUG weewx.reportengine: Report 'FTP' not enabl
ed. Skipping.
Jan 12 11:38:02 Raspi weewx[25741] DEBUG weewx.reportengine: Report 'RSYNC' not ena
bled. Skipping.
`

The info about readings for signals and battery its via extensions info 

`class CloudClient(Collector):
    """Poll the netatmo servers for data.  Put the result on the queue.
    noise is measured in dB
    co2 is measured in ppm
    rain is measured in mm
    temperatures are measured in C or F
    the user object indicates the units of the download
      unit: 0: metric, 1: imperial
      windunit: 0: kph, 1: mph, 2: m/s, 3: beafort, 4: knot
      pressureunit: 0: mbar, 1: inHg, 2: mmHg
      lang: user locale
      reg_locale: regional preferences for date
      feel_like_algo: 0: humidex, 1: heat-index

    rf_status is a mapping to rssi (+dB)
      0: 90, 1: 80, 2: 70, 3: 60

    wifi_status is a mapping to rssi (+dB)
      0: 86, 1: 71, 2: 56

   battery_vp measures battery capacity 
  indoor: 6000-4200; 0: 5640, 1: 5280, 2: 4920, 3: 4560
  rain/outdoor: 6000-3600; 0: 5500, 1: 5000, 2: 4500, 3: 4000
  wind: 6000-3950; 0: 5590, 1: 5180, 2: 4770, 3: 4360
    """ `
I have tryed to figuring out how to do to get the correct reading for this information but i cant get it. 
I even tryed to change the extended schema but with out respons so i change it back there to basic again. 
Tryed now for 3 weeks and now i ask for your help. 

I have write this as issue in netatmo extension to Matthewwall to but didnt get respons so i ask for help here. 

I hope someone can help me to how i can configuring this in essy way or if someone mabey have a working configfile to share with me that works 👍

tkeffer, 2021-01-12:
This issues list is for bug reporting and tracking. You are better off posting to weewx-user (which, I see, you have).

closed 2021-01-12 by tkeffer

RebeckaJohansson, 2021-01-12:
Thanks for answer. I post it to weewx-user 👍