weewx/weewx #640: WOW uploads too frequent, sucesses marked as failure

poohbeth, 2021-01-19:
I've been noticing in my logs that MetOffice WOW uploads are often failing. WOW only accepts new observations once an hour - as seen from those actually published. 

In the below snippet (logging everything ) it fails to upload at Jan 19 19:55:26 except when I look at WOW itself the upload did actually succeed as the timestamp matches. 

Jan 19 19:45:25 - failed
Jan 19 19:55:26 - failed - except it appears in WOW
Jan 19 20:00:15 - failed 

```
Jan 19 19:45:16 roo weewx[17796] DEBUG weeutil.rsyncupload: rsyncupload: cmd: [['rsync', '--archive', '--stats', '-e ssh', '/home/weewx/public_html/', 'beth@pooh:/var/www/html/weewx']]
Jan 19 19:45:25 roo weewx[17796] DEBUG weewx.restx: WOW: Failed upload attempt 1: timed out
Jan 19 19:45:40 roo weewx[17796] DEBUG weewx.restx: WOW: Failed upload attempt 2: timed out
Jan 19 19:45:53 roo weewx[17796] DEBUG weewx.restx: WOW: Failed upload attempt 3: HTTP Error 429: 
Jan 19 19:45:53 roo weewx[17796] ERROR weewx.restx: WOW: Failed to publish record 2021-01-19 19:45:00 GMT (1611085500): Failed upload after 3 tries
Jan 19 19:50:14 roo weewx[17796] INFO weewx.manager: Added record 2021-01-19 19:50:00 GMT (1611085800) to database 'weewx'
Jan 19 19:50:14 roo weewx[17796] INFO weewx.manager: Added record 2021-01-19 19:50:00 GMT (1611085800) to daily summary in 'weewx'
Jan 19 19:50:14 roo weewx[17796] DEBUG weewx.restx: StationRegistry: wait interval (1800 < 604800) has not passed for record 2021-01-19 19:50:00 GMT (1611085800)
Jan 19 19:50:14 roo weewx[17796] DEBUG weewx.reportengine: Running reports for latest time in the database.
Jan 19 19:50:14 roo weewx[17796] DEBUG weewx.reportengine: Running report 'SeasonsReport'
Jan 19 19:50:14 roo weewx[17796] DEBUG weewx.reportengine: Found configuration file /home/weewx/skins/Seasons/skin.conf for report 'SeasonsReport'
Jan 19 19:50:14 roo weewx[17796] DEBUG weewx.cheetahgenerator: Using search list ['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Station', 'weewx.cheetahgenerator.Current', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInfo', 'weewx.cheetahgenerator.Extras']
Jan 19 19:50:14 roo weewx[17796] DEBUG weewx.manager: Daily summary version is 3.0
Jan 19 19:50:15 roo weewx[17796] INFO weewx.restx: WOW: Published record 2021-01-19 19:50:00 GMT (1611085800)
Jan 19 19:50:16 roo weewx[17796] DEBUG weewx.manager: Daily summary version is 3.0
Jan 19 19:50:16 roo weewx[17796] DEBUG weewx.reportengine: Report 'SmartphoneReport' not enabled. Skipping.
Jan 19 19:50:16 roo weewx[17796] DEBUG weewx.reportengine: Report 'MobileReport' not enabled. Skipping.
Jan 19 19:50:16 roo weewx[17796] DEBUG weewx.reportengine: Report 'StandardReport' not enabled. Skipping.
Jan 19 19:50:16 roo weewx[17796] DEBUG weewx.reportengine: Report 'FTP' not enabled. Skipping.
Jan 19 19:50:16 roo weewx[17796] DEBUG weewx.reportengine: Running report 'RSYNC'
Jan 19 19:50:16 roo weewx[17796] DEBUG weewx.reportengine: Found configuration file /home/weewx/skins/Rsync/skin.conf for report 'RSYNC'
Jan 19 19:50:16 roo weewx[17796] DEBUG weeutil.rsyncupload: rsyncupload: cmd: [['rsync', '--archive', '--stats', '-e ssh', '/home/weewx/public_html/', 'beth@pooh:/var/www/html/weewx']]
Jan 19 19:55:15 roo weewx[17796] INFO weewx.manager: Added record 2021-01-19 19:55:00 GMT (1611086100) to database 'weewx'
Jan 19 19:55:15 roo weewx[17796] INFO weewx.manager: Added record 2021-01-19 19:55:00 GMT (1611086100) to daily summary in 'weewx'
Jan 19 19:55:15 roo weewx[17796] DEBUG weewx.reportengine: Running reports for latest time in the database.
Jan 19 19:55:15 roo weewx[17796] DEBUG weewx.reportengine: Running report 'SeasonsReport'
Jan 19 19:55:15 roo weewx[17796] DEBUG weewx.restx: StationRegistry: wait interval (2100 < 604800) has not passed for record 2021-01-19 19:55:00 GMT (1611086100)
Jan 19 19:55:15 roo weewx[17796] DEBUG weewx.reportengine: Found configuration file /home/weewx/skins/Seasons/skin.conf for report 'SeasonsReport'
Jan 19 19:55:15 roo weewx[17796] DEBUG weewx.cheetahgenerator: Using search list ['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Station', 'weewx.cheetahgenerator.Current', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInfo', 'weewx.cheetahgenerator.Extras']
Jan 19 19:55:15 roo weewx[17796] DEBUG weewx.manager: Daily summary version is 3.0
Jan 19 19:55:16 roo weewx[17796] DEBUG weewx.manager: Daily summary version is 3.0
Jan 19 19:55:17 roo weewx[17796] DEBUG weewx.reportengine: Report 'SmartphoneReport' not enabled. Skipping.
Jan 19 19:55:17 roo weewx[17796] DEBUG weewx.reportengine: Report 'MobileReport' not enabled. Skipping.
Jan 19 19:55:17 roo weewx[17796] DEBUG weewx.reportengine: Report 'StandardReport' not enabled. Skipping.
Jan 19 19:55:17 roo weewx[17796] DEBUG weewx.reportengine: Report 'FTP' not enabled. Skipping.
Jan 19 19:55:17 roo weewx[17796] DEBUG weewx.reportengine: Running report 'RSYNC'
Jan 19 19:55:17 roo weewx[17796] DEBUG weewx.reportengine: Found configuration file /home/weewx/skins/Rsync/skin.conf for report 'RSYNC'
Jan 19 19:55:17 roo weewx[17796] DEBUG weeutil.rsyncupload: rsyncupload: cmd: [['rsync', '--archive', '--stats', '-e ssh', '/home/weewx/public_html/', 'beth@pooh:/var/www/html/weewx']]
Jan 19 19:55:26 roo weewx[17796] DEBUG weewx.restx: WOW: Failed upload attempt 1: timed out
Jan 19 19:55:32 roo weewx[17796] DEBUG weewx.restx: WOW: Failed upload attempt 2: HTTP Error 429: Too frequent observations.
Jan 19 19:55:37 roo weewx[17796] DEBUG weewx.restx: WOW: Failed upload attempt 3: HTTP Error 429: Too frequent observations.
Jan 19 19:55:37 roo weewx[17796] ERROR weewx.restx: WOW: Failed to publish record 2021-01-19 19:55:00 GMT (1611086100): Failed upload after 3 tries
Jan 19 20:00:14 roo weewx[17796] INFO weewx.manager: Added record 2021-01-19 20:00:00 GMT (1611086400) to database 'weewx'
Jan 19 20:00:14 roo weewx[17796] INFO weewx.manager: Added record 2021-01-19 20:00:00 GMT (1611086400) to daily summary in 'weewx'
Jan 19 20:00:14 roo weewx[17796] DEBUG weewx.restx: StationRegistry: wait interval (2400 < 604800) has not passed for record 2021-01-19 20:00:00 GMT (1611086400)
Jan 19 20:00:14 roo weewx[17796] DEBUG weewx.reportengine: Running reports for latest time in the database.
Jan 19 20:00:14 roo weewx[17796] DEBUG weewx.reportengine: Running report 'SeasonsReport'
Jan 19 20:00:14 roo weewx[17796] DEBUG weewx.reportengine: Found configuration file /home/weewx/skins/Seasons/skin.conf for report 'SeasonsReport'
Jan 19 20:00:14 roo weewx[17796] DEBUG weewx.cheetahgenerator: Using search list ['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Station', 'weewx.cheetahgenerator.Current', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInfo', 'weewx.cheetahgenerator.Extras']
Jan 19 20:00:14 roo weewx[17796] DEBUG weewx.manager: Daily summary version is 3.0
Jan 19 20:00:15 roo weewx[17796] INFO weewx.restx: WOW: Published record 2021-01-19 20:00:00 GMT (1611086400)
Jan 19 20:00:15 roo weewx[17796] DEBUG weewx.manager: Daily summary version is 3.0
Jan 19 20:00:18 roo weewx[17796] DEBUG weewx.reportengine: Report 'SmartphoneReport' not enabled. Skipping.
Jan 19 20:00:18 roo weewx[17796] DEBUG weewx.reportengine: Report 'MobileReport' not enabled. Skipping.
```

[image]

chaunceygardiner, 2021-01-19:
> WOW only accepts new observations once an hour - as seen from those actually published.

Perhaps I’m missing something here, but I see updates every 5 minutes for my station:

https://wow.metoffice.gov.uk/observations/details?site_id=e7824f12-c455-ea11-b698-0003ff59771b

tkeffer, 2021-01-20:
Mine uploads every 5 minutes as well, although there are frequent failures. For example, from this morning:

```
Jan 19 09:20:27 hummingbird weewx[28563] INFO weewx.restx: WOW: Published record 2021-01-19 09:20:00 PST (1611076800)
Jan 19 09:25:30 hummingbird weewx[28563] DEBUG weewx.restx: WOW: Failed upload attempt 1: timed out
Jan 19 09:25:41 hummingbird weewx[28563] DEBUG weewx.restx: WOW: Failed upload attempt 2: HTTP Error 429: Too frequent observations.
Jan 19 09:25:48 hummingbird weewx[28563] DEBUG weewx.restx: WOW: Failed upload attempt 3: HTTP Error 429: Too frequent observations.
Jan 19 09:25:48 hummingbird weewx[28563] ERROR weewx.restx: WOW: Failed to publish record 2021-01-19 09:25:00 PST (1611077100): Failed upload after 3 tries
Jan 19 09:30:21 hummingbird weewx[28563] INFO weewx.restx: WOW: Published record 2021-01-19 09:30:00 PST (1611077400)
Jan 19 09:35:24 hummingbird weewx[28563] INFO weewx.restx: WOW: Published record 2021-01-19 09:35:00 PST (1611077700)
Jan 19 09:40:21 hummingbird weewx[28563] INFO weewx.restx: WOW: Published record 2021-01-19 09:40:00 PST (1611078000)
Jan 19 09:45:30 hummingbird weewx[28563] DEBUG weewx.restx: WOW: Failed upload attempt 1: timed out
Jan 19 09:45:46 hummingbird weewx[28563] DEBUG weewx.restx: WOW: Failed upload attempt 2: timed out
Jan 19 09:45:51 hummingbird weewx[28563] DEBUG weewx.restx: WOW: Failed upload attempt 3: HTTP Error 429: Too frequent observations.
Jan 19 09:45:51 hummingbird weewx[28563] ERROR weewx.restx: WOW: Failed to publish record 2021-01-19 09:45:00 PST (1611078300): Failed upload after 3 tries
Jan 19 09:50:16 hummingbird weewx[28563] INFO weewx.restx: WOW: Published record 2021-01-19 09:50:00 PST (1611078600)
Jan 19 09:55:21 hummingbird weewx[28563] INFO weewx.restx: WOW: Published record 2021-01-19 09:55:00 PST (1611078900)
Jan 19 10:00:31 hummingbird weewx[28563] DEBUG weewx.restx: WOW: Failed upload attempt 1: timed out
```

In any case, if you want it less often, there's already an option for that: `post_interval`.

I'm not seeing a bug here, so I'm closing this.

closed 2021-01-20 by tkeffer

chaunceygardiner, 2021-01-20:
Agreed on the frequent failures.  For me, 34 yesterday and 14 so far today.

poohbeth, 2021-01-20:
tkeffer - so you are seeing the issue in the logs, so is chaunceygardiner but it's not a problem.

I see it as a problem that'll come up again until someone sorts out the  default config, at least.  Shrug.

tkeffer, 2021-01-20:
Take a look at the logs, including your own. There are many examples where weewxd successfully uploaded more than once an hour. The failures come at random times. 

Take a look at the WOW API: https://wow.metoffice.gov.uk/support/dataformats. It explicitly says,

> It is recommended that the interval between automatic observations from your AWS is at least 5 minutes.

In short, I see no evidence that WOW requires uploads only once an hour. In any case, if you don't believe that, use option `post_interval`.