weewx/weewx #648: `wee_database --calc-missing` fails if the archive table is not named `archive`

gjr80, 2021-02-14:
If `wee_database --calc-missing` is executed with a user specified `--binding` then `wee_database` will fail if the archive table table name is not `archive`.

Refer comments in commit 9cecd13d0c5f5026800e471d656850562b10b9d7 (for some reason this commit just appeared, it is not under a PR nor any other issue)

gjr80, 2021-02-14:
**yesjpm**:

Tried to fix wee_database --calc-missing for non 'wx_binding' binding
I have my doubts about the second change (the 'StdArchive'). It works. But It still doesn't use the --binding parameter.
// Jaap

**gjr80**:

Can I ask for some details of what problem we are trying to fix here? Not that any are required, but I haven’t seen any issues raised or problems posted in the user group. I would like to understand the problem before considering the solution.

**yesjpm**:

munckjes@DS77U:/home/weewx/bin$ ./wee_database --config=../weewx2.conf --binding=p1_binding --calc-missing --date=2021-02-12
Using configuration file ../weewx2.conf
Using database binding 'p1_binding', which is bound to database 'p1_mysql'
Traceback (most recent call last):
File "/home/weewx/bin/weedb/mysql.py", line 52, in guarded_fn
return fn(*args, **kwargs)
File "/home/weewx/bin/weedb/mysql.py", line 206, in columnsOf
column_list = [row[1] for row in self.genSchemaOf(table)]
File "/home/weewx/bin/weedb/mysql.py", line 206, in
column_list = [row[1] for row in self.genSchemaOf(table)]
File "/home/weewx/bin/weedb/mysql.py", line 179, in genSchemaOf
cursor.execute("""SHOW COLUMNS IN %s;""" % table)
File "/usr/lib/python3/dist-packages/MySQLdb/cursors.py", line 209, in execute
res = self._query(query)
File "/usr/lib/python3/dist-packages/MySQLdb/cursors.py", line 315, in _query
db.query(q)
File "/usr/lib/python3/dist-packages/MySQLdb/connections.py", line 226, in query
_mysql.connection.query(self, query)
MySQLdb._exceptions.ProgrammingError: (1146, "Table 'p1.archive' doesn't exist")

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "./wee_database", line 1023, in
main()
File "./wee_database", line 165, in main
calc_missing(config_dict, db_binding, options)
File "./wee_database", line 652, in calc_missing
with weewx.manager.Manager.open(manager_dict['database_dict']) as dbmanager:
File "/home/weewx/bin/weewx/manager.py", line 132, in open
dbmanager = cls(connection, table_name)
File "/home/weewx/bin/weewx/manager.py", line 83, in init
self.sqlkeys = self.connection.columnsOf(self.table_name)
File "/home/weewx/bin/weedb/mysql.py", line 61, in guarded_fn
raise klass(e)
weedb.NoTableError: (1146, "Table 'p1.archive' doesn't exist")

**yesjpm**:

This is de config used:
```
```
[DataBindings]
```

    [[p1_binding]]
        # The database must match one of the sections in [Databases].
        # This is likely to be the only option you would want to change.
        database = p1_mysql
        # The name of the table within the database
        table_name = p1elect
        # The manager handles aggregation of data for historical summaries
        #manager = weewx.wxmanager.WXDaySummaryManager
        manager = weewx.manager.DaySummaryManager
        # The schema defines the structure of the database.
        # It is *only* used when the database is created.
        schema = user.p1.p1_schema

    [[p1_gas_binding]]
        database = p1_mysql
        table_name = p1gas
        #manager = weewx.wxmanager.WXDaySummaryManager
        manager = weewx.manager.DaySummaryManager
        schema = user.p1.p1gas_schema
```
```

**gjr80**:

Thanks. `wee_database` change seems appropriate but I think rather that making a change to `wxservices.py` you need to make a change to the `class DatabaseFix() __init__()` method similar to what you did for `wee_database`. I will need to have a closer look when home and in front of the pc, browsing GitHub on a mobile device is less than optimal.

Also, your commit seems to have randomly appeared in the GitHub repo, I presume you have committed to your fork and pushed to upstream? Let’s just work through this here and since the changes appear simple we can just commit to the WeeWX repo separately rather than you raising a PR as normal.

**yesjpm**:

Yeah. Something has gone wrong: tried to create a PR (and was surprised you found it). 

This is the crash for the second change (so after first one was fixed):
Munckjes@DS77U:/home/weewx/bin$ ./wee_database --config=../weewx2.conf --binding=p1_binding --calc-missing --from=2020-11-11
Using configuration file ../weewx2.conf
Using database binding 'p1_binding', which is bound to database 'p1_mysql'
Missing derived observations will be calculated from 2020-11-11 00:00:00 CET (1605049200) through to the end (2021-02-13 22:10:00 CET (1613250600)).
Proceed (y/n)? y
Calculating missing derived observations...
Traceback (most recent call last):
  File "/home/weewx/bin/weewx/manager.py", line 636, in get_manager_dict_from_config
    manager_dict = dict(config_dict['DataBindings'][data_binding])
  File "/usr/lib/python3/dist-packages/configobj.py", line 554, in __getitem__
    val = dict.__getitem__(self, key)
KeyError: 'wx_binding'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./wee_database", line 1024, in <module>
    main()
  File "./wee_database", line 165, in main
    calc_missing(config_dict, db_binding, options)
  File "./wee_database", line 709, in calc_missing
    calc_missing_obj.run()
  File "/home/weewx/bin/weecfg/database.py", line 415, in run
    wxcalculate = weewx.wxservices.StdWXCalculate(engine, self.config_dict)
  File "/home/weewx/bin/weewx/wxservices.py", line 92, in __init__
    self.db_manager = engine.db_binder.get_manager(data_binding=data_binding, initialize=True)
  File "/home/weewx/bin/weewx/manager.py", line 531, in get_manager
    manager_dict = get_manager_dict_from_config(self.config_dict,
  File "/home/weewx/bin/weewx/manager.py", line 638, in get_manager_dict_from_config
    raise weewx.UnknownBinding("Unknown data binding '%s'" % e)
weewx.UnknownBinding: Unknown data binding ''wx_binding''
munckjes@DS77U:/home/weewx/bin$

yesjpm, 2021-02-15:
@gjr80: I agree that the change in `wxservices.py` is not  #sufficient.
Here is a more complete version:  [https://github.com/yesjpm/weewx/commits/master](url)

gjr80, 2021-05-22:
Getting back to this...finally.

There are a couple of issues here:

1. The `StdWXCalculate` service is used by `wee_database --calc-missing` to calculate missing derived obs. `StdWXCalculate` requires access to archive data for calculating some derived obs, the database used by `StdWXCalculate` is specified via the `[StdWXCalculate]` `data_binding` config option in `weewx.conf`. This option is silent (and before `v4.6.0` undocumented) and defaults to `wx_binding`. The `[StdWXCalculate]` `data_binding` config option may be different to the data binding used by the `StdArchive` service which is set via `[StdArchive]` `data_binding` (I struggle to understand why a user might want such a config but it is possible and supported). The upshot is when using custom binding names it is quite easy for `StdWXCalculate` to continue to use the default `wx_binding`. At best this will cause some derived obs to be inaccurate or (more likely) not be calculated, at worst an error will be thrown when running `wee_database --calc-missing`.
2. Whilst `wee_database --calc-missing` accepts a `--binding` command line option when accessing the database specified by `--binding` `wee_database` does not pass the `table_name` parameter to weewx.Manager and hence the default `archive` table name is used irrespective of the `table_name` setting for the specified binding.

I have fixed these issues at commit e472e402e9b54e56c4c5c1891a03711e1db5dddb. In addition I have tidied up the handling of non-existent bindings by `wee_database` to give the user a bit better feedback rather than a raw error trace.

gjr80, 2021-05-22:
Wrapping this issue up with a note to the `wee_database` `--calc-missing` section of the [Utilities Guide](http://weewx.com/docs/utilities.htm) regarding the linkage between the binding (implicit or explicit) used by `--calc-missing` and the binding used by the `StdWXCalculate` service. Commit 2b4c165d4bdf64b751b722de0681e2a8f72c2785 refers.

closed 2021-05-22 by gjr80