weewx/weewx #649: weewx 4.4.0: No such table: archive_day_wind

poblabs, 2021-02-14:
Fresh install of weewx 4.4 on Ubuntu 20.04

```
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine: Caught unrecoverable exception in generator 'weewx.imagegenerator.ImageGenerator'
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****  no such table: archive_day_wind
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****  Traceback (most recent call last):
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weedb/sqlite.py", line 29, in guarded_fn
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      return fn(*args, **kwargs)
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weedb/sqlite.py", line 211, in execute
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      return sqlite3.Cursor.execute(self, *args, **kwargs)
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****  sqlite3.OperationalError: no such table: archive_day_wind
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****  During handling of the above exception, another exception occurred:
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****  Traceback (most recent call last):
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/reportengine.py", line 196, in run
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      obj.start()
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/reportengine.py", line 281, in start
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      self.run()
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/imagegenerator.py", line 41, in run
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      self.genImages(self.gen_ts)
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/imagegenerator.py", line 176, in genImages
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      start_vec_t, stop_vec_t ,data_vec_t = weewx.xtypes.get_series(var_type,
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/xtypes.py", line 85, in get_series
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      return xtype.get_series(obs_type, timespan, db_manager, aggregate_type,
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/xtypes.py", line 618, in get_series
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      return ArchiveTable.get_series(obs_type, timespan, db_manager, aggregate_type,
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/xtypes.py", line 140, in get_series
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      agg_vt = get_aggregate(obs_type, stamp, do_aggregate, db_manager)
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/xtypes.py", line 101, in get_aggregate
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      return xtype.get_aggregate(obs_type, timespan, aggregate_type, db_manager,
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/xtypes.py", line 787, in get_aggregate
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      row = db_manager.getSql(sql, timespan)
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/manager.py", line 430, in getSql
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      _cursor.execute(sql, sqlargs)
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weedb/sqlite.py", line 37, in guarded_fn
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****      raise weedb.NoTableError(e)
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****  weedb.NoTableError: no such table: archive_day_wind
Feb 14 16:05:18 weewxflagpole weewx[8769] ERROR weewx.reportengine:         ****  Generator terminated
```

tkeffer, 2021-02-14:
There have been several threads on this issue. Here's one: https://groups.google.com/g/weewx-user/c/sunTiSNUn6s/m/qaLVDckeBgAJ . In particular, see [this post](https://groups.google.com/g/weewx-user/c/sunTiSNUn6s/m/KbXBM-QfBgAJ).

closed 2021-02-14 by tkeffer

poblabs, 2021-02-14:
@tkeffer My concern here is that it happened on a virgin install of weewx, with a brand new archive. Shouldn't that not be the case?

tkeffer, 2021-02-14:
So, we're good?

poblabs, 2021-02-14:
Yep, think so. Thanks again