weewx/weewx #650: Updates to PWSWeather are sent as plaintext

genesispc, 2021-02-17:
Looking through my router log I noticed that updates to PWSWeather are sent as plaintext, meaning that the password used to update the values is visible to all. Can I suggest changing the following line (line 720) in /bin/weewx/restx.py

from:

    archive_url = "http://www.pwsweather.com/pwsupdate/pwsupdate.php"

to

    archive_url = "**https**://www.pwsweather.com/pwsupdate/pwsupdate.php"

I've tried it locally and it works a treat.

chaunceygardiner, 2021-02-17:
I tested this change, and also changed the URL for WOW (in the same file) to use https.  Both worked for me.  It looks like the others are already using https.

tkeffer, 2021-02-17:
Fixed in commit https://github.com/weewx/weewx/commit/bada58c979fa8572391657960cd3c993c28c4f00

Thanks, @genesispc  for the fix, and John (@chaunceygardiner) for testing it!

closed 2021-02-17 by tkeffer