weewx/weewx #653: Reference to StdWXCalculate in XType Documentation

roe-dl, 2021-03-03:
I wrote an XType extension. The documention describes how to do that and how to make the extension loaded by WeeWX. After that the observation types calculated by the extension can be displayed in skins, but they are *not* included in the record and the MQTT export.

To include the values there, it is necessary to add lines in the `[StdWXCalculate]` `[[Calculations]]` section of weewx.conf like

    observation_type = software

I think it would be helpful to have that mentioned in the XType documentation.

(And I am not sure if I could and am allowed to add it there myself.)

tkeffer, 2021-03-03:
There is already language to this effect in the wiki. See the section [_Populate packets and records_](https://github.com/weewx/weewx/wiki/WeeWX-V4-user-defined-types#populate-packets-and-records) 

Was something unclear about that? If so, anyone can edit the wiki. Give it a try!

roe-dl, 2021-03-04:
For my thinking the section "Overview of adding new types" is like a checklist, saying what to do step by step. There, all steps are mentioned except that last step involving "StdWXCalculate".

The section you referenced is a subsection under "A comprehensive example". I considered that as additional or deeper information. As I found all the information I needed in the previous section I did not read that additional section. 

I will try to add, what I think would help improving the documentation.

roe-dl, 2021-03-05:
I wrote it yesterday evening, and I hope the result is ok for you.

closed 2021-03-09 by roe-dl